realsenseai/librealsense #5973: How to record video and audio simultaneously via intel realsense camera D415 and external microphone

IFICL, 2020-03-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.11.6.250 | 
| Operating System & Version |   {Win 10 / Linux (Ubuntu 18) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC|
| SDK Version                            |  { legacy / 2.0 }                          | 
| Language                            |  C/python/                          | 
| Segment			|  others             | 

### Issue Description
Is there any way to record video and audio simultaneously via intel realsense camera D415 and external microphone? If I use python to control both of them, It seems cannot record in the same time.

MartyG-RealSense, 2020-03-04:
There is no mechanism in the current RealSense SDK to record audio, as the 400 Series cameras do not have microphones.  The last time that a similar question was asked, I recommended trying to do the audio and video capture with Windows via Microsoft's DirectShow API.

https://github.com/IntelRealSense/librealsense/issues/5350#issuecomment-559947283

There was someone who created a tutorial article for accessing DirectShow with Python.

https://www.codeproject.com/Articles/1274094/Capturing-Images-from-Camera-using-Python-and-Dire

IFICL, 2020-03-06:
Thanks. I will try another way.

closed 2020-03-06 by IFICL