realsenseai/librealsense #5976: Openni .exe run problem

alexx600, 2020-03-04:
Camera Model: D435
Realsense: WIN10-2.32.1.1297
OS : Windows 10

I compiled openni2 wrapper to get rs2driver.dll. Then i copied this file and Intel RealSense SDK 2.0\bin\x64\rs2driver.dll to C:\Program Files\OpenNI2\Samples\Bin\OpenNI2\Drivers.

My Global variable are:
OPENNI2_DIR C:\Program Files\OpenNI2
REALSENSE2_DIR C:\Program Files (x86)\Intel RealSense SDK 2.0

In Intel RealSense Viewer stream is avaible.
In OpenNI2 Simple read i got 

C:\Program Files\OpenNI2\Samples\Bin>"C:\Program Files\OpenNI2\Samples\Bin\SimpleRead.exe"
Couldn't open device DeviceOpen using default: no devices found

I would appreciate any advice.
Best Regards

dorodnic, 2020-03-05:
Hi @alexx600 
I didn't build the samples but was able to run NiViewer by copying binaries also to `C:\Program Files (x86)\OpenNI2\Tools\OpenNI2\Drivers`. Please note that NiViewer is 32-bit application.
I'd recommend checking 64/32 versions, and check that both `realsense2.dll` and `rs2driver.dll` are in the right place.

alexx600, 2020-03-05:
Hi @dorodnic thank you very much for advice. I noticed that openni2 i have installed was x64. So i installed x86 version and copied dll. Now it workds.

closed 2020-03-05 by alexx600