realsenseai/librealsense #5978: SR305: The installer failed to detect an Intel® RealSense™ 3D camera on this system. The installation was terminated.

dzoech, 2020-03-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR305 | 
| Firmware Version                   | Here is the issue... | 
| Operating System & Version |  Windows 10, Version 1909, Build 18363.657  | 
| Platform                                 | PC  |

### Issue Description
I try to setup the SR305 camera on my workstation (notebook) but when installing the Depth Camera Manager (DCM), the setup fails with the following error message:

  The installer failed to detect an Intel® RealSense™ 3D camera on this system. The installation was terminated.

[image]

I use this setup: intel_rs_dcm_sr300_3.3.27.5718.exe
Since there is no setup dedicated for the SR305, I guess the one for the SR300 should be correct.

I allready disabled power saving for the USB ports and tried all ports even the one on the docking station. 
[image]

Device Manager:
[image]

I tried every possible approach I found online, but nothing helped. Do you have any idea what the problem could be?

MartyG-RealSense, 2020-03-05:
In 2017 Intel adopted an online installation method for the SR300, where instead of using the DCM program, the drivers could be installed automatically via Windows Update just by plugging the camera into the USB port on a PC with an internet connection.  I assume the same is true for the SR305, since it has an SR300 camera board inside it.

It looks as though you already have the drivers correctly installed in the 'Cameras' section of your Device Manager **(Intel RealSense(TM) Depth Camera SR305** and **Intel RealSense(TM) RGB Camera SR305**).

Could you try testing the camera by downloading and installing the RealSense SDK software and launching the 'Intel RealSense Viewer' program that the installation places on your desktop please?

The RealSense SDK can be downloaded on Windows by going to the bottom of the download page in the link below and left-clicking on the **Intel.RealSense.SDK-WIN10-2.33.1.1360.exe** link to start downloading the installer file in your browser.  When installation is complete, please run the file to install the RealSense SDK software and associated programs such as the RealSense Viewer.

dzoech, 2020-03-06:
Thank you @MartyG-RealSense for the quick reply. I already had installed the SDK but couldn't fix a specific error, therefore I thought I'm missing something...

The RealSense Viewer works fine, both, for RGB and depth.

Maybe you could help me out with my initial problem (or should I create a new issue for it?):

I tried several libraries built on top of the RealSense SDK, e.g. Microsoft's Project Gesture, Nuitrack, Intel/cubemos Skeleton SDK, etc. Some don't officially support the SR305 but I though I'd still give it a shot. But with all libraries I tested, the result is that no sensor/camera could be initialized and a common error message - seemingly returned by the RealSense SDK - is:

> Couldn't resolve requests

Is there a possibility to gather more information what this means or what the underlying problem might be? I came across possible incompatibilities with resolution or FPS but If I understand correctly 640x480@30 FPS should work.

MartyG-RealSense, 2020-03-06:
Although Project Gesture says that it can work with the SR300 camera, I cannot rule out the possibility that there is an incompatibility with SR305 (since Intel have said that the older SR300-compatible RealSense SDK for Windows from 2016 cannot be used with the SR305).

I tested the Cubemos SDK with an SR300 the other day.  Although a couple of the graphical samples would not detect the camera, the webcam example program did work and overlay joint information over an RGB video.  The non-graphical plain text console examples also detected a skeleton and returned information.  One RealSense user of SR305 wondered if they might be able to rewrite the non-graphical examples to render graphics in a way that worked with their camera.

If the webcam sample program is treating the camera like a webcam instead of a RealSense camera then there is a good chance that it will work with SR305.

[image]

RealSenseCustomerSupport, 2020-03-06:
----------------------------------------------

Hi dzoech,
The DCM is only support SR300 not SR305
They have different USB PID/VID, so they could be  enumerated as different name under device manager.
So that is a expected behavior that fail to install the SR300 DCM for SR305 even they looks the same.

The new has SDK has different software architecture that no need DCM to handle the depth streaming. that is why you can streaming the both RGB and depth without installing DCM..

dzoech, 2020-03-06:
Thank you both for your fast support @MartyG-RealSense and @RealSenseCustomerSupport 

The whole situation seems a bit complicated but I think I now understand most of the issue.

My situation is that I'd like to setup a prototype for gesture recognition, which leaves me with two options:

1. Microsoft Project Gesture that uses SDK 2016 and requires an SR300 (Device ID: 0x0AA5). Is there a way to simulate the SR305 (Device ID: 0x0B48) as an SR300? I found this comment https://github.com/IntelRealSense/librealsense/issues/4435#issuecomment-525617614 but I am unable do find a download for firmware version V3.26.1.0 or V3.25
2. Nuitrack which is built upon RealSense SDK V2 but doesn't officially support SR300 nor SR305 but according to a comment I've found, it might be possible to use it. The problem here is the error message from above `Couldn't resolve requests`, which I'm having troubles investigation further.

dzoech, 2020-03-06:
Quick update: I now installed the RealSense SDK 2.25 which allowed me to downgrade the firmware like explained in the comment mentioned above. The SR305 camera is now recognized as an SR300 and seems to work now with Project Gesture.

MartyG-RealSense, 2020-03-06:
I'm very glad that you found a workaround that provides a solution for you.  Thanks for the update!

MartyG-RealSense, 2020-03-06:
As you got the camera working successfully with a tracking solution, I will close the case.  Good luck with your work!

closed 2020-03-06 by MartyG-RealSense