realsenseai/librealsense #5981: How to display depth at full range in Realsense viewer?

bradsteiner-pff, 2020-03-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | latest | 
| Operating System & Version |   Ubuntu 16 | 
| Kernel Version (Linux Only)    |   | 
| Platform                                 | PC |
| SDK Version                            |  latest  | 
| Language                            |  C++                     | 
| Segment			|      robotics    | 

### Issue Description
I noticed when using the Realsense viewer that the color range for the depth image seems to cap out at about 3 meters. How can I modify this so that the highest color value is something larger (like 20m)?

MartyG-RealSense, 2020-03-05:
It sounds as though what you are describing is the Threshold filter in the 'Post Processing' section of the Viewer's options side-panel being enabled.  This is set to 4 meters by default and excludes depth detail beyond that distance from the camera.  Turning the Threshold filter off will display the depth image as far as the camera's 10 meter depth range can see.

bradsteiner-pff, 2020-03-05:
I have post-processing turned off in my viewer, so I am able to see all of the distances in the display. However, in this attached image, you can see that most of the pixels are colored the same shade of maroon, due to the usable portion of the color scale only covering a range of about 3 meters. How can I modify this display so that the color range covers a larger range of distances?

[image]

MartyG-RealSense, 2020-03-05:
Apologies for the delay in responding further, I was doing extensive tests.  I do not usually experience the dark color that you showed when I test during the daytime - it is usually brightly lit in multi-color right up to the 10 meter maximum range I usually test at.   

But it is 11 pm at night in a dim bulb-lit room with the curtains closed when I am testing now and I am getting exactly the same brown-ish color shade starting at roughly the same distance.  

Are you testing in a room with dim artificial lighting, please?

bradsteiner-pff, 2020-03-05:
Toggling does not seem to affect the issue I am having. As you can see on the distance ruler on the right of the screenshot above, it is correctly identifying the maximum distance of 24 meters. However, the non-maroon colors are all in the 0-3 meter range. Basically I'd like to modify how it maps these colors to distances.

bradsteiner-pff, 2020-03-06:
The room I am testing in is well lit with a mix of sunlight and artificial lighting. I do think this is merely a visualization issue though, and not a problem with the data that is being collected. It seems to be correctly picking up the right range (as seen by the ruler on the right), however the color scale does not stretch to this range. Do you know where this color range is determined in the viewer code?

MartyG-RealSense, 2020-03-06:
400 Series cameras can depth sense to a maximum range of 10 meters by default.  If the scale at the side of the image is forced to a higher maximum scale than 10 by reducing the Depth Units value (e.g to 24 or 48 meters scale maximum) then the results may become more unpredictable.

There may not be a benefit to pushing beyond 10 meters though, as accuracy of measurements starts to drift noticably after about 3 meters.  So by pushing the scale to a high level, you may end up with an image that looks good but does not have depth measurement values that accurately correspond to the real world scene.

bradsteiner-pff, 2020-03-06:
That's ok, I am not too concerned about the accuracy of the farther depths, I would just like to visualize them anyway. Is this possible?

MartyG-RealSense, 2020-03-06:
I have seen the scale go beyond 10 in images so few times that I cannot even find a reference image, unfortunately (though I have seen it done, and obviously was able to replicate it myself by minimising the Depth Unit setting).  

I will tag into this conversation the CTO of the RealSense Group at Intel to see if they can help, since they will have complete knowledge of the camera hardware.  @agrunnet

MartyG-RealSense, 2020-03-19:
Case closed due to no further comments after 2 weeks.

closed 2020-03-19 by MartyG-RealSense