realsenseai/librealsense #5982: D435 color space standard

sh0, 2020-03-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   Linux (Gentoo)  | 
| Kernel Version (Linux Only)    |  5.5.6  | 
| Platform                                 | PC |
| SDK Version                            |  git master | 
| Language                            |  C | 
| Segment			|  others | 

### Issue Description

I am trying to figure out what color space standard is D435 color camera stream using. There does not seem to be any details in code or in RealSense datasheets.

From `color-formats-converter.cpp` it seems that YUYV and UYVY formats are using:
* limited range YCbCr quantization
* ITU-R BT.601 YCbCr encoding
* linear transfer function
* unknown color primaries

This seems somewhat odd choice of settings. Do you have some better info as to what color standard was the stream designed to use?

MartyG-RealSense, 2020-03-07:
Unfortunately I could not find any reference to color space either other than the discussion in the link below.

https://github.com/IntelRealSense/librealsense/issues/2310

@agrunnet  could you comment on this subject please?

MartyG-RealSense, 2020-03-19:
Case closed due to no further comments after 12 days.

closed 2020-03-19 by MartyG-RealSense