realsenseai/librealsense #5985: RealSense D435i doesn't detected in realsense viewer.

ulthra, 2020-03-06:
| Camera Model | D435i |
| Firmware Version | 5.11.1.100 |
| Operating System & Version | Win 10 |

I used RealSense D453i on windows 10 but didn't detected by RealSense viewer.
[image]

RealSense D453i was detected as camera on windows 10 when I checked device manager. (I could use RealSense D453i by CyberLink YouCam 5)
[image]

What should I do? help me please.

MartyG-RealSense, 2020-03-06:
Is the camera detected please if you unplug it from the USB port and then plug it back in?  It should be inserted with a firm, quick motion and not put in slowly.

ulthra, 2020-03-06:
didn't work. What meaning of it?

MartyG-RealSense, 2020-03-06:
And it does not appear if you use the Add Source button to find the camera?

ulthra, 2020-03-06:
Hmm... Nothing. 
When I used RealSense D453i with other computer it detected so I think that firmware update is fine.
I think something is happening in windows 10.

[image]

MartyG-RealSense, 2020-03-06:
The firmware is stored inside the camera, while the drivers in the Device Manager are stored on the computer.  So the firmware gets carried to whatever computer you attach the camera to.

Please check if the Windows camera privacy settings on that particular computer are set to 'On' for access to camera applications.   If camera hardware privacy is set to **Off** (allowing the Device Manager to detect the camera) but the application privacy is set to **On**, then this would give the results you are seeing (the Device Manager sees the camera but the RealSense Viewer application software cannot) 

Instructions for checking the camera privacy settings are in the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/360037039794/comments/360009388833

ulthra, 2020-03-06:
Thank you for teaching me so kindly.
But, I checked it in first step and didn't work.

MartyG-RealSense, 2020-03-06:
The camera hardware and firmware seem to be fine, since the D435i works on your other computer.

On the computer that it does not work on, did you install the full 635 mb RealSense SDK, and not just the RealSense Viewer program please?

[image]

ulthra, 2020-03-06:
Yes. I installed it.
(On the computer that it worked, didn't installed)

MartyG-RealSense, 2020-03-06:
Do you have a different USB 2 or USB 3 port on the computer that you can try the camera in?  Or is is not detected in any of the computer's USB ports?

ulthra, 2020-03-06:
I did and both didn't worked (USB 2 and USB 3).
On the computer detected RealSense worked whichever (USB 2 and USB 3).

ulthra, 2020-03-06:
I googled many of troubleshooting but many of cases are not about on windows 10.
So I think this one is very rare case...

MartyG-RealSense, 2020-03-06:
There was a case similar to this where the D435i was detected in Device Manager but not the Viewer.  It is a bit complicated but may be worth reading.

https://github.com/IntelRealSense/librealsense/issues/3644

Also, the RealSense Viewer and the Depth Quality Tool programs depend on the computer's graphics hardware supporting **OpenGL**.   Popular brands such as Nvidia, AMD Radeon and Intel Iris HD should support OpenGL.   you could try upgrading the graphics drivers on your computer and see if that makes a difference to the Viewer.

ulthra, 2020-03-06:
???
I must to change the value of registry about below?

Computer \ HKEY_LOCAL_MACHINE \ SOFTWARE \ Microsoft \ Windows \ CurrentVersion \ DeliveryOptimization \ Config \ DODownloadMode 

Computer \ HKEY_CURRENT_USER \ Software \ Microsoft \ Windows \ CurrentVersion \ DeliveryOptimization \ SystemSettingsDownloadMode

MartyG-RealSense, 2020-03-06:
Editing the registry of a Windows computer is very dangerous if you are not skilled in doing it and can easily break Windows.  So it is best not to do this unless you are sure that you can do it well.

ulthra, 2020-03-06:
mmm......
It's means that there is a very very rare case that RealSense D453i doesn't work on Windows 10 ...?

MartyG-RealSense, 2020-03-06:
It may be a power issue on the USB ports of that computer, where the power is not stable enough for the camera to be detected.  Not all USB ports are made equal, and a camera may work fine on one computer but not on another.  Plugging the camera into a mains electricity powered USB hub instead of plugging it directly into the computer's USB ports may help when non-detection is caused by a USB power problem.

ulthra, 2020-03-06:
Is it possible that device manager detect the camera when the power is not stable enough? My computer's device manager detect the RealSense and able to use only rgb camera.

MartyG-RealSense, 2020-03-06:
Yes, it is common for a camera to be detectable in the Device Manager but not appear, or rapidly disconnect, in a program.

ulthra, 2020-03-07:
I think the problem related to this...
https://github.com/IntelRealSense/librealsense/issues/1098

I could use IR camera by RealSense Calibrator. 
https://github.com/IntelRealSense/librealsense/issues/3644

[image]

**error log**
 07/03 12:57:05,895 ERROR [12188] (types.h:307) CoCreateInstance(CLSID_SensorManager, NULL, CLSCTX_INPROC_SERVER, IID_PPV_ARGS(&pSensorManager)) returned: HResult 0x800704ec: "This program is blocked by group policy. For more information, contact your system administrator."
 07/03 12:57:05,964 WARNING [12188] (ds5-factory.cpp:948) DS5 group_devices is empty.
 07/03 12:57:05,964 WARNING [12188] (ds5-factory.cpp:948) DS5 group_devices is empty.

MartyG-RealSense, 2020-03-07:
If you are experiencing the group policy error, you could try methods 1 and 2 in the page that @dorodnic linked to in the discussion that you showed me.

https://www.isunshare.com/blog/this-program-is-blocked-by-group-policy-error-windows-10/

ulthra, 2020-03-07:
I really Thank you MartyG-RealSense (I took many of times from you so sorry....)
I finally could solve the problem to try method 3 below.
https://www.isunshare.com/blog/this-program-is-blocked-by-group-policy-error-windows-10/

I wrote down the methods to solve the problem to make sure.

1. Open the Registry Editor.
2. Delete all items in Microsoft folder.
HKEY_LOCAL_MACHINE\Software\Policies\Microsoft

I really thank you MartyG-RealSense!
I hope all of users can use great device: RealSense D435i.

closed 2020-03-07 by ulthra

MartyG-RealSense, 2020-03-07:
It's no trouble at all, I'm pleased I could help you find a solution.  Thanks so much for the update!  I'm very glad that you were able to succeed in editing the Windows registry.  Good luck!

ulthra, 2020-03-07:
I really thank you. See ya!

MigVega, 2021-03-27:
> I really Thank you MartyG-RealSense (I took many of times from you so sorry....)
> I finally could solve the problem to try method 3 below. [...]

Thanks, It worked for me!
Just delete the complete Microsoft folder.

MartyG-RealSense, 2021-03-28:
Thanks very much @MigVega for sharing your own particular solution!