realsenseai/librealsense #5986: Saved images has sometimes a line.

hojunson, 2020-03-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (05.12.01.00) | 
| Operating System & Version |   { Linux (Ubuntu 14/16/17)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PNVIDIA Jetson  |
| SDK Version                            |                     | 
| Language                            |  {C }                          | 
| Segment			|  {Robot}                        | 

### Issue Description
<Describe your issue / question / feature request / etc..>
[image]

Here is an example image. 
I am confused why the line in the image was happened. 
I am using a Jetson nano and saving data streaming into a .bag file.

I cannot see the line in the Laptop. It might be Jetson problem or anything I can do?

Thanks.

MartyG-RealSense, 2020-03-06:
It does it look like it is just a line across the screen, as the top section of the image above the line does not align with the rest of the image beneath and has different color shading in places.

It appears as though the RGB is "buffering" , like when an internet video stutters or a television has a bad signal reception.  It looks as though the top of the image is a second ahead of the rest of the image, like a video of someone walking down a corridor where the top of the video is showing the position in the corridor that the person is currently at and the rest of the picture shows where they were at a second ago before they reached their current position.

If the image is okay on the laptop then it is probably not the florescent ceiling strip lights affecting the image and it is instead a consequence of recording to a bag with the Nano's lesser hardware capabilities.  You could try reducing the resolution and / or the FPS speed to put less processing demands on the Nano.

hojunson, 2020-03-07:
Thanks.
Updating SDK and reducing resolution was helpful.

MartyG-RealSense, 2020-03-07:
I'm glad the advice helped.  Thanks for the update!

MartyG-RealSense, 2020-03-12:
@hojunson   Do you require further assistance or is it okay to close this case now, please?   Thanks!

hojunson, 2020-03-12:
No. Thanks

MartyG-RealSense, 2020-03-12:
Ok, thanks so much!

closed 2020-03-12 by MartyG-RealSense