realsenseai/librealsense #5987: pyrealsense2 RuntimeError: No device connected.

CXYCarson, 2020-03-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | BlasterX Senz3D | 
| Firmware Version                   |  | 
| Operating System & Version | Win10 | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC  |
| SDK Version                            |  2.33.1                           | 
| Language                            |  python                           | 
| Segment			|   | 

### Issue Description
I was using pyrealsense2 and I used the following code to read from a BlasterX Senz3D camera:

`import pyrealsense2 as rs`
`pipe = rs.pipeline()`
`profile = pipe.start()`

But after my computer was dead and I forced a restart while the camera was still plugged in, I got:

`RuntimeError: No device connected. ` with the same code and dependencies

I have already tried to reinstall different versions of pyrealsense2 but that didn't help. I have also tested on another computer with the same code and it was working fine on the other computer. Moreover, I tried to display image inputs using Intel RealSense Viewer but the software couldn't detect any inputs, and I also tried that on the other computer and it also worked fine. In the meanwhile, my computer can detect the camera in the device manager, and it can also display the inputs in the built-in camera App.

I'm thinking that something in my computer is broken due to the forced restart and SDK was not able to detect the camera. But I'm not sure about this since my Win10 computer is still able to recognize the device. Could someone please help me diagnose this issue?

MartyG-RealSense, 2020-03-06:
The first step that I would recommend is to try updating the graphics drivers on your computer.  RealSense Viewer requires OpenGL graphics in order to work correctly.  If the Viewer worked on your computer before the crash then it is likely that the graphics GPU component in your PC is OpenGL-compatible.

CXYCarson, 2020-03-09:
I just updated my Nvidia graphics driver and also checked OpenGL is installed in my PC. But I'm still getting `RuntimeError: No device connected. `
My OpenGL version is 4.6 with Nvidia driver 442.19 on a RTX2080. Any ideas on how to solve this?

CXYCarson, 2020-03-09:
> The first step that I would recommend is to try updating the graphics drivers on your computer. RealSense Viewer requires OpenGL graphics in order to work correctly. If the Viewer worked on your [...]

I just updated my Nvidia graphics driver and also checked OpenGL is installed in my PC. But I'm still getting RuntimeError: No device connected.
My OpenGL version is 4.6 with Nvidia driver 442.19 on a RTX2080. Any ideas on how to solve this?

MartyG-RealSense, 2020-03-09:
If you are using Windows 10, could you provide an image please of your Device Manager so that I can see if the 'Depth' and 'RGB' camera drivers are installed okay on that particular computer?  These are typically found in the "Cameras" section.  On some PCs they may also appear in the "Imaging Devices" section of the Device Manager.

[image]

CXYCarson, 2020-03-09:
> If you are using Windows 10, could you provide an image please of your Device Manager so that I can see if the 'Depth' and 'RGB' camera drivers are installed okay on that particular computer? These [...]

Screenshot's here, it seems that it's able to detect the camera.
[image]

MartyG-RealSense, 2020-03-09:
That's a good sign that the drivers are installed.  How did you install them please?   Did you use a Depth Camera Manager (DCM) program, or did you just insert the USB cable into the computer and Windows automatically downloaded and installed them, please?

CXYCarson, 2020-03-09:
> That's a good sign that the drivers are installed. How did you install them please? Did you use a Depth Camera Manager (DCM) program, or did you just insert the USB cable into the computer and Windows automatically downloaded and installed them, please?

I'm using a BlasterX Senz3D camera, so I downloaded the BlasterX Acoustic Engine Pro Software Pack for Windows from here:   [https://support.creative.com/Products/ProductDetails.aspx?catID=1170&catName=Peripherals&subCatID=1171&prodID=22791&prodName=BlasterX%20Senz3D&_ga=2.163539001.440221987.1583765826-978425700.1582558152](url)

MartyG-RealSense, 2020-03-09:
The BlasterX Senz3D by Creative is virtually identical to the official Intel SR300, as Creative manufactured the official SR300 for Intel.  You could try right-clicking on each of the drivers (RGB and Depth) and selecting 'Uninstall' from the menu.  Then when both are removed, unplug the camera, wait a couple of seconds and re-insert it.  It should hopefully download the drivers over the internet, and have them appear as modern "SR305" drivers.

CXYCarson, 2020-03-09:
> The BlasterX Senz3D by Creative is virtually identical to the official Intel SR300, as Creative manufactured the official SR300 for Intel. You could try right-clicking on each of the drivers (RGB [...]

It didn't download the drivers over the internet, and I also tried to select 'Update driver' but it didn't start downloading either.

MartyG-RealSense, 2020-03-09:
Assuming you were able to restore the drivers after the uninstall, did you get them back using Creative's driver install software please?

CXYCarson, 2020-03-09:
> Assuming you were able to restore the drivers after the uninstall, did you get them back using Creative's driver install software please?

I got 'SR300 RGB' back on the list but depth is gone.

MartyG-RealSense, 2020-03-09:
Does your Device Manager have an "Imaging Devices" section in it now when the camera is plugged in?  If it does, is the depth driver under that?

CXYCarson, 2020-03-09:
> Does your Device Manager have an "Imaging Devices" section in it now when the camera is plugged in? If it does, is the depth driver under that?

You are right, 'SR300 depth' is under 'Imaging Devices'

MartyG-RealSense, 2020-03-09:
Ok, this is normal for the SR300.  The drivers can end up on different places - either all in one place or divided between Cameras and Imaging Devices.

Could you now try the camera with the RealSense Viewer to see if it works?

CXYCarson, 2020-03-09:
> Ok, this is normal for the SR300. The drivers can end up on different places - either all in one place or divided between Cameras and Imaging Devices.
> 
> Could you now try the camera with the RealSense Viewer to see if it works?

RealSense Viewer doesn't detect any devices, the code still doesn't work either.

MartyG-RealSense, 2020-03-09:
Apologies for the length of this process.  Diagnosing SR300 cameras can take a while!

Forgive asking the question, but do you have the camera plugged in a USB 3 port.  Unlike the later 400 Series RealSense camras that can work with USB 2.0, SR300 requires USB 3.

CXYCarson, 2020-03-09:
> Apologies for the length of this process. Diagnosing SR300 cameras can take a while!
> 
> Forgive asking the question, but do you have the camera plugged in a USB 3 port. Unlike the later 400 Series RealSense camras that can work with USB 2.0, SR300 requires USB 3.

No problem, I am aware that it will take some time to debug as I already looked through other issues lol and I appreciate your help greatly.
I have the camera plugged in a USB 3 port and I also tried all USB 3 ports on my pc but the code was still unable to detect any devices.

MartyG-RealSense, 2020-03-09:
thanks for your patience!

Next, please go the text box at the bottom of your Windows screen (the one next to the corner button where you shut down) and type **privacy** into it to find the Privacy Settings window.

Open the window and scroll down the options side-panel until you get to **Cameras**.   Left-click on it and please check that the '**Allow Apps to access your camera** is set to 'On'.

[image]

CXYCarson, 2020-03-09:
> Thanks for your patience!
> 
> Next, please go the text box at the bottom of your Windows screen (the one next to the corner button where you shut down) and type **privacy** into it to find the Privacy Settings window.
> 
> Open the window and scroll down the options side-panel until you get to **Cameras** and left-click on it to bring up the camera privacy settings. Please check that '**Allow Apps to access your camera** is set to 'On'.
> 
> [image]

Verified it's 'ON'. I can get inputs through built-in 'Camera' App but still getting the same error.

MartyG-RealSense, 2020-03-09:
That makes sense.  When an SR300 is unable to function as a RealSense depth camera, it can still be used as an ordinary RGB webcam.

The fact that your other computer can detect the camera and work with it fine does seem to confirm that it is an issue local to your computer and not a camera hardware problem.  

At this point we have to start considering whether it is a USB power problem - that something has changed on the state of the USB ports that means that they are providing insufficiently stable power for the camera to work.   Sometimes this can be fixed by connecting the camera to a mains electricity powered USB hub instead of plugging it directly into the computer.   

Do you have a mains powered hub available to try please?

If you do not, but you do have a USB chargable device such as a tablet, a workaround for free can be to attach the tablet to the computer via its charging cable whilst it is switched off so it begins charging.  Then attach the camera to another port and see if it can then be detected.

CXYCarson, 2020-03-09:
> That makes sense. When an SR300 is unable to function as a RealSense depth camera, it can still be used as an ordinary RGB webcam.
>
> The fact that your other computer can detect the camera and work [...]

I don't have a powered hub so I tried the other way. I attached an iphone to the computer while it wasn't switched off and it was charging. Then I attached the camera to another port and device manager was still able to detect both RGB and depth.

MartyG-RealSense, 2020-03-09:
Did you test whether the SR300 appeared in the RealSense Viewer when the tablet + camera combination was used please?

CXYCarson, 2020-03-09:
> Did you test whether the SR300 appeared in the RealSense Viewer when the tablet + camera combination was used please?

SR300 didn't appear in the RealSense Viewer, and the code didn't work either when the combination was used.

MartyG-RealSense, 2020-03-09:
Sometimes SR300s are not detected because a third driver called the **Virtual Driver** is not present.  A complication though is that sometimes the SR300 works if the Virtual Driver is missing and sometimes it does not work.  Sometimes SR300 owners have given up without ever finding out why their camera stopped working, because there were so many different possibilities.

There comes a point where a user has done everything reasonably possible to get their SR300 to work and then they have to give it up before they end up spending weeks on it with no guarantee of success.

The last thing worth trying may be to use the **Windows Troubleshooter** tool to see if it can find a hardware problem and  auto-correct it.

https://forums.intel.com/s/question/0D50P00004McnzkSAB/i-have-a-intel-sr300-but-i-cannot-install-the-realsense-camera-sr300-virtual-driver

CXYCarson, 2020-03-09:
> Sometimes SR300s are not detected because a third driver called the **Virtual Driver** is not present. A complication though is that sometimes the SR300 works if the Virtual Driver is missing and [...]

That didn't work. Do you know any other reasons that may cause this error? Could it be some dependencies incompatible or not installed?

MartyG-RealSense, 2020-03-09:
I remembered that a couple of month ago, a RealSense user shared a detailed process on the Intel support RealSense forum that they say fixed their SR300 on a modern Windows version.  It is one more avenue of investigation.

https://support.intelrealsense.com/hc/en-us/community/posts/360038275074-Managed-SR300-to-work-on-latest-Windows-10-18363-535-

CXYCarson, 2020-03-09:
> I remembered that a couple of month ago, a RealSense user shared a detailed process on the Intel support RealSense forum that they say fixed their SR300 on a modern Windows version. It is one more [...]

Just followed all steps but the camera is still not working. Are there any other potential reasons?

MartyG-RealSense, 2020-03-09:
Sometimes an SR300 has stopped working when Windows updates to a new version (build), like when Windows auto-downloads a new version and then installs it when the computer is restarted.  It may be worth seeing which Windows build version the computer the camera works with has compared to the non working computer's build.

You can find the build by putting 'system info' in the bottom text box of Windows to find the System Information page.

MartyG-RealSense, 2020-03-19:
@CXYCarson   Do you require further assistance please or can this case be closed?  Thanks!

closed 2020-04-12 by MartyG-RealSense

TuanTran1412, 2021-03-24:
in Linux to fix

MartyG-RealSense, 2021-03-24:
@TuanTran1412   Please start a new case using the **New Issue** button on the front page of the forum at the link below.  Thanks!

https://github.com/IntelRealSense/librealsense/issues/