realsenseai/librealsense #5990: Announcement:  Highlighting a study on use of RealSense with Unreal Engine

MartyG-RealSense, 2020-03-07:
Hi everyone,

As highlighted by the official RealSense Twitter account, RealSense user **zuga** has authored a study on using RealSense with Unreal Engine.  

https://www.zugakousaku.com/archives/ue4_study/2165

I have auto-translated the article from Japanese into English and attached it to this discussion as a PDF.

[Introduction of RealSense-recommendation of futuristic drawing.pdf](https://github.com/IntelRealSense/librealsense/files/4301608/Introduction.of.RealSense-recommendation.of.futuristic.drawing.pdf)

closed 2020-03-09 by MartyG-RealSense

Sashi-Mee, 2020-10-06:
This is great ! But building fails and I don't know how to make it manually... would you know how to fix that ?

MartyG-RealSense, 2020-10-06:
Hi @Sashi-Mee   Did you follow the official setup instructions for the RealSense Unreal Engine wrapper in the lnk below, please?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unrealengine4#getting-started

Sashi-Mee, 2020-10-06:
Hi ! Thanks for your quick reply !
I managed to build the files, my .NET package was out of date !

So now I can open the unreal project, but the materials are just black & white checkers, and there's no point cloud.

I'm using the L515 , and it's working fine with the viewer (so the camera works )... Any idea ?

MartyG-RealSense, 2020-10-06:
Great news that you made progress!

I am not involved in L515 related cases as I am a 400 Series and SR30x specialist.  Could you create a new case on the Github please using the **New Issue** button on the front page of the RealSense GitHub and include L515 in the message title.  Then the appropriate RealSense team member can answer your question,.   Thanks!

https://github.com/IntelRealSense/librealsense/issues/