realsenseai/librealsense #5993: Disposing of Accel+Gyro Streams in Unity wrapper takes 20 seconds

BlueprintBen, 2020-03-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC |
| SDK Version                            |  2.31.0 | 
| Language                            |  Unity (2019.2.21)  | 

### Issue Description
This issue arises if you're trying to read back IMU values from a D435i device by including the Accel and/or Gyro streams in the profile for an RsDevice, and can be seen in-editor or in standalone builds.
There isn't an example scene provided that actually exemplifies reading IMU data but the attached screenshot shows the one change that you can make to an example scene to reproduce the issue. 

[image]

The addition of each stream adds 10 seconds to the rs2_delete_pipeline function call, which is a significant concern and basically bars us from using this functionality until the issue's resolved. 

I'd previously chimed in on [this thread](https://github.com/IntelRealSense/librealsense/issues/5807) thinking our issues were connected but I felt with my additional investigation it was worth opening it's own ticket. I'll try upgrading SDKs and Firmware Versions in case that has an impact and report back.

MartyG-RealSense, 2020-03-08:
Have you calibrated the IMU on your D435i please? The non-IMU sensors of the D435i are calibrated in the factory but its IMU is not and must be self-calibrated by the user in order to access it.  Not calibrating the IMU can cause errors when trying to access it.

For example, trying to use gyro or Accel from an uncalibrated D435i IMU in the RealSense Viewer may generate this error, which disappears when the IMU features are turned off.

**hwmon command 0x4f failed. Error type: No data to return (-21).**

If your IMU is uncalibrated then I wonder if there is a delay on exit because Unity is struggling to work out how to close the IMU streams.  I have seen Unity do similar prolonged delays with non-RealSense problematic elements when exiting from a test run back to the editor.

Intel have a white paper guide document on calibrating the D435i IMU.

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_Depth_D435i_IMU_Calib.pdf

BlueprintBen, 2020-03-08:
You're saying that our clients who want to use the D435i with our software will also have to install Python and perform this calibration in order for our software to read IMU data? This doesn't seem advisable nor what's currently implemented.

I haven't seen any explicit prompt to calibrate and definitely haven't run an IMU calibration script, but am able to read both Accel and Gyro data from my D435i in both the RealSense Viewer and my SDK app.

[image]

MartyG-RealSense, 2020-03-08:
The advice as far as I have known it was that self-calibration was a necessity to avoid errors being generated from IMU use.  I tested my own D435i's IMU just now (I do not usually use this function myself) and it worked fine like yours.  So something may have changed in the RealSense SDK to avoid those problems in the long period since the D435i's launch.  I do apologise.

BlueprintBen, 2020-03-08:
No worries; I appreciate the responsiveness. One of the reasons we like the IMU functionality is for its ability to circumvent some need for user intervention in our software, so the extra instructions threw me for a loop! 
I did expect that the direct IMU access was probably getting less eyes on it than say the SLAM features but it would be great to have an example scene in Unity where it's working as intended as a base test.

MartyG-RealSense, 2020-03-08:
The RealSense developers @matkatz or @ogoshen may be best placed to answer this question, since they develop for the Unity wrapper. 

On the supplied image that the CPU does seem to indeed be getting caught up on disabling RsDevice, specifically in its **rs2_delete_pipeline()** function.

I couldn't find anything to use as a sample to test the D435i IMU either, unfortunately.

It has been stated that the RealSense developers are not producing any further tracking sample programs for the T265.   I do not have knowledge if that same policy about tracking related samples is being applied to the D435i IMU.

BlueprintBen, 2020-03-09:
Understood. A couple additional details I can note:
- upgrading the device firmware to latest didn't help the situation
- maybe 1 in 10 test runs actually completes within 1-2 seconds, getting my hopes up, but the rest are still closer to that 10/20 seconds.

Updating our version of the SDK to latest and/or testing on that is a bit more work from us so I'll wait the day or two to see if anyone else known the path to a solution. Given the reproducibility I imagine sourcing the cause to be fairly straightforward if you have the full stack at hand. 

Cheers

MartyG-RealSense, 2020-03-09:
Re-reading your case from the start, the impression I get is that the **BackgroundSegmentation** sample did not originally include an IMU-accessing element (and so was not designed for its inclusion) but you added it yourself because there was not a pre-made Unity sample program for the D435's IMU.  Is that correct please?

I managed to track down an example of a RealSense user who used the D435i and IMU with Unity to create a program.  It is not in English and so will need to be put through a browser auto-translater or Google Translate, but you can get an understanding of its setup through the images.

https://qiita.com/NWLab/items/69f1a9f8102096a3d451

BlueprintBen, 2020-03-09:
Yes your first paragraph is correct; I made that modification to the provided example in order to illustrate an example 'project' that reproduces the issue consistently. 

The source code at your link does mirror our own for the most part. I was following a different article/github project which walked through the setup to read the Accel/Gyro streams in Unity, which as I mentioned did get us to the point of functionality, but this Dispose issue is the only remaining hurdle. I can't seem to resurface that webpage (of course

MartyG-RealSense, 2020-03-09:
Thanks so much.   I'm glad you were able to find some use in that article.

I do not currently have the RealSense Unity programming knowledge to provide an analysis of your code, unfortunately.  I will re-tag @matkatz and @ogoshen in the hope that one of them will be able to provide advice.  I will add @ev-mp too as I believe they have had some dev experience with the Unity wrapper.

ogoshen, 2020-03-11:
These kind of delays seem like frame timeouts, those are set up to be 5 seconds by default.
Sensors will always wait for their frames & callbacks before closing.

Double check disposal of frame resources, looks like you're missing one in `frame.As<MotionFrame>()`.

MartyG-RealSense, 2020-03-11:
@ogoshen  Thanks so much for taking the time to help on this case!

BlueprintBen, 2020-03-11:
Thanks for jumping on @ogoshen. Good catch on the dispose, but it unfortunately didn't impact the duration of the hitch. 
[image]

BlueprintBen, 2020-03-12:
To summarize:

This is a timing bug that manifests when closing a device (hardware is D435i) that was streaming Accel or Gyro streams in the Unity SDK. This bug causes any call to **rs2_delete_pipeline()** to block for almost exactly 10 seconds if one of the above streams is requested, and **20 seconds** if both were requested. Any modifications to stream parameters require calling rs2_delete_pipeline(), meaning this issue in its current state basically forces us to choose between having responsive UI and supporting the D435i IMU :( 

**Testbed (in Unity Editor):**
1. New Scene
2. Add GameObject
3. Add RsDevice (can be set to either Unity Thread or Multithread)
4. Set requested streams based on Tests
5. Press Play button and wait for initialization to complete (D435i should be playing)
6. Press Play button (to stop playing). You could either disable the GameObject or Component instead.
7. Observe amount of time taken for Unity Editor to become responsive again.

**Tests:**
- With Color, Infrared, or Depth streams (or a combination thereof) requested, closing takes **<2 seconds**. 
- With Accel and Gyro streams requested, closing takes **~20 seconds**. 

Doesn't matter if I've registered a listener or not. Hope that inspires some thought; the 10 seconds per stream did feel like it would spark something.

MartyG-RealSense, 2020-03-22:
At this point I will add a "Bug" label to the case so that it can be tracked as a possible bug by the RealSense team and investigated.

closed 2020-05-29 by MartyG-RealSense

Carlo-Gezapp, 2020-10-12:
Any new info? Still same problem with T265 in unity! Thanks

MartyG-RealSense, 2020-10-12:
Hi @Carlo-Gezapp  I am not responsible for T265 related issueas as another member of the RealSense team handles that camera.  I recommend creating a new case using the **New Issue** button on the front page of the RealSense GitHub forum and include the word **T265** in the message title.   Thanks!

https://github.com/IntelRealSense/librealsense/issues

Carlo-Gezapp, 2020-10-12:
Thanks. https://github.com/IntelRealSense/librealsense/issues/5807 already exist

MartyG-RealSense, 2021-06-24:
Hi everyone!  After a recent report of a 20-second delay when closing down an IMU-enabled session in the Unity wrapper, a official feature request has been filed with Intel to consider the issue (please see the link below).  Thanks very much for your helpful past reports!

https://github.com/IntelRealSense/librealsense/issues/9278

Vangos, 2021-11-17:
Version 2.49 has resolved the issue for me.