realsenseai/librealsense #6001: rs-convert: .png extraction premature exit

alberttissot, 2020-03-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | (05.11.06.250) | 
| Operating System & Version |   {Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.0 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

Hello, 

I have been having issues with the extraction of images from bag files with the _rs-convert_  tool.
The extraction stops at a random moment in the execution, but without printing a particular error message, e.g: 

_rs-convert -i /media/user/Elements/path/20191217_132804.bag -p /mnt/disk/path/ -c
2%
PNG converter
	23 Color frame(s) processed_
(this bag is supposed to output 1365 images according to _rosbag info_)

I have tried to use other scripts I have found on github when researching the issue, unsuccesfully.
When playing the bag file in realsense-viewer, the bags plays correctly from start to finish, despite some occasional lag in the video stream

Has this issue been solved already ? Is there a way to extract the remaining images ?

Thank you

MartyG-RealSense, 2020-03-09:
Some users pref an alternative bag to image converter called **rs_bag2image**, which was created by a RealSense community member called **UnaNacyOwen**.  Compared to **rs-convert**, it does not tend to have reports of problems.

https://github.com/UnaNancyOwen/rs_bag2image

alberttissot, 2020-03-12:
Hello, 

I tried that code already and it gets stuck the same way, only extracting a small fraction of the images, without printing any error message or exiting the execution (exiting has to be done manually)

MartyG-RealSense, 2020-03-12:
Having to sometimes manually exit execution of **rs-convert** is a known problem that has occurred for other users of it in the past.  A workaround bug-fix was incorporated into the SDK in February 2019, though it is possible that there are still some cases that slip past the fix.

https://github.com/IntelRealSense/librealsense/issues/3297

It is normal to have some missing frames during conversion.  I have never seen a frame conversion amount as low as 23 out of a total frame amount of over a thousand though, so it definitely appears that something is going wrong.  It looks as though it is crashing almost as soon as it begins converting.

Are you writing the images to a path that allows writing, please, and not a path that is read-only.

alberttissot, 2020-03-12:
Hello, 

The same thing happens when running **rs-convert** with the **-d** option, the output path also allows reading and writing. 
When comparing the output of **rosbag info** of the "broken" bag file to a "sane" bag file, I noticed a different line on the "broken" file:
**/device_0/sensor_0/notification/Hardware Error    5 msgs    realsense_msgs/Notification**
I can't acccess the message though, do you know how to ? 

Thank you

MartyG-RealSense, 2020-03-12:
I do not know about the messages, sorry.

Did you create the bag file with the RealSense SDK or with ROS, please?  If it was not recorded from the RealSense SDK, that could be a source of incompatibility.

https://github.com/IntelRealSense/librealsense/issues/3098#issuecomment-458029589

MartyG-RealSense, 2020-03-22:
This case will be closed after 7 days from the date of writing this if there are no further responses.   Thanks!

alberttissot, 2020-03-23:
Sorry about the late reply !
I have recorded the .bag files with RealSense Viewer. This problem occurs with 2 .bag recorded in the same session (the third one of that session works fine), the other .bag files I recorded during other sessions do not show problems
The available extraction tools do not extract the images but the RealSense Viewer seems to be able to play the entire bag file somehow

MartyG-RealSense, 2020-03-23:
Bag files may report problems such as being 'unindexed' if the recording was not closed properly.  

A RealSense community member recommended a ROS bag fixing tool called **rosbag reindex**, though I'm not sure if it will work with bags recorded in the RealSense SDK.

http://wiki.ros.org/rosbag/Commandline#reindex

alberttissot, 2020-03-23:
I tried **rosbag reindex** already and it doesn't have any effect on the bag files, the extraction still stop around 2%

MartyG-RealSense, 2020-03-23:
On another discussion about bags yesterday, the alternative option of converting a bag into a conventional video file using screen capture software was discussed.  Examples of such programs mentioned were:

Examples of such programs that have a Linux version are :

OBS (Open Broadcaster Software).
https://obsproject.com/

**ffmpeg**
https://www.ffmpeg.org/download.html

Once the bag was played in full-screen whilst video capture software was running, the resultant video file could have a video to png extractor program used on it to get the PNG image-set.  google for 'video to png extractor' for ways to do this.

MartyG-RealSense, 2020-04-12:
Closed due to no further comments being received.

closed 2020-04-12 by MartyG-RealSense