realsenseai/librealsense #6002: Dynamic calibrator V2.6.8.0

scizors, 2020-03-09:
Hello,
 I tried to calibrate the RGB camera using D415 
first time
It was successful and it created a CalibrationTable.xml inside bin the values after using  Dynamic calibrator V2.6.8.0 are 
    <param name = "ResolutionRGB">
        <value>1920</value>
        <value>1080</value>
    </param>
    <param name = "FocalLengthRGB">
        <value>1388.65</value>
        <value>1388.21</value>
    </param>
    <param name = "PrincipalPointRGB">
        <value>945.008</value>
        <value>564.178</value>
    </param>

second,third and so on time why?
It stopped in the app Dynamic calibrator V2.6.8.0
in step 15 it says start scale calibration

MartyG-RealSense, 2020-03-09:
The instructions for depth scale calibration are in section 2.2.1 of the Dynamic Calibrator user guide.

https://dev.intelrealsense.com/docs/intel-realsensetm-d400-series-calibration-tools-user-guide

To quote the instructions:

**************
1.  Take the images from L and R camera, including the depth stream (in real-time)

2.  Detect the target on the smartphone in both images

3.  Similarly to target-less calibration, user moves the phone so that it covered most of the image, repeating steps 1-2

4.  Once done, user just keeps taking images by positioning the phone anywhere in the image but must move the phone every time

5.  **After taking 15 images** in step 4, the process is complete

6.  The process checks for rectification error (absolute Y difference) but also compares the measured pattern size with ground truth

scizors, 2020-03-11:
Hello,
 Thanks for the reply I need to calibrate RGB camera for align intrinsics for 12801*720 is it possible 

as you see my previous post the output xml is
param name = "ResolutionRGB"
value 1920value
value>1080value
 
but i am intrested in 1280*720

MartyG-RealSense, 2020-03-11:
The appendix of the Dynamic Calibrator instructions says about phone screen calibration: "The target image is dynamically generated on the fly at runtime.  The image will be exact match to screen resolution".

So I wonder if the issue is related to the screen settings on your phone target.

scizors, 2020-03-11:
how can I fix this please provide the solution for this problem

MartyG-RealSense, 2020-03-11:
Re-reading the information about the phone screen calibration, I think the app is just scaling the stripe image to the size of the phone's screen and it doesn't actually mean anything in terms of calibrating the camera.  There does not appear to be a way to change the image displayed on the phone, and there is probably no need to do so.

The information that you got:

value 1920value
value>1080value

May just be referring to the fact that this is the maximum resolution (the 'Active Pixels') of the RGB sensor of the 400 Series cameras.

scizors, 2020-03-11:
sorry i am interested in intrinsic camera parameters for resolution of 1280*720 like focal length(x,y), principal point(x,y) and distortion parameter  using dynamic camera calibration app

MartyG-RealSense, 2020-03-11:
The free public version of the Dynamic Calibrator program only provides for calibration of extrinsic parameters.  

[image]

There is an OEM version of the Calibrator that can calibrate both intrinsics and extrinsics, but it would be hard to obtain unless you had a Non Disclosure Agreement (NDA) with Intel or purchased the $1500 USD OEM calibrator target board in the RealSense Store.

MartyG-RealSense, 2020-03-22:
This case will be closed after 7 days from the date of writing this if there are no further responses.   Thanks!

closed 2020-03-28 by MartyG-RealSense