realsenseai/librealsense #6003: Possible ways of aligning depth and RGB streams after being sent through WiFi

johnny-wang16, 2020-03-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I have a D435i connected to an on board computer(intel NUC) and I am able to send RGB and depth streams over WiFi but they have different latencies when arriving the host computer. I know the rs-align example demonstrates aligning depth and rgb streams but the two streams are no longer realsense objects at the host computer.  Is there maybe a constructor that converts the realtime streams  to some realsense objects so i can utilize the rs-align example to align both streams?

MartyG-RealSense, 2020-03-09:
There was a case in the past week of a RealSense user whose project needed real-time communication between an Android tablet and a desktop computer.  I suggested a remote computer control application to take full control of the target computer wirelessly from the tablet screen.

https://github.com/IntelRealSense/librealsense/issues/5971#issuecomment-596127497

The Chrome Remote Desktop application I suggested can also use a computer to take control of another computer.  So I wonder if the host computer could run the Remote Desktop application and take real-time control of the NUC by an operator sitting at the desktop.  That way, you would be accessing the original unconverted camera data instead of sending the streams over wi fi and breaking the alignment.

johnny-wang16, 2020-03-10:
Thank you @MartyG-RealSense for your response. I think my intended system setup is a bit different from that user. My project is to send depth and rgb streams wirelessly from the client side (D435i + intel NUC) to the host computer and display it in Unity 3D (installed on host computer). Thus, I am not sure how taking control over the target computer would help me in this case. 

The reason I asked this question is because I think there should be a way of doing it. In the end, all these data streams and frames are just bytes. I am guessing that the reason rs-align works is because there are timestamps encoded in each frame of both rgb and depth streams and under the hood, rs-align just matching those timestamps. If that's the case, aligning both streams on the host computer should be straightforward.

MartyG-RealSense, 2020-03-10:
Intel did a multicam demonstration in January 2018 where they captured a raw point cloud with a camera on one computer and sent the data to another computer along a network for final processing with the help of Unity.

https://www.intelrealsense.com/intel-realsense-volumetric-capture/

So it should certainly be possible to capture raw data and send it to another computer and do the alignment in Unity at the destination instead of before transmission.  Is it a project requirement please that alignment is done at the host end and not at the destination?

RealSense transmits large volumes of data at higher resolutions, which is why cabled transmission is usually more practical than Wi-Fi.

johnny-wang16, 2020-03-12:
Alignment can happen anywhere and for my purpose,  it makes more sense that the alignment happens after transmission. Is this project you mentioned about open source? If so, do you mind including the link of it? Thank you very much for your help

MartyG-RealSense, 2020-03-12:
Intel didn't make that particular volumetric capture project open source.  They did later present a long seminar on similar techniques though and publish it on YouTube. 

https://www.youtube.com/watch?v=VSHDyUXSNqY

It is probably going to be easier to go with your planned approach of one camera, one wi-fi connection.   Though wired connections are best, I have previously looked at RealSense wi-fi setups that use flight drone tech to accomplish the streaming.

https://github.com/IntelRealSense/librealsense/issues/4819#issuecomment-529093164

MartyG-RealSense, 2020-03-22:
This case will be closed after 7 days from the date of writing this if there are no further responses.   Thanks!

closed 2020-03-28 by MartyG-RealSense