realsenseai/librealsense #6005: Edge Map Preset

rupak-d, 2020-03-09:
| Camera Model                         | {D400 } | 
| Firmware Version                    | v2.29.0 | 
| Operating System & Version   |  Linux (Ubuntu 16  | 
| Kernel Version (Linux Only)     | 4.15.0-88 | 
| Platform                                   | PC |
| SDK Version                            | 2.29 | 
| Language                                | C  | 
| Segment			          | Robot | 

### Issue Description
Got the edge json preset from here: [https://raw.githubusercontent.com/wiki/IntelRealSense/librealsense/d400_presets/EdgeMapD435.json](url)
Which are the values in the file responsible for getting the edges? 
Will this work with a camera without the RGB module?

MartyG-RealSense, 2020-03-09:
A little more information on the Edge Map preset is available in a comment by a RealSense team member.

https://github.com/IntelRealSense/librealsense/issues/5188#issuecomment-549886043

Looking at the last 4 lines of the json file, it only makes reference to defining a depth stream.

I would speculate with an educated guess, though am not certain, that **param-scanlineedgetaub** is responsible for edges in the json.  This is based on the variable having a much higher value in the 'Default' json file, which has clean edges as one of its characteristics.

rupak-d, 2020-03-09:
Got it to work with a depth camera by removing the "controls-color" parameters.
Also found that it's actually the **param-texturecountthresh** & **param-texturedifferencethresh** values that are responsible for the edges. Would help if I got some more info regarding these 2 parameters.

MartyG-RealSense, 2020-03-09:
I'm glad you found a solution that works for you!  And thanks for the edge information.

Jsons are largely based on the SDK's "Advanced Mode", which doesn't have much documentation.  This is because Advanced Mode parameters interact with each other in complex ways and so Intel chose to handle them automatically with machine learning algorithms instead of document them fully.  Users are still free to do trial and error experimentation with the settings though to see how it affects the image.

I am off-shift for the day now but if you would like further assistance, please leave a comment below and I will pick it up when I return in 7 hours time and be happy to continue the conversation.   Good luck!

rupak-d, 2020-03-10:
Can we get more information regarding how it's doing the edge finding?

MartyG-RealSense, 2020-03-10:
If you are seeking cleaner edges, using a "Spatial Edge-Preserving Filter" post-processing filter has more extensive documentation.  Would this meet your needs please?

https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#spatial-edge-preserving-filter

MartyG-RealSense, 2020-03-22:
This case will be closed after 7 days from the date of writing this if there are no further responses.   Thanks!

closed 2020-03-28 by MartyG-RealSense