realsenseai/librealsense #6008: Motion Module Failure

anandmohan4143, 2020-03-10:
| | |
|---|---|
|**librealsense**|2.33.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435I|
|**Serial Number**|919122070909|
|**Firmware Version**|05.12.03.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|924123050848|
|**Firmware Update Id**|924123050848|

Please provide a description of the problem
Motion Module Error notification from hardware while turning on the module in the viewer. And the gyroscope and accelerometer output looks to be interchanged. Please have a look at the problem. Looking forward for a solution. Thank you.
[image]
[image]
[image]

MartyG-RealSense, 2020-03-10:
I ran some tests with my own D435i on Windows in SDK version 2.33.1.  Whe you say the data is "interchanged", do you mean that the data display appears on one side in one picture and on the other side on the other picture?  If so, this is how the Viewer works.  The left and right sides (gyro and accel) display and hide their read-out depending on whether the mouse cursor is hovered over their panel.

When I repeated your process (start the motion module, stop it and then start it again), I got the same error box as you when re-starting after the stop.  On my Viewer, the IMU streams still kept working normally immediately after that though, so I wonder if it is a momentary glitch that corrects itself immediately after start-up.

anandmohan4143, 2020-03-10:
Actually, The gyroscopic measure is displayed in accele window and accelerometer measure is displayed in gyro window as shown in the first two images

MartyG-RealSense, 2020-03-10:
The values shown in the two windows are consistent with the ones that I got for each window in my own tests with SDK 2.33.1.  I checked against some reference pictures of the motion module of D435i on Google Images though and found that the values do indeed seem to be flipped.

Example from Google Images:

[image]

MartyG-RealSense, 2020-03-10:
I checked the values in your test and mine against reference images on Google Images of the motion module.  They do seem to be correct (Gyro has values near 0 and Accel has values between 0 and 9.8)

anandmohan4143, 2020-03-10:
But gyro should give consistent value as if we change the position of the camera. It shouldn't go back to 0 and accelerometer should give acceleration to gravity and goes to zero when it is not accelerating. But in our images, both are interchanged.

MartyG-RealSense, 2020-03-10:
The IMU documentation makes reference to how you may get non-zero angular velocities due to it not having internal calibration.

https://github.com/IntelRealSense/librealsense/blob/master/doc/d435i.md#imu-calibration

anandmohan4143, 2020-03-10:
Thank you for the documentation. I will haver a look at it. I turned off the motion module and turned it on after some time, but still getting the module error. Is it hardware or software issue you think?

MartyG-RealSense, 2020-03-10:
Given that the issue was replicated exactly on my own camera, it is likely an SDK software issue rather than a hardware fault.  I will label this discussion as "bug" so that the RealSense team can track and investigate it.

anandmohan4143, 2020-03-10:
Please let me know as fast as you can so that I can work on it without
much trouble. Thank you

MartyG-RealSense, 2020-03-10:
I am not part of the RealSense developer team and are part of the GitHub support team.  So I can highlight issues with tags for other team members to follow up on but cannot influence their development timetable, so I cannot provide a time estimate for a bug-fix if it is decided to be a bug.  Thanks for your patience..

justin-chau, 2020-03-19:
I am also experiencing issues with Motion Module Failure when running RealSense viewer on Ubuntu 18.04. The message appears, but the viewer still works fine. It does affect the ROS Rtabmap example in realsense-ros though. Occasionally, the description for the error reads: "Motion Module force pause". I am also running firmware 05.12.03.00 and librealsense 2.33.1.

Any help would be greatly appreciated!

MartyG-RealSense, 2020-03-19:
@justin-chau  I did extensive further research on the D435i motion module 'failure' and 'force pause' errors.  There does not seem to be a clear solution available at the moment, unfortunately.  Raspberry Pi 4 in particular seems to be vulnerable to having IMU problems.

https://github.com/IntelRealSense/librealsense/issues/4979#issuecomment-540456439

I added a "Bug" label to this discussion 9 days ago, meaning that the issue will be kept open and will be able to be tracked by the RealSense team.

justin-chau, 2020-03-19:
Thanks for letting me know. I may try the installation method you described if I keep having issues. If I do, I will update my findings here.

Thanks again!

MartyG-RealSense, 2020-05-26:
Case closed due to reported bug not being reported by other users recently.

closed 2020-05-26 by MartyG-RealSense

Andyshen555, 2020-06-02:
I am experiencing the Motion Module failure error on 2.34.0 and FW 05.12.03.00. Everything looks fine on the realsense viewer, but I receive 'Warning: Some linear velocity entries in parameter imu0_config are listed true, but an sensor_msgs/Imu contains no information about linear velocities' when launching with opensource_tracking.launch.

MartyG-RealSense, 2020-06-02:
@Andyshen555   I researched your problem extensively but could not find any information on the error message that you quoted.  Are you using the D435i SLAM guide please?

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

Andyshen555, 2020-06-02:
Yes I'm using that. It worked for us for over a year and my last successful usage was yesterday. I'm sure that I have not hit the camera or have done anything that can physically damage the camera. By the way, I'd like to report that my laptop (ubuntu 18.04) recognizes the IMU as it belongs to the computer and rotates my desktop when I rotate the camera. Could that be a problem?

MartyG-RealSense, 2020-06-02:
@Andyshen555   A physical knock would likely have not stopped the IMU from functioning if it had occurred.  And I have not heard of an IMU-equipped RealSense camera being able to rotate a computer desktop display when a RealSense application is not running.  It makes sense for computers to be able to detect gyroscopes though in order to rotate the screen on tablet computers.  I recommend turning off screen rotation in Ubuntu.

https://askubuntu.com/questions/1035209/how-to-turn-off-screen-rotation-in-ubuntu-18-04-lts

Do you still get the error messages if you unplug the camera and then put it back in after the messages occur?

Andyshen555, 2020-06-02:
For the first time I turn on IMU, it reports nothing, but if I switch it off and turn on again, I get the error every time. However, unplugging does nothing to the ROS error. I'll send the ubuntu log in the next comment

Andyshen555, 2020-06-02:
andy1234@andyThinkpad:~$ roslaunch realsense2_camera opensource_tracking.launch ... logging to /home/andy1234/.ros/log/78fa3306-a49f-11ea-b7f1-d46d6d655e19/roslaunch-andyThinkpad-6239.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://andyThinkpad:41673/

SUMMARY
========

CLEAR PARAMETERS
 * /ukf_se/

PARAMETERS
 * /ImuFilter/_publish_tf: False
 * /ImuFilter/_world_frame: enu
 * /ImuFilter/use_mag: False
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
[... 319 lines cut ...]
[ImuFilter-4] killing on exit
[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
 02/06 03:05:51,532 WARNING [140607137761024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/06 03:05:51,654 WARNING [140607137761024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/06 03:05:51,718 WARNING [140607137761024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/06 03:05:51,957 WARNING [140607137761024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/06 03:05:52,006 WARNING [140606155687680] (sync.cpp:171) Device destroyed
 02/06 03:05:52,006 WARNING [140606155687680] (sync.cpp:171) Device destroyed
 02/06 03:05:52,006 WARNING [140606155687680] (sync.cpp:171) Device destroyed
 02/06 03:05:52,006 WARNING [140606155687680] (sync.cpp:171) Device destroyed
 02/06 03:05:52,006 WARNING [140606155687680] (sync.cpp:171) Device destroyed
 02/06 03:05:52,006 WARNING [140606155687680] (sync.cpp:171) Device destroyed
 02/06 03:05:52,026 WARNING [140606129272576] (sync.cpp:171) Device destroyed
 02/06 03:05:52,028 WARNING [140606129272576] (sync.cpp:171) Device destroyed
 02/06 03:05:52,028 WARNING [140606129272576] (sync.cpp:171) Device destroyed
 02/06 03:05:52,029 WARNING [140606129272576] (sync.cpp:171) Device destroyed
corrupted size vs. prev_size
rtabmap: Saving database/long-term memory... (located at /home/andy1234/.ros/rtabmap.db)
rtabmap: Saving database/long-term memory...done! (located at /home/andy1234/.ros/rtabmap.db, 0 MB)
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

MartyG-RealSense, 2020-06-02:
@Andyshen555    It is difficult to tell from the log what is going on.   I note the error **Accel Sensitivity:hwmon command 0x4f failed. Error type: No data to return (-21)** though.  This is typically generated when the D435i's IMU has not been calibrated by the user.  Have you performed a self-calibration of the IMU since you purchased the D435i, please?

https://dev.intelrealsense.com/docs/depth-camera-d435i-imu-calibration

Andyshen555, 2020-06-02:
After calibration, realsense-viewer still gives ERROR and IMU still doesn't work on ROS. Also, some new problems arise. The accelerometer reads around 9.6 for all 6 axis before calibration and now it reads 9.4 camera facing down and 10.2 camera facing up, 9.5 cable facing down and 10.1 cable facing up. This is exactly what happened to my old camera which I used during SDK 2.17~2.20. I though the camera was broken and brought a new one (current one). What should I do now when I can't get a good calibration? I have recalibrated dozens of times lol

MartyG-RealSense, 2020-06-02:
Another user had similar values to yours after IMU calibration.  They were advised that after IMU calibration, an option called **Enable Motion Correction** should become available in the Viewer.  This corrects the IMU values so that they approach closer to the ideal of 9.8.

https://github.com/IntelRealSense/librealsense/issues/4751#issuecomment-527138458

Correction of acceleration values is necessary because the IMU component used in the D435i does not have built-in calibration.

Andyshen555, 2020-06-02:
Mine are almost identical no matter the Enable Motion Correction is on or off. They still read something close to 9.5, 10.1 though.

MartyG-RealSense, 2020-06-02:
If you still have the old "broken" D435i camera, I wonder if you could try swapping over their USB cables to eliminate the possibility that the ROS log errors are related to a bad USB cable.

Andyshen555, 2020-06-02:
Nothing changed doing a cable swap.

Andyshen555, 2020-06-02:
I got this when running the rs_camera.launch

02/06 15:04:17,549 WARNING [140687055988480] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ WARN] [1591124657.685268432]: Hardware Notification:RT IC2 Config error,1.59112e+12,Error,Hardware Error
[ERROR] [1591124657.685460059]: Performing Hardware Reset.

MartyG-RealSense, 2020-06-02:
Were all the tests with the current and previous camera carried out on the same computer please?  If yes, do you have the option of trying the camera on a different computer?

Andyshen555, 2020-06-02:
[ INFO] [1591125014.176830764]: insert Accel to Motion Module
 02/06 15:10:14,218 WARNING [139725679617792] (uvc-device.cpp:664) interrupt event received: 1, 3, 0, 7, 0, 10, 
[ INFO] [1591125014.471291072]: SELECTED BASE:Depth, 0
[ INFO] [1591125014.526446177]: RealSense Node Is Up!
 02/06 15:10:14,588 WARNING [139725889337088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/06 15:10:14,649 WARNING [139725889337088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/06 15:10:14,752 WARNING [139725889337088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/06 15:10:14,813 WARNING [139725889337088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/06 15:10:14,894 WARNING [139725889337088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ WARN] [1591125014.945867324]: Hardware Notification:Motion Module failure,1.59113e+12,Error,Hardware Error
 02/06 15:10:15,075 WARNING [139725889337088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/06 15:10:15,257 WARNING [139725889337088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

Andyshen555, 2020-06-02:
Currently I don't have any other Ubuntu machines available, but the IMU failure message showed up in the realsense viewer appeared on 2 win10 and a Ubuntu 18.04.

MartyG-RealSense, 2020-06-02:
Doronhi the RealSense ROS wrapper developer has said in the recent past that the **Resource temporarily unavailable** error message can be related to lack of system resources - the app can't keep up with the incoming flow of images.  Enabling CUDA support can help with this if the computer has an Nvidia graphics GPU.

Andyshen555, 2020-06-02:
1. Bandwidth / system resource could be a problem as you said, so I disabled RGB, **Resource temporarily unavailable** error message is gone.

2. The IMU calibration seems to have fixed itself somehow. The norms read in every axis goes back to 9.8, which is great.

3. The problem comes back to IMU not giving out info as it should ::
`[ WARN] [1591132803.683814185]: Warning: Some linear velocity entries in parameter imu0_config are listed true, but an sensor_msgs/Imu contains no information about linear velocities`
`[ WARN] [1591132809.978068997]: Could not obtain transform from camera_imu_optical_frame to camera_link. Error was "camera_link" passed to lookupTransform argument target_frame does not exist. `
`[ WARN] [1591132810.196124894]: /rtabmap/rgbd_odometry: Did not receive data since 5 seconds!`

MartyG-RealSense, 2020-06-03:
@Andyshen555  Great news that you made progress with the errors!

Regarding the remaining errors in ROS: there is no information about 2 of the 3 listed errors at all on the internet, so it is difficult to diagnose the problem from them.  The **imu0_config** that it refers to though seems to be related to robot localization.  Whilst **sensor_msgs/Imu** holds data from an IMU.

There is a lot more information available about the third error **/rtabmap/rgbd_odometry: Did not receive data since 5 seconds!**.  This may tie in to the availability of processing resources.  Others who have experienced this error suggest checking **rqt_graph** to see what is happening during processing.  If 30 FPS is being used then a publishing frequency of **30 hz** would be expected to be displayed on the rqt_graph.  Significantly less than 30 may suggest that there is a problem that is causing the errors.

http://wiki.ros.org/rqt_graph