realsenseai/librealsense #6009: v2.33.1 does not compile with -DBUILDEASYLOGGINGPP=OFF

andreaszwengINS, 2020-03-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version | Linux (Linux Mint 19.3) |
| SDK Version                            |  2.33.1                           | 
| Language                            |  C++                          | 

### Issue Description
Since 2.33.1, we cannot compile without easyloggingpp anymore. The reason is simple: In src/log.h, there is a new struct `log_message` in lines 11-17 defined within the preprocessor directive `#if BUILD_EASYLOGGINGPP`. In src/rs.cpp this struct is then used in lines 1122-1207 outside this directive, resulting in compiler errors.
Our quick fix was to wrap this part in the same preprocessor directive which compiles successfully. We did not investigate further if this causes any side effects, but so far everything works as expected.

Thank you,
Kind Regards,
Andreas

MartyG-RealSense, 2020-03-10:
Thanks so much for your report and details of your workaround!

I considered how best to label this discussion for the attention of the RealSense team.  easyloggingpp is already a part of the SDK via its 'Third Party Tools' folder, so 'feature request' is not appropriate as it already exists.  On the basis that easyloggingpp is an existing part of the SDK that worked before and now does not work for you without manual code changes according to your report, I will label it as "bug".  I hope that this approach works for you.

andreaszwengINS, 2020-03-10:
Sure, I just forgot the label - thanks!

7oud, 2020-06-30:
So if I wanna use easyloggingpp with the rs lib at the same time, is there any workaround now? The easyloggingpp is really useful and convenient.

MartyG-RealSense, 2020-06-30:
The listed known Issue in the SDK release notes regarding problems using **easyloggingpp** is named to be with the 2.33.1 SDK version specifically.  I don't know if anyone has tested easyloggingpp with newer SDK versions to see the problem it was isolated to just that one SDK.

7oud, 2020-06-30:
@MartyG-RealSense In the latest [v2.35.2 release note](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2352), #6009 is still in the known issues

MartyG-RealSense, 2020-06-30:
It looks like the "pull" to clear up issues is still an ongoing project.

https://github.com/IntelRealSense/librealsense/pull/6304

MartyG-RealSense, 2020-09-03:
Posting to keep this discussion open.

MartyG-RealSense, 2020-09-13:
It looks as though the pull fix https://github.com/IntelRealSense/librealsense/pull/6304 by @maloel was merged into the SDK 17 days ago at the time of writing this.   I will therefore close this case as the issue has been addressed by the pull.

closed 2020-09-13 by MartyG-RealSense