realsenseai/librealsense #6010: Android 9 Support

erikgallegos, 2020-03-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Android 9.0  | 
| Platform                                 | Galaxy S8 |
| SDK Version                            |  2.30                         | 
| Language                            |  kotlin                       | 
| Segment			| Smartphone    | 

### Issue Description
In the [releases page](https://github.com/IntelRealSense/librealsense/releases) I see that only Android 7 and 8 are listed under supported platforms, however I've been able to successfully use my application on phones with Android 9. 

I'm interested in knowing what it would take for Android 9 to be officially supported. Are there any blockers? What are the biggest hurdles?

MartyG-RealSense, 2020-03-10:
Like yourself, others have been able to use the Android wrapper with Android 9 devices.

One of the main developers of the Android wrapper - @matkatz - reported in mid 2019 that they had tried the wrapper with a couple of Android 9 devices.

https://github.com/IntelRealSense/librealsense/issues/4091#issuecomment-500792067

Later on In August 2019, a RealSense community member reported that they had used Android 9.

https://github.com/IntelRealSense/librealsense/issues/4753

As Android 9 support apparently is not an official status at this time though (and I could find no information explaining the reason for it), I will mark this discussion as a Feature Request.  This will help Intel to track it and also gauge user demand for Android 9 full support.

kafan1986, 2020-03-12:
@erikgallegos Based on my testing the Android 9 is as well supported as the official mentioned Android 7/8. I think android 10 introduced some changes and it broke certain things and intel team is working on that with Android team directly.

MartyG-RealSense, 2020-05-30:
Given that Android 9 is proven to work fine with Librealsense and Android 10 also works if you set a lower **targetSdkVersion,** of around '27', I will close this case now.

closed 2020-05-30 by MartyG-RealSense