realsenseai/librealsense #6012: When the camera suddenly disconnects and reconnects from the USB cable during normal frame transmission, is there any solution that can make the camera resume frame transmission quickly?

Dontla, 2020-03-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.11.15.00| 
| Operating System & Version |  Win10  | 
| Platform                                 | PC  |
| SDK Version                            | 2.29.0                     | 
| Language                            |  python3.6.8                           | 
| Segment			|  Robot                       | 

We use multiple D435 cameras to transmit in a scene with strong vibration. Sometimes the USB cable will suddenly disconnect and then connect again, but the frame transmission of the program will be disconnected. Is there a specific solution code for this scenario that can quickly restore the previous frame transmission after the camera is offline and then online?

MartyG-RealSense, 2020-03-11:
The stereo depth algorithms in the 400 Series cameras are very robust in regard to vibration, but there does come a point where disruptions like the ones you describe can occur if vibration becomes severe enough.  An example of past cases of this is when the camera is mounted to agricultural machines traveling over very rough ground.

If your camera is experiencing mobile motion that may jostle the cable in the USB port, USB cables with locking screws can be used to secure the cable firmly in the port.

https://github.com/IntelRealSense/librealsense/issues/3118#issuecomment-456392939

The main remedy for camera vibration though is to put vibration dampening pads somewhere near the camera (its mounting point, on the feet of a tripod, etc).  Google for 'vibration suppression pad' to see the products available.

In regard to resuming streaming automatically in Python after disconnection, I hope that the link below will be helpful.

https://github.com/IntelRealSense/librealsense/issues/5537

MartyG-RealSense, 2020-03-22:
This case will be closed after 7 days from the date of writing this if there are no further responses.   Thanks!

closed 2020-03-28 by MartyG-RealSense