realsenseai/librealsense #6013: Dynamic calibrator command line, no cameras detected

tispratik, 2020-03-11:
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.03.00| 
| Operating System & Version |   Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.15.0-88-generic                                         | 
| Platform                                 |   |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  python                         | 
| Segment			|  Robot                          | 

### Issue Description
When i enumerate the camera, both cameras show up, but when i run command line dynamic calibration tool, the tool says no devices connected.

```
$ /usr/bin/Intel.Realsense.DynamicCalibrator -cli
No device detected, please connect camera to calibrate.

$ rs-enumerate-devices -S
Device Name                   Serial Number       Firmware Version
Intel RealSense D415          840412060865        05.12.03.00
Intel RealSense D415          851112062361        05.12.03.00
Device info:
    Name                          : 	Intel RealSense D415
    Serial Number                 : 	840412060865
    Firmware Version              : 	05.12.03.00
    Recommended Firmware Version  : 	05.12.03.00
    Physical Port                 : 	2-3-2
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0AD3
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	846623022810
    Firmware Update Id            : 	846623022810

Device info:
    Name                          : 	Intel RealSense D415
    Serial Number                 : 	851112062361
    Firmware Version              : 	05.12.03.00
    Recommended Firmware Version  : 	05.12.03.00
    Physical Port                 : 	2-5-3
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0AD3
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	844213020021
    Firmware Update Id            : 	844213020021
```

MartyG-RealSense, 2020-03-11:
The command line instruction **-list** or **-show** should list the attached cameras when entered into the command line interface of the Dynamic Calibrator.  

You can launch the CLI version of the Calibrator with:

Intel.RealSense.DynamicCalibratorCLI

tispratik, 2020-03-11:
$ ls -l /usr/bin/Intel*
-rwxr-xr-x 1 root root 13716640 Nov 21  2018 /usr/bin/Intel.Realsense.CustomRW
-rwxr-xr-x 1 root root 33638336 Nov 21  2018 /usr/bin/Intel.Realsense.DynamicCalibrator

I only see the following two tools. The DynamicCalibrator has a -cli option.

tispratik, 2020-03-11:
$ /usr/bin/Intel.Realsense.DynamicCalibrator -cli
No device detected, please connect camera to calibrate.

$ /usr/bin/Intel.Realsense.CustomRW -g
CustomRW for Intel RealSense D400, Version: 2.6.8.0

Error: cannot find or open device with serial number

MartyG-RealSense, 2020-03-11:
Do you have both D415 cameras plugged in at the same time please?

Are the cameras still not detected if only one of them is plugged in?

tispratik, 2020-03-11:
Yes both are connected at the same time. I can check if single one works.

tispratik, 2020-03-11:
Even after unplugging one camera, same issue.

$ /usr/bin/Intel.Realsense.DynamicCalibrator -cli
No device detected, please connect camera to calibrate.

tispratik, 2020-03-11:
Unplugged both cameras, then plugged in one camera. Now i got further. But still an error.

$ /usr/bin/Intel.Realsense.DynamicCalibrator -cli -serial 851112062361

Serial Number: 851112062361
Device Name: Intel RealSense D415
Device PID:  0AD3
FW version:  05.12.03.00

Calibration Type: Targeted
Laser Power: On
Auto Exposure: On
AE SetPoint: Auto
Resolution: 1280x720x30
freeglut failed to open display ''

tispratik, 2020-03-11:
I am on an ssh connection to the machine which has realsense cameras connected. A field rep is helping me plug/unplug. I am using my mac to ssh.

MartyG-RealSense, 2020-03-11:
It has been a while since I have handled cases involving SSH.  Are you on a "headless" system please (operating without a monitor and controlled over a network connection).   The final line of the above log, "freeglut failed to open display", gives me that impression.

tispratik, 2020-03-11:
Yes, we're in a headless system. The remote machine is a 18.04 Ubuntu without UI. That is why I am interested in -cli mode of calibration.

MartyG-RealSense, 2020-03-11:
I found that there is an open case on another discussion about difficulties with using CLI mode.

https://github.com/IntelRealSense/librealsense/issues/3990

In that long discussion, another user had problems with CLI on Ubuntu 18.0 and said that they also had a friend who also could not get it to work in 18.04 but had no problems when using it in Windows.   That would not be an option for your Ubuntu based network setup though.  It may be worth joining that case so the RealSense support team member on that case can offer their perspective.

tispratik, 2020-03-14:
Thats correct. So, can I assume that the cli mode of calibration is not supported on Linux?

MartyG-RealSense, 2020-03-14:
In regard to your earlier comment about not being able to find the Intel.RealSense.DynamicCalibratorCLI executable:

_I only see the following two tools. The DynamicCalibrator has a -cli option._

I had another read of the Calibrator's manual and learned that whilst the Windows version has a separate Intel.RealSense.DynamicCalibratorCLI application, the Linux version only has a single Intel.RealSense.DynamicCalibrator application that supports both GUI and CLI interfaces.

So Linux is certainly meant to have a working CLI interface, according to the user manual, and your earlier-stated method of using **-cli** seems to be correct.

MartyG-RealSense, 2020-03-14:
Reading the manual, as far as I can tell it isn't required to use the -cli term as using a CLI term in the manual seems to be enough for it to be recognised as CLI without having to say so.

For example, to check for detection of all cameras, it suggests using:

**/usr/bin/Intel.Realsense.DynamicCalibrator –show**

tispratik, 2020-03-14:
Thanks for checking the manual Marty. True, the calibration does not require the -cli option, but I do get the same error.

MartyG-RealSense, 2020-03-14:
Are you able to check individual cameras by serial number?  The command for that is -sn.  For example:

**/usr/bin/Intel.Realsense.DynamicCalibrator –sn 722312071055**

tispratik, 2020-03-14:
No :(. I get the same error.

MartyG-RealSense, 2020-03-14:
I will seek technical advice and get back to you when I have a response.  I apologise for the wait in the meantime.

tispratik, 2020-03-14:
Thank you Marty. In general for all these technical utilities, mostly used by engineers or field engineers, it is good to throw exceptions and show detailed stack trace than give a friendly one liner like `No device detected, please connect camera to calibrate.`

MartyG-RealSense, 2020-03-14:
I had a thought just now that may be important, having re-read the case.  You are using an SSH connection to access a computer that the cameras are attached to.  So I assume that means that the cameras are not attached to the computer that you are issuing the Dynamic Calibrator CLI commands from.   this may be why the software is not detecting the cameras.

The OEM version of the Dynamic Calibrator software can connect remotely to cameras via Linux servers.  I don't think the free public version of the Dynamic Calibrator that you are using has theis functionality though.

tispratik, 2020-03-14:
Thanks for the information Marty. We have paid for the OEM version, problem is that the OEM target is very large for our conveyor belt which is only 2 ft wide while the OEM target is 6ft x 5 ft wide.

MartyG-RealSense, 2020-03-14:
It's great that you were able to buy the OEM software!

I'm not sure it strictly matters whether you are using the actual full size board, just that the chessboard pattern is sized proportionately.  The size of the board is designed for calibrating multiple cameras.  so you could try the OEM software with a chessboard image designed for a single camera.

Besides which, I would imagine that the calibration board would not need to actually be on the conveyor, but the cameras would be removed from the conveyor and put in front of the board on the rare occasions that they needed re-calibrating.

tispratik, 2020-03-15:
Thanks Marty. For most accuracy, I would prefer it calibrating on site. Can you guide on what kind of smaller sized chess board is pre-configured in the calibration tool?

MartyG-RealSense, 2020-03-15:
Intel have a printable print target for the dynamic calibration here:

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/print-target-fixed-width.pdf

tispratik, 2020-03-15:
Thanks Marty. I meant a target for the OEM calibration that is other than the one here https://store.intelrealsense.com/buy-intel-realsense-d400-cameras-calibration-target.html

MartyG-RealSense, 2020-03-15:
That is what I meant when I provided the print target - to try it with the OEM calibration process at the conveyor for a single camera instead of having the full size OEM board on the conveyor.

MartyG-RealSense, 2020-03-22:
This case will be closed after 7 days from the date of writing this if there are no further responses.   Thanks!

closed 2020-03-28 by MartyG-RealSense