realsenseai/librealsense #6014: unable to open "realsense-file.lib" when building matlab wrapper

FancyBrush, 2020-03-11:
OS: windows 10
device: SR300
matlab: 2018a
VS2019
1. Installed realsense SDK 2.0 following steps in readme.md
2. Installed realsense driver
    When I tried to build librealsense_mex  in  matlab wrapper, LINK error arose, which are "libmx.lib" and "realsense2.lib", and I solved them by adding their path in the project.
    Then, I build this again, another LINK error jumps out, saying, unable to open "realsense-file.lib". I find a folder named "realsense-file", but with no .lib file in it. 
    Where is "realsense-file.lib"? How can I generate it?
    
[image]

MartyG-RealSense, 2020-03-11:
If you are building the MATLAB wrapper from source, there is a Windows installer version of the wrapper that can be easier to use.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/matlab#windows-installer

FancyBrush, 2020-03-12:
Thank you for your reply! @MartyG-RealSense 
       This issue is succesfully settled, and procedures are as followed :
      1) It turns out that the command "cmake .. -D... -DBUILD_MATLAB_BINDINGS:BOOL=ON" in the README.md file from "F:\librealsense-master\wrappers\matlab" is actually "cmake .. -DBUILD_MATLAB_BINDINGS:BOOL=ON".
      2) After CMake the CMakelist.txt in "F:\librealsense-master\wrappers\matlab", a \matlab folder with RealsenseMatlabWrappers.sln  is generated.   Open RealsenseMatlabWrappers.sln generated by CMake, we can find librealsense_mex.sln, which is the solution that README.md want us to build, instead of the same name solution in "F:\librealsense-master\wrappers\matlab". I failed lots of times because of building librealsense_mex.sln in "F:\librealsense-master\wrappers\matlab". 
      BUILD **RealsenseMatlabWrappers.sln** GENERATED BY CMAKE !!!
      Finishing building librealsense_mex.sln, it generates a new \debug folder. 
      3) Then, we can copy +realsense from SDK 2.0 to the \debug folder, and set this \debug folder to MATLAB's working directory. 
      All done.

MartyG-RealSense, 2020-03-12:
Thanks so much for the update and sharing details of your method with the RealSense community.  I'm very pleased that you found a solution!   :)

Is it okay to close this case now if you have no more problems, please?

FancyBrush, 2020-03-12:
@MartyG-RealSense  
No more problems, thank you!

MartyG-RealSense, 2020-03-12:
Thanks!   I will close the case now.   :)

closed 2020-03-12 by MartyG-RealSense

ranggo467, 2022-01-01:
@FancyBrush can u help me to install this intelrealsense in matlab . im stuck already at this step 
1. Download the Git repository. Opt in the Matlab wrapper for LibRs with cmake .. -D... -DBUILD_MATLAB_BINDINGS:BOOL=ON when configuring CMake.

2. Build the librealsense_mex target.

MartyG-RealSense, 2022-01-01:
Hi @ranggo467  Are you attempting to install the MATLAB wrapper on Windows 10 or Linux, please?  The wrapper and its installation instructions are designed for Windows 10.

ranggo467, 2022-01-01:
@MartyG-RealSense i want install matlab wrapper in windows 10 but i dont understand this step... can u teach me from the beginning

MartyG-RealSense, 2022-01-01:
If you are using Windows 10 then it would be easier to install the full RealSense SDK on Windows using the auotmated installer on the **Releases** page.  This installer is set up to install the MATLAB wrapper as part of the overall SDK installation.

[image]

On the installer, you should only need to click the 'Next' button a few times to complete SDK installation without changing anything on the list of components that are installed by default.

The 'Releases' page is at the link below.  At the bottom of the page, there is an 'Assets' file list where the SDK installer program can be downloaded and run.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

[image]

ranggo467, 2022-01-01:
i already do this step.... but my problem is how to build this, i stuck at step 1 to 3

Building from Source
1. Download the Git repository. Opt in the Matlab wrapper for LibRs with cmake .. -D... -DBUILD_MATLAB_BINDINGS:BOOL=ON when configuring CMake.
2. Build the librealsense_mex target.
3. After compiling the project, set build/Debug or build/Release as your Matlab working directory. Alternatively copy the +realsense folder from there to a place where Matlab can find it.
4. Start exploring with Realsense devices by typing at the Matlab prompt:
realsense.depth_example - Capture and render Depth frame with a live device.
realsense.rosbag_example - Playback pre-recorded video and data streams stored in rosbag format.
realsense.pointcloud_example - Generate and visualize 3D point cloud from live stream.
realsense.depth_view_example - Capture and render Depth example based on apps.AppBase component.
realsense.capture_example - Capture and render multiple streams

MartyG-RealSense, 2022-01-01:
If you have used the SDK installer to install the full SDK and the MATLAB wrapper along with it then you do not have to do any of the instructions in the **Building from Source** section.  You can follow the method below instead.

1.  Right-click on the launch shortcut icon for the RealSense Viewer on your Windows desktop and select the **Open file location** menu option.  This will take you to the SDK folder and its **Tools** sub-directory.

2.  In the folder window's address bar, left-click on 'Intel RealSense SDK 2.0' to navigate to that directory.  You should see that there is a folder called **matlab** in the list of folders.  Open the matlab folder.

[image]

3.  In the matlab folder there is a folder called **+realsense**

[image]

Copy and paste this +realsense folder to a location on your computer where MATLAB can find it, or add it to MATLAB's path.

If you choose to define a path for MATLAB to look in, you can add the +realsense folder to the MATLAB search path using the guidance at https://github.com/IntelRealSense/librealsense/issues/2400

ranggo467, 2022-01-01:
thanks bro u help me alot i appreciate that

MartyG-RealSense, 2022-01-01:
You are very welcome, @ranggo467  :)

ranggo467, 2022-01-01:
Btw do u have another way to communicate more easier with u to ask any problem related with intelrealsense?

MartyG-RealSense, 2022-01-01:
You can create a new question on this support forum using the green **New Issue** button on the front page of the forum at the link below.

https://github.com/IntelRealSense/librealsense/issues

If you use ROS and have a ROS related question, then the RealSense ROS GitHub will be the best place to post the question.

https://github.com/IntelRealSense/realsense-ros/issues/

ranggo467, 2022-01-01:
bro i already try the every example in matlab but can i know how to use other function like depth_frame, depth_sensor, depth_stereo_sensor, disparity frame and other.

[image]

MartyG-RealSense, 2022-01-01:
Script code for the MATLAB wrapper is almost 1:1 with the SDK's C++ code, with some differences. A good way to research a particular function is to search using the keywords **librealsense matlab** and add on the end of the search instruction the information that you are looking for.  For example: librealsense matlab depth_sensor

https://github.com/IntelRealSense/librealsense/issues/3358#issuecomment-475489478 has an example of scripting for using **depth_sensor** in the MATLAB wrapper.

ranggo467, 2022-01-01:
Thank bro i will try and update to you later if have any problem

ranggo467, 2022-01-02:
@MartyG-RealSense 
bro do u know how to use this API...can u teach me from the start....where to put this code in matlab? 

https://github.com/IntelRealSense/librealsense/wiki/API-How-To#controlling-the-laser

MartyG-RealSense, 2022-01-03:
There are no tutorial lessons for the MATLAB wrapper, so the best learning method is to study how the examples that you have already tried work.  Usually when someone wants to know how to program a specific function in the wrapper, they post a question on this support forum and then I research it to see if someone else has posted an answer in the past (my own MATLAB programming knowledge is limited).  

In regard to your question about controlling the laser, the https://github.com/IntelRealSense/librealsense/issues/3358#issuecomment-475489478 link provided earlier had MATLAB wrapper code for enabling / disabling the IR emitter.  Bear in mind that there is an apparent typing mistake in the script in that link where the section highlighted below should be **option.emitter_enabled** (a full stop inserted after 'option').

[image]

When looking at a piece of code, you can get hints about where to insert it in your own script by looking for where the **pipe.start()** line is.

If the code is below the pipe.start() line in the example then also put the code below pipe.start in your own script.

[image]

Similarly, if the code is above the pipe.start() like then also put it above pipe.start() in your own script.

ranggo467, 2022-01-04:
thanks bro, can i combine the depth_view_example code with align code together in one coding? because i want get the live view and also to align image at the same time?
or do you have other solution?

MartyG-RealSense, 2022-01-04:
A RealSense user shared their MATLAB wrapper script for depth-color alignment in https://github.com/IntelRealSense/librealsense/issues/9773#issuecomment-925553900

Another RealSense user shared their own MATLAB approach to alignment and then generating a pointcloud in https://github.com/IntelRealSense/librealsense/issues/6763#issuecomment-657217234

ranggo467, 2022-01-04:
thanks again bro, sorry if i ask to many question, pointcloud function is to store the 3d data or to show 3d data... do you have example how to store data in pointcloud and function for imwrite?

MartyG-RealSense, 2022-01-04:
There is not a limit on the number of questions that you can ask, so don't worry  :)  For future questions though, it would be best to create your own new case on this support forum using the green **New Issue** button on the front page of the forum at the link below.

https://github.com/IntelRealSense/librealsense/issues

I found a Chinese-language example of a MATLAB RealSense guide for generating a pointcloud and saving depth and RGB with imwrite.  

https://zhuanlan.zhihu.com/p/384836137

I converted it to English in the PDF document below.

[MATLAB-imwrite.pdf](https://github.com/IntelRealSense/librealsense/files/7808372/MATLAB-imwrite.pdf)

ranggo467, 2022-01-05:
thanks bro, but i get error at
imu_accel_ = fs.first_or_default( RS2_STREAM_ACCEL );
can i know how to solve this error?

MartyG-RealSense, 2022-01-05:
Has your script defined the **fs** reference that the IMU instruction relies on, please?

[image]

There is also another MATLAB script at https://github.com/IntelRealSense/librealsense/issues/7222 that accesses the IMU in a similar way.  Whilst that script did not work correctly, that may have been because an L515 camera was being used in that case.  So it may be worth trying if your camera is a 400 Series model.

If you are unable to access IMU data through the RealSense SDK, a discussion in the link below looks at ways to access the data with a MATLAB software tool called the **Navigation Toolbox**.

https://support.intelrealsense.com/hc/en-us/community/posts/1500001312921-Execution-of-motion-module-of-IMU-sensor-by-Matlab

ranggo467, 2022-01-05:
thanks bro, do you know how to stream rgb and depth image with both image align. i still dont get it how to do this while stream in matlab

MartyG-RealSense, 2022-01-05:
Another approach to depth-color alignment in the MATLAB wrapper is shared in https://github.com/IntelRealSense/librealsense/issues/3735

ranggo467, 2022-01-06:
i already view that but i dont understand this 
Note that you should also be able to get intrinsic directly from the "aligned_depth" frame in a similar manner (frame->frame_profile->video_frame_profile->get_intrinsic), but I don't have the Matlab environment to provide a reference code.

MartyG-RealSense, 2022-01-06:
https://github.com/IntelRealSense/librealsense/issues/2541#issue-370575706 provides an example of scripting for get_intrinsic, though with non-aligned frames.

ranggo467, 2022-01-09:
bro i have a question, how to save depth image and rgb image simultaneously, when i try to save by using imwrite its only save rgb image only. I want to save both the image simultaneously. what should i add to my code? This is my code
clc
close all
clear all

% Make Pipeline object to manage streaming
pipe = realsense.pipeline();

% Make Colorizer object to prettify depth output
colorizer = realsense.colorizer();

% Start streaming on an arbitrary camera with default settings
profile = pipe.start();

for j=1:5
align_to = realsense.stream.color;
alignedFs = realsense.align(align_to);
% Get streaming device's name
dev = profile.get_device();
name = dev.get_info(realsense.camera_info.name);
% Get frames. We discard the first couple to allow
% the camera time to settle
%while(true)
for i = 1:25
fs = pipe.wait_for_frames();
end
aligned_frames = alignedFs.process(fs); depth = aligned_frames.get_depth_frame();
% get depth image parameters
depthSensor = dev.first('depth_sensor');
depthScale = depthSensor.get_depth_scale();
depthWidth = depth.get_width();
depthHeight = depth.get_height();

% vidWriterdepthMap = VideoWriter('framesdepth1.avi','Uncompressed AVI');
% open(vidWriterdepthMap);

% retrieve UINT16 depth vector
depthVector = depth.get_data();
% reshape vector, and scale to depth in meters
depthMap = double(transpose(reshape(depthVector, [depthWidth,depthHeight]))) .* depthScale;
%align the depth frames to the color stream
aligned_frames = alignedFs.process(fs);
depth = aligned_frames.get_depth_frame();
%Select rgb frame
color = fs.get_color_frame();
colordata=color.get_data();
img = permute(reshape(colordata',[3,color.get_width(),color.get_height()]),[3 2 1]);
imwrite(depthMap,sprintf('%d.jpg',j))
imwrite(img,sprintf('%d.jpg',j))

% figure(j)
% subplot(2,1,1);
% imshow(depthMap)
% impixelinfo;
% title('Depth Map (m)');
% subplot(2,1,2);
% imshow(img)
% impixelinfo;
% title('Color Map');
v=normalize(depthMap,'range');
v=uint8(v*255);
c = jet(256);
depthMap = ind2gray(v,c);
pause(1)
j
end
 
pipe.stop();

MartyG-RealSense, 2022-01-09:
https://github.com/IntelRealSense/librealsense/issues/9773#issuecomment-926777520 has an example of MATLAB wrapper code written by a RealSense user for saving depth data to PNG file with imwrite.

ranggo467, 2022-01-10:
do you have example how to setting the distance on depth map using matlab?

MartyG-RealSense, 2022-01-10:
Do you mean converting the raw depth values to real-world distances in meters, please?  If that is what you mean then the MATLAB scripting in https://github.com/IntelRealSense/librealsense/issues/9923#issuecomment-961845156 may be helpful.

ranggo467, 2022-01-13:
tq bro

saroshpatel, 2023-11-21:
Hello,
I am trying to use my D405 with MATLAB. I used the windows installer [Intel® RealSense™ SDK 2.0 (v2.54.2)](https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.2) .

During installation I did not see any MATLAB wrapper selection checkbox.
[image]

After installation I do not see any matlab folder in the RealSenseSDK2.0 folder.

Please help

MartyG-RealSense, 2023-11-21:
Hi @saroshpatel  I can confirm that the MATLAB developer package component is not in the 2.54.2 installer program.  It is in the installer for **2.54.1** though.  

[image]

As 2.54.2 was a minor release containing bug-fixes only, it should be fine to use 2.54.1 to install the MATLAB component.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.1

saroshpatel, 2023-11-25:
Thank you for your help ! that worked out.