realsenseai/librealsense #6015: I captured many abnormal depth frame with some white area using realsense D435i

hwenhui1997, 2020-03-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |   Ubuntu 16.04 | 
Kernel Version (Linux Only)    |  4.15.0-88-generic                                        | 
| Platform                                 | PC |
| SDK Version                            |  RealSense 2.0                         | 
| Language                            |  {c++}                          | 
### Issue Description
One of the abnormal frames is as following!
[image]
I have tried to adjust exposure using following code, and it works a little. 
The distance of object matters too.
But I don't know what its real key problem is. 
```
    rs2::device selected_device = profile.get_device();
    auto color_sensor = selected_device.first<rs2::color_sensor>();
    color_sensor.set_option(RS2_OPTION_EXPOSURE,6000);

```

MartyG-RealSense, 2020-03-11:
The stated firmware version (5.10.13) on your D435i is very old, dating back to November 2018.  Could you try updating to the current firmware please and see if that helps?

If you have access to the RealSense Viewer program, an easy way to perform the firmware update is to go to the 'More' option at the top of the Viewer's options side-panel and select from its menu the "Install Recommended Firmware" option.  This will automatically install the latest firmware without you having to provide a firmware file.

[image]

If you are unable to use the RealSense Viewer, the firmware can also be updated using a tool called **rs-fw-update**.

https://dev.intelrealsense.com/docs/firmware-update-tool

dorodnic, 2020-03-14:
Hi @hwenhui1997 
I recommend looking at the Infrared stream. This type of artifact can occur when the infrared image is completely black or completely white. This should almost never happen with Auto-Exposure enabled, but is easy to cause by manually setting the wrong exposure.

hwenhui1997, 2020-03-15:
> Hi @hwenhui1997
> I recommend looking at the Infrared stream. This type of artifact can occur when the infrared image is completely black or completely white. This should almost never happen with Auto-Exposure enabled, but is easy to cause by manually setting the wrong exposure.

Yes, I have set exposure manually,but when I tried to enabled Auto-Exposure, it went wrong with the following error:
```
RealSense error callingrs2_set_option(options:0x1c270c0, option:Auto Exposure Mode, value:1):
object doesn't support option #21
```

dorodnic, 2020-03-15:
Auto exposure is enabled by default when you connect the camera

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

MartyG-RealSense, 2020-03-22:
This case will be closed after 7 days from the date of writing this if there are no further responses.   Thanks!

closed 2020-03-28 by MartyG-RealSense