realsenseai/librealsense #6016: d435i gravity removal

nych9811, 2020-03-11:
Hi.
I'm using D435i, and I want to estimate pose by imu sensor inside D435i.
Before I start to estimate, I finished calibration by tutorial uploaded in Intel website.

I learned we need to remove gravity from accelerometer value to estimate pose, so I have questions.

After I calibrated my IMU sensor by tutorial, should I remove gravity from value myself? If I must do, how can I do?

For example, I got these values after calibrated my imu sensor

Writing files:
accel_A and B.txt
gyro_A and B.txt
[-0.00216658 -0.00114198 -0.00042399]
[1000 1000 1000 1000 1000 1000]
using 6000 measurements.
[[ 1.02186281  0.03536714 -0.00926439]
 [ 0.02256644  1.01664358  0.05166774]
 [-0.00436534 -0.0454576   1.01037547]
 [ 0.00597615  0.02938021  0.33118931]]
residuals: [ 51.3077819   85.60967507 320.59129084]
rank: 4
singular: [443.57067753 433.00927674 418.03927512  77.32178072]
norm (raw data  ): 9.651282
norm (fixed data): 9.802097 A good calibration will be near 9.806650
Would you like to write the results to the camera's eeprom? (Y/N)Y
Writing calibration to device.
  Device PID:  ****
  Device name:  Intel RealSense D435I
  Serial number:  ************
  Firmware version:  05.12.03.00
SUCCESS: saved calibration to camera.
Done.

Please Help me!

Thanks.

MartyG-RealSense, 2020-03-11:
Only the T265 Tracking Camera, which has the same IMU component as the D435i, has built-in support for 6 degrees of freedom (6-DOF) pose. 

[image]

It is possible though to get pose on a D435 camera by using OpenCV or ROS.

https://github.com/IntelRealSense/realsense-ros/issues/1026#issuecomment-566029045

nych9811, 2020-03-12:
Dear @MartyG-RealSense 
Thanks a lot! It makes sense!

closed 2020-03-12 by nych9811

MartyG-RealSense, 2020-03-12:
You're very welcome, I'm glad I could help.  Good luck!   :)