realsenseai/librealsense #6017: Ubuntu realsense-viewer error

nych9811, 2020-03-11:
Hi. I'm using ubuntu 16.04, and realsense-viewer v2.33.1. When I use realsense-viewer in ubuntu, there's error like this picture.

[image]

Does anyone knows solution about this problem?

Please help me....

dorodnic, 2020-03-11:
Hi @nych9811 
This is a warning, not an error, so you can safely ignore it. 
However, it indicates that due to some limitation of the CPU / USB controller, some frames are getting lost. What kind of platform are you using? Also, did you install the SDK from source or via apt-get?

nych9811, 2020-03-11:
Thanks, @dorodnic 
I installed realsense-SDK by tutorial uploaded in github realsense.
I use realsense-viewer in Ubuntu 16.04 in virtual machine, so it can be hard to process it.

Thanks alot.

closed 2020-03-16 by nych9811