realsenseai/librealsense #6018: color fov vs depth fov

omerbrandis, 2020-03-11:
Dear support,

i've noticed that the fov data returned for the depth stream does not match that of the color stream ( for a given resolution).

can you please tell me which of the two is the "effective" fov when I perform aligment of the depth to the rgb ?   ( rs2::align align(RS2_STREAM_COLOR); ) 

using rs version 2.16.1.
firmware version 05.10.06.00.
d415,d435.

getting fov from stream is performed by 
rs2::config   myconfig;
    myconfig.enable_stream(RS2_STREAM_DEPTH, -1,PixelWidth, PixelHeight, RS2_FORMAT_Z16 , 6);
    myconfig.enable_stream(RS2_STREAM_COLOR, -1,PixelWidth, PixelHeight, RS2_FORMAT_RGB8, 6);
    myconfig.enable_device(cameraid);
MyPipelineProfile = myp.start(myconfig);

// get & store depth intrinsics
    auto DepthStream = MyPipelineProfile.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>();
    auto ColorStream = MyPipelineProfile.get_stream(RS2_STREAM_COLOR).as<rs2::video_stream_profile>();
    DepthIntrinsics = DepthStream.get_intrinsics();
    ColorIntrinsics = ColorStream.get_intrinsics();

omerbrandis, 2020-03-11:
( seems my previous post was cut short )
rest of the code for getting fov :
rs2_fov(&ColorIntrinsics, ...);
rs2_fov(&DepthIntrinsics, ...);

thanks,
Omer

MartyG-RealSense, 2020-03-11:
The D435 camera has a wider IR imager component than the one on the D415, so you may experience FOV differences on the D435 due to a wider depth FOV.  The subject is discussed in the link below, which states that the **Align** class can be used to generate a synthetic depth image from the color viewpoint.

https://github.com/IntelRealSense/librealsense/issues/2279

omerbrandis, 2020-03-11:
Dear support, 

I have reviewed #2279, i did not see an answer to my question.
i shall try to restate.
for a given camera ( d435 or d415 , actuall for both of them ) 
the fov provided by the rs2_fov function , for the color stream and the depth stream is not the same.
since i perform alignment, i do not know which fov ( depth or color ) is the one i should use for the "aligned" stream.

thanks,
Omer

MartyG-RealSense, 2020-03-11:
I considered your question carefully.   I believe that if you are aligning depth to color then use the color extrinsics.

dorodnic, 2020-03-17:
Taking FOV of the aligned stream is a nice idea, but it won't work unfortunately.
@omerbrandis - FOV is a property of a camera sensor and that's what we write in the data sheet. 
"Overlapped FOV" is not really well defined (because of occlusion and partial overlap) so we don't have such information to provide

MartyG-RealSense, 2020-03-28:
Closed due to no further responses from the customer.

closed 2020-03-28 by MartyG-RealSense