realsenseai/librealsense #6019: D435 Changes in Exposure dont effect the image

kremerf, 2020-03-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | (Open RealSense Viewer) | 
| Operating System & Version |  Ubuntu 18.04  |
| Platform                                 | NUC  |
| Language                            |  C++                          | 

I am trying to change the exposure time of my RGB-Camera (D435) when launching my code.
The way I was trying to do that is:
```
	pipeline p;
	auto prof = p.start();
	rs2::device dev = p.get_active_profile().get_device();
	dev.query_sensors()[1].set_option(rs2_option::RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);
	auto range = dev.query_sensors()[1].get_option_range(RS2_OPTION_EXPOSURE);
	dev.query_sensors()[1].set_option(RS2_OPTION_EXPOSURE, range.min);
	std::cout << dev.query_sensors()[1].get_info(RS2_CAMERA_INFO_NAME) << std::endl;
	std::cout << "exposure = " << dev.query_sensors()[1].get_option(rs2_option::RS2_OPTION_EXPOSURE) << std::endl;
```
The output in my Terminal:
```
RGB Camera
exposure = 1
```
But unfortunately the images in ROS remained the same.

When changing the value via the RealSense-Viewer parallel to running my programm the results also weren't as expected. Chaning the exposure from 166 to 1 resulted in a brighter image, changing it to a higher value made it darker and then brighter again, when increasing even further.

Any tips how to get actual changes?

PS: The code I used was from several other topics in here such as:
https://github.com/IntelRealSense/librealsense/issues/1624

https://github.com/IntelRealSense/librealsense/issues/4140
Stated, that exposure levels below 50 should work as intended as of Firmware Version 5.11, i am using 5.12.

RealSenseViewer Shows "3.2" next to the USB-Connection, so I think the cable is working.

MartyG-RealSense, 2020-03-11:
If you are using ROS, you can change the exposure with a mechanism called Dynamic Reconfigure that is run immediately after the roslaunch instruction.

https://github.com/IntelRealSense/realsense-ros#set-camera-controls-using-dynamic-reconfigure-params

kremerf, 2020-03-11:
Thanks for the quick reply, the Dynamic Reconfigure produces the same result:

Normal View, Exposure is 166 and Auto_Exposure is active
[image]

Exposure set to 1 and auto_exposure is off
[image]

The expected image should look like this:
[image]

kremerf, 2020-03-11:
It might be the "1".. I tried putting the time to 2 and 3 and the image actually went dark.

Changing the initial code to said value unfortunately doesnt work. It seems as if the settings are being overwritten by something. I dont call any other functions than the ones above.

MartyG-RealSense, 2020-03-11:
I tested your settings with my D435i on a USB 3 connection on a NUC in the RGB section of the RealSense Viewer's controls with RGB auto-exposure disabled and got the results that you expected (black at '1', increasing in brightness as the exposure value is increased, and the brightness did not go dark and then bright when increasing).

Does it work as expected if you try the Viewer on its own?

kremerf, 2020-03-11:
Yes, on both cameras (I have two setups, one with a NUC and one where just with the camera on windows currently). Both even start up with the values set to 1 and being dark.

MartyG-RealSense, 2020-03-11:
I wonder if you are disabling the correct auto-exposure in your script, as the depth and RGB have separate auto-exposure settings.  I believe that both use the RS2_OPTION_ENABLE_AUTO_EXPOSURE instruction, but they are distinguished by the **set.option instruction that precedes it.

https://github.com/IntelRealSense/librealsense/issues/3224#issuecomment-461430639

kremerf, 2020-03-11:
I do it like this:

```
auto colorSensors = prof.get_device().query_sensors()[1];
	colorSensors.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);
	colorSensors.set_option(RS2_OPTION_EXPOSURE, 2);
	//auto range = dev.query_sensors()[1].get_option_range(RS2_OPTION_EXPOSURE);
	colorSensors.set_option(RS2_OPTION_EXPOSURE, 2); // set minimal allowed exposure
	std::cout << colorSensors.get_info(RS2_CAMERA_INFO_NAME) << std::endl;
	std::cout << "exposure = " << colorSensors.get_option(RS2_OPTION_EXPOSURE) << std::endl;
	std::cout << "exposure2 = " << colorSensors.get_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE) << std::endl;

RGB Camera
exposure = 2
exposure2 = 0

```

The Camera info says "colorSensors" points to a RGB Camera.

MartyG-RealSense, 2020-03-11:
Manual exposure is a somewhat tricky subject.  When auto-exposure is disabled, the camera reverts to the last known manual exposure value used before auto-exposure was active.

https://github.com/IntelRealSense/librealsense/issues/1624#issuecomment-389085387

kremerf, 2020-03-11:
Yes, I remember that one. I got until:

```
terminate called after throwing an instance of 'rs2::wrong_api_call_sequence_error'
  what():  start() cannot be called before stop()
```

> I am still learning the more advanced aspects of Librealsense programming

Looks like there is a lot do discover ( :

MartyG-RealSense, 2020-03-11:
Librealsense is a hugely flexible language, which means that there is also plenty to learn.  As with most things involving study, it is best to focus on one aspect at a time so that it does not become overwhelming.

Advice about that error is given here:

https://github.com/IntelRealSense/librealsense/issues/1889#issuecomment-426601102

kremerf, 2020-03-11:
`<arg name="fps"                 value="6"/>`
 So 6.

It starts with 12, wow.

```
exposure = 8500
meta exposure = 8500
exposure = 8500
meta exposure = 8500

```

MartyG-RealSense, 2020-03-11:
That was what I was thinking about the FPS.  that error has been reported when 6 FPS is used.  6 FPS is prone to problems occurring that disappear when a minimum of 15 FPS is used.

kremerf, 2020-03-11:
Currently I get this one:

```
Stereo Module
terminate called after throwing an instance of 'rs2::backend_error'
  what():  get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
```

The code looks like this:

```
device dev = p.get_active_profile().get_device();
auto sensor = dev.query_sensors()[0];
cout << sensor.get_info(RS2_CAMERA_INFO_NAME) << std::endl;
sensor.set_option(rs2_option::RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);
std::this_thread::sleep_for(std::chrono::seconds(1));
cout << "exposure = " << sensor.get_option(rs2_option::RS2_OPTION_EXPOSURE) << endl;
auto frameset = p.wait_for_frames();
cout << "meta exposure = " << frameset.get_depth_frame().get_frame_metadata(rs2_frame_metadata_value::RS2_FRAME_METADATA_ACTUAL_EXPOSURE) << endl;
sensor.set_option(rs2_option::RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);
std::this_thread::sleep_for(std::chrono::seconds(1));
cout << "exposure = " << sensor.get_option(rs2_option::RS2_OPTION_EXPOSURE) << endl;
frameset = p.wait_for_frames();
cout << "meta exposure = " << frameset.get_depth_frame().get_frame_metadata(rs2_frame_metadata_value::RS2_FRAME_METADATA_ACTUAL_EXPOSURE) << endl;
```

Or I get the following if I remove all the sleepFors

```
Stereo Module
exposure = 8500
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  metadata not available

```

MartyG-RealSense, 2020-03-11:
This subject goes beyond my current programming knowledge to be honest.  I will tag in another RealSense team member for advice.   @dorodnic

kremerf, 2020-03-11:
Thank you very much so far. I have some points to look through - thanks to you. Maybe I get some new pieces of Information until then.

kremerf, 2020-03-11:
```
[ INFO] [1583937810.185421665]: Initializing nodelet with 8 worker threads.
[ INFO] [1583937810.401519833]: RealSense ROS v2.2.12
[ INFO] [1583937810.401584320]: Running with LibRealSense v2.32.1
RGB Camera
166
[ INFO] [1583937810.466036837]:
[ INFO] [1583937810.530661116]: Device with serial number 740112071197 was found.

[ INFO] [1583937810.530748641]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video0 was found.
[ INFO] [1583937810.530786595]: Device with name Intel RealSense D435 was found.
[ INFO] [1583937810.531612993]: Device with port number 2-3 was found.
[ INFO] [1583937810.531699940]: Resetting device...
RGB Camera
2
RUN 19
[ INFO] [1583937816.590447047]:
[ INFO] [1583937816.604553504]: Device with serial number 740112071197 was found.

[ INFO] [1583937816.604656020]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video0 was found.
[ INFO] [1583937816.604724877]: Device with name Intel RealSense D435 was found.
[ INFO] [1583937816.605538457]: Device with port number 2-3 was found.
[ INFO] [1583937816.622052178]: getParameters...
RGB Camera
166
RUN 18
RGB Camera
166
RUN 17
RGB Camera
166
RUN 16
[ INFO] [1583937817.287272083]: setupDevice...
[ INFO] [1583937817.287375442]: JSON file is not provided
[ INFO] [1583937817.287438358]: ROS Node Namespace: D435
[ INFO] [1583937817.287499379]: Device Name: Intel RealSense D435
[ INFO] [1583937817.287567714]: Device Serial No: 740112071197
[ INFO] [1583937817.287698259]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video0
[ INFO] [1583937817.287787730]: Device FW version: 05.12.02.100
[ INFO] [1583937817.287853750]: Device Product ID: 0x0B07
[ INFO] [1583937817.287934067]: Enable PointCloud: On
[ INFO] [1583937817.288011089]: Align Depth: On
[ INFO] [1583937817.288075983]: Sync Mode: On
[ INFO] [1583937817.288255148]: Device Sensors:
[ INFO] [1583937817.288404417]: Stereo Module was found.
[ INFO] [1583937817.288506869]: RGB Camera was found.
[ INFO] [1583937817.288645673]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1583937817.288717065]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1583937817.288777095]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1583937817.288851663]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1583937817.288928094]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1583937817.288992183]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1583937817.289108116]: Add Filter: pointcloud
[ INFO] [1583937817.290892758]: num_filters: 1
[ INFO] [1583937817.290972289]: Setting Dynamic reconfig parameters.
[ INFO] [1583937817.687161245]: Done Setting Dynamic reconfig parameters.
[ INFO] [1583937817.690183628]: depth stream is enabled - width: 1280, height: 720, fps: 30, Format: Z16
[ INFO] [1583937817.690692551]: infra1 stream is enabled - width: 1280, height: 720, fps: 30, Format: Y8
RGB Camera
166
RUN 15
[ INFO] [1583937817.711998548]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1583937817.716331373]: setupPublishers...
[ INFO] [1583937817.733484720]: Expected frequency for depth = 30.00000
[ INFO] [1583937818.008168538]: Expected frequency for infra1 = 30.00000
RGB Camera
166
RUN 14
[ INFO] [1583937818.242389963]: Expected frequency for aligned_depth_to_infra1 = 30.00000
[ INFO] [1583937818.478713095]: Expected frequency for color = 30.00000
RGB Camera
166
RUN 13
[ INFO] [1583937818.713672993]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1583937818.959888955]: setupStreams...
[ INFO] [1583937818.961550014]: insert Depth to Stereo Module
[ INFO] [1583937818.961650916]: insert Color to RGB Camera
[ INFO] [1583937818.961693046]: insert Infrared to Stereo Module
 11/03 15:43:39,035 WARNING [140690289493760] (backend-v4l2.cpp:1622) Metadata frame arrived in idle mode.
[ INFO] [1583937819.035521142]: SELECTED BASE:Depth, 0
[ INFO] [1583937819.146011921]: RealSense Node Is Up!
terminate called after throwing an instance of 'rs2::backend_error'
  what():  xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy

[...] process has died [pid 4775, exit code -6, [...]

```

Big chunk of code, but there can be seen, that the previously mentioned loop itself works until "RealSense Node Is Up!" Where my ROS-Package dies.
But the saved Exposure Time gets overwritten almost immediately (from 2 back to 166).

The "Run xy" is a personal countdown, so that I dont run my core functions before everything is properly loaded.

kremerf, 2020-03-11:
We are making progress here!
I decided to set the param every loop (which worked first) but it killed the package as well.
So I stopped that right after the "[ INFO] [1583937816.622052178]: getParameters...". Which worked like a charm - well, not without an additional error of course. But still, I do have an alive package now, that has the exposure set to 1.

```
hwmon command 0x75 failed. Error type: Invalid parameter (-6).
[ERROR] [1583938276.185642930]: hwmon command 0x75 failed. Error type: Invalid parameter (-6).
```

This error does make me feel, like we still aren't 100% done yet..

I hope my additional comments are ok, i can also just edit previous comments..^^

MartyG-RealSense, 2020-03-11:
Providing additional information is always welcome.  :)   Thanks!

Perhaps not directly related to your case, but I found a case where someone else caught that error when trying to set a Region of Interest for the exposure.

https://github.com/IntelRealSense/librealsense/issues/5308

kremerf, 2020-03-11:
I think his solution was to not call "something" multiple times. I (at this moment) dont call my code more than one time, so i unfortunately dont really see how to deviate.

I get an additional warning now:
`WARNING [139728192501504] (backend-v4l2.cpp:973) Frames didn't arrived within 5 seconds`
Mentioned in https://github.com/IntelRealSense/librealsense/issues/1086 . Looking into how I can use hardware_reset(), which currently does not work.

I will work on this tomorrow.

MartyG-RealSense, 2020-03-22:
This case will be closed after 7 days from the date of writing this if there are no further responses.   Thanks!

closed 2020-03-28 by MartyG-RealSense