realsenseai/librealsense #6020: Calculate height of moving object under camera

tispratik, 2020-03-11:
We have cardboard boxes flowing on a conveyor belt which a robotic system has to pick up. It is important to gauge the height of the object to the millimeter accuracy level. The camera D415 is mounted at 24" height from conveyor belt facing the conveyor belt. As the object enters the FOV of the camera we start measuring distances of the center of the boxes as seen from the camera.

We know the distances of each pixels when the conveyor belt is empty. So, when we measure distance to a pixel on the box, we need to know the angle of the pixel from the vertical plane of the camera in order to do the appropriate trignometry to compute the height of object.

How can i determine the angle (theta) of a particular pixel relative to the axis vertical to the plane of the depth sensors?

What we have:
- Distance of pixel when no objects on conveyor (a)
- Distance of pixel on the box when a box is placed on the conveyor (b)

What we need to compute:
- Distance between a and b which will give us height of the cardboard box

MartyG-RealSense, 2020-03-11:
I am guessing that you have the camera centered over the conveyor belt so that when a box passes underneath, the center of the box roughly corresponds to the center of the camera image?

If you want to be sure that the box is centered under the camera and you aren't reading its edge as it enters or exits the camera view, a couple of possibilities come to mind:

1.  Define a 'bounding box' in the center of the camera image so that the camera ignores depth data outside of that box.  

A recent article described a project where Lego bricks move under a vertically mounted downward-pointing camera on a makeshift conveyor and have bounding boxes defined around them to identify them.  This may give some useful insights about use of bounding boxes on conveyor objects.

https://github.com/IntelRealSense/librealsense/issues/5393

Or;

2.  Set up a physical light sensor positioned just past the camera centerline so that the camera only performs a capture of the center of the box when the edge of the box that reaches the light sensor first passes through the light beam and breaks it whilst the box center is under the camera.

An alternative method might be to put a 'tag' image on the center of the top of the box that can be detected by the camera and trigger the robotic system to respond.  For a 400 Series depth camera, an **Aruco** tag could be used.

https://docs.opencv.org/3.1.0/d5/dae/tutorial_aruco_detection.html

If you just need to measure between two points though, the **Measure** C++ sample program may meet your needs.

https://github.com/IntelRealSense/librealsense/tree/master/examples/measure

tispratik, 2020-03-14:
Thank you for the ideas Marty. Will explore further. We're looking at simple trignometry to achieve the results. Also, do you know if depth calibration smoothes out the effects of lens warping?

MartyG-RealSense, 2020-03-14:
If you are referring to concave and convex distortion effects such as barrel distortion and pincushion distortion, @dorodnic offers advice on the subject here:

https://github.com/IntelRealSense/librealsense/issues/4939#issuecomment-535458739

tispratik, 2020-03-14:
Thank you for the link Marty. Will do some tests on the two distortion types on empty conveyor and test objects.

MartyG-RealSense, 2020-03-14:
You're very welcome.  Let us know how the tests turn out please!

MartyG-RealSense, 2020-03-22:
This case will be closed after 7 days from the date of writing this if there are no further responses.   Thanks!

closed 2020-03-28 by MartyG-RealSense