realsenseai/librealsense #6023: Vertical camera - rotate depth frames?

keithahern, 2020-03-11:
| Camera Model                       | {D400  | 
| Firmware Version                   | V2.31.0 | 
| Operating System & Version |   Linux / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2.X }                          | 
| Language                            |  {python}                          | 
| Segment			|  {/others }                          | 

### Issue Description

I have a developed application in python and some tests indicate the performance of the camera would be better if rotated 90 degrees.  It would be extremely time consuming (and confusing!) to rewrite to change all the code which assumes x is horizontal and y is vertical.  We use a lot of librealsense API calls.  Is there an easer to to rotate the camera and still have x, y behave correctly? e.g. in 480x640 resolution?

MartyG-RealSense, 2020-03-11:
I believe that you wish to rotate the camera 90 degrees but rotate the image 90 degrees in the opposite direction so that it appears in its normal orientation, please?

If you are able to use OpenCV in your application, it has the ability to rotate an image.  the link below gives guidance about doing this with Python.

https://note.nkmk.me/en/python-opencv-numpy-rotate-flip/

keithahern, 2020-03-11:
no, thats not correct.

I want to turn the camera 90 degrees and still have X axis be horizontal, Y is vertical.  So, for example in landscape the camera is 640x480 and in portrait it is 480x640.  

If the camera is NOT rotated the follow code is correct
dist = depth_frame.get_distance(x, y)

if the camera IS rotated then the code needs to be
dist = depth_frame.get_distance(y, x)

which will get confusing very quickly, every x must swapped for y.  

A solution could be done at the frame level
depth_frame = depth_frame.rotate(rs2.90_DEGREES_CLOCKWISE)
dist = depth_frame.get_distance(x, y)

but would probably be better at the device level so the color and IR frames also do the 'right thing'
config = rs.config()
config.device_orientation(rs2.90_DEGREES_CLOCKWISE)
config.enable_stream(rs.stream.depth, 480, 640, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 480, 640, rs.format.bgr8, 30)

Or something like that.  Does that help?

MartyG-RealSense, 2020-03-11:
I considered your question carefully.   It sounds as though it would be much easier to keep the camera in its normal orientation and seek to improve the output results by other means.  Could you tell me please what problems you have with the camera in its normal orientation that seem to be better when it is rotated 90 degrees please?   Thanks very much!

keithahern, 2020-03-11:
I am measuring a tall vertical groove at close range.  When the camera is horizontal the cameras can not see deeply into the groove. When the camera is vertical it can.

dorodnic, 2020-03-11:
Hi @keithahern 
We have such capability on the upcoming L500 camera, and we may be able to port this to D400 cameras in one of the future releases. I agree it's not as easy as rotating the image, all calibration data and other parameters need to be updated, and we do want to enable it in the SDK at some point

keithahern, 2020-03-11:
Great.

Thank you Sergey and Marty.

RMichaelPickering, 2020-05-12:
@dorodnic @MartyG-RealSense I would like to upvote this request, as we would like to use it in our application, which involves detecting human faces. The point is that having the largest field of view in the vertical (i.e. a portrait orientation rather than landscape) helps to allow for a wider range of subject heights, potentially including people sitting in wheelchairs, etc. It would be most helpful to have the requested feature for our application, too!!

MartyG-RealSense, 2020-05-12:
@RMichaelPickering   Thanks so much for your supportive input!

You could also increase the vertical field of view by having two cameras stacked vertically with their fields of view overlapping, and this should avoid the image rotation implications that Dorodnic mentioned.  I can appreciate that this has hardware cost implications for projects though.

Kevin-Delnoije, 2021-02-24:
any update on this? a vertical orientation parameter or a function that can rotate realsense frames would be very helpfull!

keithahern, 2021-02-24:
My solution has been to just used RealSense for initial depth and color frame acquisition and then move to the (also Intel backed) Open3D library and do my processing from there (i.e. I'm not using any RealSense projection/deprojection APIs.  In Open3d it's much easier to merge and manipulate point clouds, render to 2D, create cross platform UIs etc. The documentation is lacking in places but it's allowed me to move forward.

See below on how to get an rs frameset into an open3d PointCloud and transform it - its very fast.

`
depth_frame = frameset.get_depth_frame()
color_frame = frameset.get_color_frame()

depth_image = np.asanyarray(depth_frame.get_data())
color_image = np.asanyarray(color_frame.get_data())

img_depth = o3d.geometry.Image(depth_image)
img_color = o3d.geometry.Image(color_image)

img_rgbd = o3d.geometry.RGBDImage.create_from_color_and_depth(img_color,img_depth)
intrinsics = profile.as_video_stream_profile().get_intrinsics()
pinhole_camera_intrinsic = o3d.camera.PinholeCameraIntrinsic(intrinsics.width, intrinsics.height, intrinsics.fx, intrinsics.fy, intrinsics.ppx, intrinsics.ppy)

pcd = o3d.geometry.PointCloud.create_from_rgbd_image(img_rgbd, pinhole_camera_intrinsic)
pcd.transform([[1, 0, 0, 0], [0, -1, 0, 0], [0, 0, -1, 0], [0, 0, 0, 1]])

`

Kevin-Delnoije, 2021-02-24:
> My solution has been to just used RealSense for initial depth and color frame acquisition and then move to the (also Intel backed) Open3D library and do my processing from there (i.e. I'm not using [...]

Thanks this solves the problem!

MartyG-RealSense, 2021-02-24:
@Kevin-Delnoije   The feature request related to this case remains active but I have no further updates to provide about it.  I see that the advice of @keithahern has solved your problem though, which is excellent news.

@keithahern  Thanks so much for the successful help you provided to @Kevin-Delnoije

qualityland27, 2021-12-03:
> Hi @keithahern We have such capability on the upcoming L500 camera, and we may be able to port this to D400 cameras in one of the future releases. I agree it's not as easy as rotating the image, all [...]

@dorodnic I am working with the L515. Could you kindly tell me how to do it in Python? I cant find it ...

RMichaelPickering, 2021-12-03:
FWIW, we've implemented some frame rotation code using OpenCV but it would
of course be far better to do as much of this as possible in the camera! If
the capability from the L500 could be ported to the D4xx cameras and fully
supported in the SDK that would be excellent!!

Mike

R. Michael Pickering CBIP
President & CEO
CloudConstable Incorporated
10271 Yonge Street, Suite 321
Richmond Hill, ON L4C 3B5
Canada
www.cloudconstable.com
***@***.***
Toronto:  416-721-1826
Orlando: 321-337-2824

MartyG-RealSense, 2021-12-03:
Hi @RMichaelPickering  The official feature request to Intel to provide support on the 400 Series cameras for rotation of depth frames remains active but there is no progress to report at this time.   

@qualityland27  At the time that the statement about L515 support for image rotation was made, the camera was in its pre-release phase.  I do not have any information regarding whether the suggested feature was included in the RealSense SDK post-launch.

FeiSong123, 2022-08-13:
> 

Hi, I'm using L515 and I also need to rotate camera 180 degree for my project. This is very helpful to me, so can you tell me more about the usage ?

MartyG-RealSense, 2022-08-13:
Hi @FeiSong123  If you rotate the L515 camera 180 degrees in the Y direction then although the image would be upside down, you would not need to make any changes to the image as its values should still be correct

If you needed to rotate / flip the image after rotating the camera 180 degrees so that the image was the 'right way up' then you could do that using the methods in this discussion or at https://github.com/IntelRealSense/librealsense/issues/4395#issuecomment-511113996 and https://github.com/IntelRealSense/librealsense/issues/9088

FeiSong123, 2022-08-13:
I also prefer to rotate at the camera level, and I see dorodinc commented that L515 has such capability. What should I do with C++(I can't find this function device_orientation with config).

MartyG-RealSense, 2022-08-13:
The comment was made before the L515 was first released on sale, and not every planned feature is released for various reasons (for example, if the feature did not work well).  As far as I am aware there is not a dedicated image rotation feature for the L515.

FeiSong123, 2022-08-14:
That's a pity. Nevertheless, I still want to thank you for your guidance.

maciejandrzejewski-digica, 2022-11-30:
@dorodnic @MartyG-RealSense According to what @dorodnic has written: "I agree it's not as easy as rotating the image, all calibration data and other parameters need to be updated, and we do want to enable it in the SDK at some point".

I would like to know if we can expect same depth/image accuracy, quality, performance, reponse, etc. if the camera is rotated 90* left/right? For some reason calibration data is mentioned. So rotating those frames manually using some other software like OpenCV is to be considered sub-optimal because camera is not positioned up-right?

MartyG-RealSense, 2022-11-30:
Hi @maciejandrzejewski-digica  There are no performance issues with physically rotating only the camera 90 degrees.  There would be a problem with calibration values if the **frame** was then rotated with software such as OpenCV to make the image display in the normal orientation.

MartyG-RealSense, 2022-11-30:
Depending on the maximum depth range required, the RealSense **D405** model (or its D401 PCB board version) may be an easy fit for a device case as it is a small square rather than wide horizontal shape.  Its ideal depth range is 0.7 cm to 50 cm, meaning that it is suited to very close range depth sensing applications rather than multiple-meter distances.

https://store.intelrealsense.com/buy-intel-realsense-depth-camera-d405.html

If a longer-distance camera model is required, a RealSense user earlier in this discussion at https://github.com/IntelRealSense/librealsense/issues/6023#issuecomment-785071263 shared a software workaround for frame rotation.

There is unfortunately not any further information available about the implications to camera parameters of frame rotation other than dorodnic's statement at https://github.com/IntelRealSense/librealsense/issues/6023#issuecomment-597615585

MartyG-RealSense, 2024-12-01:
Hi everyone, have any of you been able to test the new rotation post-processing filter implemented on the **development** branch of librealsense, please?  Thanks!

https://github.com/IntelRealSense/librealsense/tree/development

mewescott, 2024-12-02:
I would like to test it but I haven't figured out how it even works.  Do you know how to use it?  My install is using the ROS 2 wrapper which he said it is not supported yet but I would like to know how to use it without the wrapper too.  Maybe I can figure out a way to make it work that way.

Nir-Az, 2024-12-02:
> I would like to test it but I haven't figured out how it even works. Do you know how to use it? My install is using the ROS 2 wrapper which he said it is not supported yet but I would like to know how to use it without the wrapper too. Maybe I can figure out a way to make it work that way.

You can use this code reference as a usage example:

https://github.com/IntelRealSense/librealsense/blob/development/examples/post-processing/rs-post-processing.cpp

Nir-Az, 2024-12-12:
Thanks to @noacoohen  now also available on RS ROS Node.
See PR #3274

mewescott, 2024-12-14:
Thanks for the heads up.  Where can I find instructions for install/use?  Which repo branch of realsense library and wrapper do I need?

Nir-Az, 2024-12-15:
Please build from source librealsense development branch: https://github.com/IntelRealSense/librealsense/tree/development
and ros wrapper ros2-development branch: https://github.com/IntelRealSense/realsense-ros/tree/ros2-development

Both have install instructions when building from source

mewescott, 2024-12-15:
Thank you.  Can you tell me how to use the rotate filter in ROS after installation.  I just installed the development branch from both on Thursday.  Do I need to do it again or is there a way to just update them?

Nir-Az, 2024-12-15:
PR [3274](https://github.com/IntelRealSense/realsense-ros/pull/3274)  was merged on 12/12
Depends when you took it.
But you can `git pull` your forks to latest and it will be OK

mewescott, 2024-12-16:
I went through the instructions in both repo's and I can't figure out how
to use the rotation filter with ROS.  Where can I find instructions?  I
need to turn my images 90 degrees as my cameras are mounted narrow side
down.  I plan to use Aruco markers and human detection so I would like to
get the orientation of the image right first.

noacoohen, 2024-12-16:
You can find instructions for using the rotation filter with ROS in this launch file:  
[rs_launch.py on the ros2-development branch](https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/launch/rs_launch.py)

To enable and configure the rotation filter, you can use the following parameters:  
1. **rotation_filter.enable:**
   - Default: false
   - Enable or disable the rotation filter.  

2. **rotation_filter.rotation**:  
   - Default: 0.0
   - Set the rotation angle. Supported values are `0.0`, `90.0`, `-90.0`, and `180.0`.  

You can configure these parameters  to rotate the images as needed for your setup.

mewescott, 2025-01-22:
Any progress on rotating the color stream?  I need this for my process right now.  Any work arounds if it is not complete.  My camera is rotated 90 degrees and Google HRI will not conform to it. My body tracking is 90 degrees off.

Nir-Az, 2025-01-22:
Hi @mewescott ,
Rotating the color is more complex than rotating the depth/IR since color steam has much more formats. It takes some time but this feature is WIP.
Stay tuned..

Nir-Az, 2025-01-22:
For rgb permutations (rgba,argb… non YUYV) you can easily rotate using opencv methods on the user app side until the SDK will natively supports it.

mewescott, 2025-01-27:
I was able to rotate the image with opencv and use the depth rotation built into you SDK.  However, it is real slow.  I need a faster way without all the lag.  Is it possible to rotate the intrinsic matrixes or any other calibration data to get rviz and other processes to interpret the image as rotated?  If not, any ideas on how to speed this up?  I need real time detection of people for a safety application.

Nir-Az, 2025-01-27:
I wonder which part is slow, opencv or the SDK work..
Anyway as a first step we implemented the rotation with CPU implementation.
This is a native implementation that do not use HW accelerators.
Once we will see the performance is bad and we have a demanding, we can explore some accelerators implementation like (SSE3/AVX/CUDA) and leverage HW accelerators like we do with the formats converting.
But this will take time..

noacoohen, 2025-01-30:
Hey @mewescott , what are the color formats you work with?

mewescott, 2025-01-30:
Looks like RGB8 for color and Z16 for depth.

MartyG-RealSense, 2025-05-21:
The original issue was about depth frame rotation.  As depth and IR frame rotation has been implemented and the feature request has therefore been closed as done. I will close this case.

closed 2025-05-21 by MartyG-RealSense