realsenseai/librealsense #6024: sensor dimensions

scizors, 2020-03-11:
Hello,
 I am using D415 
I would like to know about RGB sensor dimension sensor width,height and diagonal in (mm)

where can I find this mechanical data information 

Thanks for the reply

MartyG-RealSense, 2020-03-11:
In the current edition of the RealSense 400 Series data sheet document, the sensor information is on pages 35 to 37.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

scizors, 2020-03-11:
Thanks for the reply Is this the latest one 
Revision 007
October 2019
if so
I need in physical dimension in (mm) but its in degrees

MartyG-RealSense, 2020-03-11:
That is the latest edition for the 400 Series, yes.

Apologies, I misunderstood your question about the information you were seeking.   Pages 102 and 106 of the data sheet have mechanical drawings for D415, but if you are seeking the exact physical size in mm of the RGB sensor then it may be difficult to work out from those drawings, as the deal with the distances between sensors.

scizors, 2020-03-11:
yes I know I am looking for one RGB sensor dimension in (mm) which is present in the camera D415

MartyG-RealSense, 2020-03-11:
The 400 Series cameras use the **OmniVision OV2740** component for the RGB sensor.  If Intel are using the component relatively unchanged then the information in the link below may help.

https://www.gophotonics.com/products/cmos-image-sensors/omnivision-technologies/21-118-ov2740

It gives the physical dimensions of the sensor as 3855 µm x 2919 µm

scizors, 2020-03-11:
perfect thanks a lot

closed 2020-03-11 by scizors

Saafke, 2022-08-18:
Just for completion, you can find all specifications of Intel Realsense sensors here: https://en.wikipedia.org/wiki/Intel_RealSense