realsenseai/librealsense #6025: Error while saving rotated depth image to ply

Allius27, 2020-03-11:
| | |
|---|---|
|**librealsense**|2.32.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435|
|**Serial Number**|915112071095|
|**Firmware Version**|05.12.02.100|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|901523021687|
|**Firmware Update Id**|901523021687|

Good evening.

When I am trying save rotated depth image I got this message:
`Error occured during execution of the processing block! See the log for more info`

Gif demo:
[image]

Code with problem:
```
#include <librealsense2/rs.hpp>
#include <vector>
#include <iostream>

#include <opencv2/core/types_c.h>
#include <opencv2/core/core.hpp>
#include <opencv2/imgproc/imgproc.hpp>
#include <opencv2/highgui/highgui.hpp>

using namespace std;
using namespace rs2;
using namespace cv;

bool isRotate;
cv::RotateFlags rotateAngle;
string angleStr;

int main(int argc, char * argv[]) try
{
    if ( argc != 2 ) {
        cout << "Use one of the next keys: 0, 90, 180, 270" << endl;
        return 1;
    }

    if ( atoi(argv[1]) == 0 ) {
        isRotate = false;
        angleStr = "0";
    }
    else
        isRotate = true;

    if ( atoi(argv[1]) == 90 ) {
        rotateAngle = cv::RotateFlags::ROTATE_90_CLOCKWISE;
        angleStr = "90";
    }
    if ( atoi(argv[1]) == 180 ) {
        rotateAngle = cv::RotateFlags::ROTATE_180;
        angleStr = "180";
    }
    if ( atoi(argv[1]) == 270 ) {
        rotateAngle = cv::RotateFlags::ROTATE_90_COUNTERCLOCKWISE;
        angleStr = "270";
    }

    rs2::pipeline pipe;
    rs2::colorizer colorizer;
    rs2::threshold_filter thr_filter;

    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_COLOR);
    cfg.enable_stream(RS2_STREAM_DEPTH);

    thr_filter.set_option(RS2_OPTION_MIN_DISTANCE, 0.3);
    thr_filter.set_option(RS2_OPTION_MAX_DISTANCE, 0.7);

    pipe.start(cfg);

    rs2::frame_queue frame_queue;
    rs2::processing_block procBlock( [&](rs2::frame f, rs2::frame_source& src )
    {
        const int w = f.as<rs2::video_frame>().get_width();
        const int h = f.as<rs2::video_frame>().get_height();

        cv::Mat image(cv::Size(w, h), CV_16UC1, (void*)f.get_data(), cv::Mat::AUTO_STEP);

        cv::rotate( image, image, rotateAngle );

        auto res = src.allocate_video_frame(f.get_profile(), f, 0, image.cols, image.rows);
        memcpy((void*)res.get_data(), image.data, w * h * 2);

        src.frame_ready(res);
    });
    procBlock.start(frame_queue);

    while ( true ) {
        rs2::frameset frameset = pipe.wait_for_frames();
        rs2::video_frame color = frameset.get_color_frame();
        rs2::frame depth = frameset.get_depth_frame();

        if ( !color || !depth )
            continue;

        depth= thr_filter.process(depth);

        if (isRotate) {
            procBlock.invoke( depth );
            depth = frame_queue.wait_for_frame();
        }

        const int w = depth.as<rs2::video_frame>().get_width();
        const int h = depth.as<rs2::video_frame>().get_height();
        cv::Mat depthMat (cv::Size(w, h), CV_8UC3, (void*)depth.apply_filter(colorizer).get_data());

        cv::imshow("depthMat", depthMat);

        char c = (char)cv::waitKey(1);

        if ( c == 27 )
            break;

        if ( (c == 99) || (c == 67) ) {
            rs2::pointcloud pc;
            rs2::points points = pc.calculate(depth);
            points.export_to_ply( "depth" + angleStr + ".ply" , rs2::video_frame{ rs2::frame{} } );
        }
    }

    return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}
```

MartyG-RealSense, 2020-03-11:
As the problem seems to be happening at the pc.calculate section of the script, I wonder if it would help to divide the line up into separate instructions like the **pointcloud** example does.

https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud

By this,  I mean first define rs2::points without making it equal to anything.

**rs2::points points;**

And then elsewhere in the script, assigning a value to the created 'points' object:

**points = pc.calculate(depth);**

dorodnic, 2020-03-12:
Hi @Allius27 
Your code indicate very good understanding of the API, and I need to reproduce and look into what exactly can be the problem, but my intuition is that pointcloud is crashing because rotated frame is still using original (not rotated) stream profile. I'd try making a copy of original stream profile using [clone](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_frame.hpp#L259), but this may not be the end of it.

MartyG-RealSense, 2020-03-22:
This case will be closed after 7 days from the date of writing this if there are no further responses.   Thanks!

closed 2020-03-28 by MartyG-RealSense