realsenseai/librealsense #6027: Recording bag frame freezes randomly

kevindehecker, 2020-03-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.13.3.0 | 
| Operating System & Version |   Ubuntu 18   | 
| Kernel Version (Linux Only)    |  5.3.0-40-generic  | 
| Platform                                 | PC  |
| SDK Version                            |  2.33.1                         | 
| Language                            | n.a.                        | 
| Segment			|  Robot                        | 

### Issue Description
I experience irregular frame loss when recording bag files. This happens anywhere between every few seconds up tens of seconds apart.
I use a callback-based method to retrieve images from the realsense but luckily I think I can also reproduce in the Realsense-viewer itself. In case of the viewer, actually my whole computer freezes up for a second or so, only when recording. I'm retrieving just the ir images on 840x468@60fps:
[image]

I can reproduce on several different types of computers, all with fast SSD's, among which a Intel NUC7i3BNK with a Crucial P1 SSD, 500GB, M.2. I say that should be fast enough.

I turned on the logs: and these errors appear when the freezes happen:
```
 11/03 15:42:20,930 ERROR [140683543287552] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 11/03 15:42:20,947 WARNING [140683543287552] (types.cpp:49) Out of frame resources!
```
My own application does not have a GUI (and also does not completely freeze up the computer), but I'm definitely missing out on frames (the callback function just skips them) in more or less the same frequency as the freezes happen in the viewer.

Any chance you guys can run a test on your end to see if the recording bags of a few minutes still performs as it used to?

dorodnic, 2020-03-11:
Hi @kevindehecker 
What percentage of frames are we talking about? Are these drops random or consecutive?

kevindehecker, 2020-03-11:
Anywhere in the order of 1-60 frames are lost. It seems to be quite random, sometimes the freezes are a minute apart, sometimes just a few secs.

kevindehecker, 2020-03-11:
E.g. 
[image]

x: time in seconds
y axis = current_frame_id - previous_frame_id. (So this is 1 if there is no problem)

In this case my drone crashed due to the big framedrop at the end, but if looked closely at the of the graph one can see the Realsense started to capture images again after the framedrop and likely would have continued on this path.

neilyoung, 2020-03-16:
I have observed the same problem with the T265. The frame delivery is stuck every now and then for a couple of seconds. If I reconstruct the video from the images extracted from the rosbag, there is a visible jump. In my experience with the T265 it happens every 10 to 20 seconds. Only if video is recorded. Poses are gap free. I have reported here https://github.com/IntelRealSense/librealsense/issues/5470 and there is also a visualization of the outages from a walk outside here https://github.com/IntelRealSense/librealsense/issues/5472

RealSenseCustomerSupport, 2020-03-25:
----------------------------------------------

Just wondering, is there any observation of sudden CPU or IO usage spike before the sequential frame drops?

neilyoung, 2020-03-25:
@RealSenseCustomerSupport I cannot confirm to have higher CPU load in that case. 

It seems to have to do with the USB load. If I record just poses then the effect is not noticeable. If I record poses and video, then I can see it twice: Once in the stuck display of positions on my floor plan, once in the extracted video afterwards (jumpy).  But there is no extraordinary load to the system.

kevindehecker, 2020-03-25:
[image]

I at least do see an effect in CPU, but not a spike but a dip. During the blue circle my computer basically froze.

neilyoung, 2020-03-25:
Ah, I didn't especially watch this effect, but makes sense. Computer frozen, no delivery of poses, outage... It hangs somewhere, I would say :)

RealSenseCustomerSupport, 2020-04-13:
----------------------------------------------

Just wondering if there is any good news on your side to overcome this problem?

kevindehecker, 2020-04-14:
Nope.
Well, since this was continuously crashing our drones, I needed this fixed asap. So actually I used the quarantine + the Eastern weekend to implement our own video recording, using h265 gstreamer vaapi... Saving the IMU data as separate csv data. Gives me a lot more freedom than rosbags, being able to just play the video with vlc for instance. Also gives me files that are a lot easier to handle, them being 50x smaller and all...

I guess you guys are too invested in ros, but an example how to just record mkv's may broaden Realsense's reach quite a bit (e.g. with gstreamer, with just a couple of apt-installs you get all the encoding you'll ever need bypassing the licensing issues you might have). m2c

RealSenseCustomerSupport, 2020-04-15:
----------------------------------------------

@kevindehecker, thanks for the update and I am glad to hear that you were able to record the stream using gstreammer api. I'd assume that the the issue went away perhaps with smaller and fewer recording files?

This case will be closed after 7 days if there is no further assistance needed.
Thank you!

closed 2020-04-25 by RealSenseCustomerSupport