realsenseai/librealsense #6030: after calibration

nych9811, 2020-03-12:
Hi. I have questions about calibration.

{
    "device_type": "D435i",
    "imus": [
        {
            "accelerometer": {
                "bias": [
                    -0.03024129862536047,
                    -0.1153773358162399,
                    0.08777283557915627
                ],
                "scale_and_alignment": [
                    1.0221061951306303,
                    0.004108292890913816,
                    0.03809794362362086,
                    0.007725218974165806,
                    1.0194643229243248,
                    0.010445030224839558,
                    0.0027681792045815452,
                    -0.011606771744018363,
                    1.0099275933787255
                ]
            },
            "gyroscope": {
                "bias": [
                    -0.0012983011651379107,
                    -0.0012028620427323353,
                    4.881655968290166e-05
                ],
                "scale_and_alignment": [
                    1.0,
                    0.0,
                    0.0,
                    0.0,
                    1.0,
                    0.0,
                    0.0,
                    0.0,
                    1.0
                ]
            }
        }
    ]
}

These messages are from calibration.json I got after I did calibration of IMU. I can't understand what's the meaning of "bias", and "scale_alignment". Can you tell me what these mean? And also, I understood when we estimate pose, we need to subtract bias from IMU data. So I think when we estimate pose, we must subtract these values(bias) from accelerometer values I got after calibration. Please tell me what is right, and what is wrong.

MartyG-RealSense, 2020-03-12:
Page 8 of Intel's white paper document on D435i IMU calibration explains the meaning of the terms 'bias' and 'scale alignment'.

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_Depth_D435i_IMU_Calib.pdf?

I have pasted an extract of the section below for your easy reference:

[image]

The white paper recommends the link below for further reading on the basics of IMU intrinsic parameters.

https://www.st.com/content/ccc/resource/technical/document/application_note/a0/f0/a0/62/3b/69/47/66/DM00119044.pdf/files/DM00119044.pdf/jcr:content/translations/en.DM00119044.pdf

In regard to pose accuracy, page 9 of the paper recommends using a 3 axis level.

[image]

nych9811, 2020-03-13:
Dear @MartyG-RealSense 
Thanks to your reference and help.

MartyG-RealSense, 2020-03-13:
Glad I could be of help!  Do you have any other questions please?

nych9811, 2020-03-13:
Perfectly done. Thanks.

closed 2020-03-13 by nych9811