realsenseai/librealsense #6034: How to use MATLAB record and save depth and RGB video into rosbag file ?

FancyBrush, 2020-03-12:
OS: window 10
matlab: 2019b
device: SR300

      I have setup the environment, and succesfully run realsense MATLAB example. However, how to save depth and RGB video with system_time and timestamp into rosbag file? Is there any example for reference? Thanks a lot.

MartyG-RealSense, 2020-03-12:
My understanding is that scripting for the MATLAB wrapper is almost 1:1 the same as C++ code for the RealSense SDK  So if there is a C++ example for writing depth and RGB to a rosbag, in theory it should be able to be adapted for the MATLAB wrapper.

The SDK has a C++ example program for recording to a bag that may act as a starting point for creating an adaptation for the MATLAB wrapper.

https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback

FancyBrush, 2020-03-12:
Thank you for your hint. @MartyG-RealSense 
I wil try that on matlab. You could close this issue.

MartyG-RealSense, 2020-03-12:
Thanks!  Good luck   :)

closed 2020-03-12 by MartyG-RealSense