realsenseai/librealsense #6036: Regression - huge latency with RGB camera

zacosik, 2020-03-12:
| | |
|---|---|
|**librealsense**|2.33.1 RELEASE|
|**OS**|Mac OS Mojave|
|**Name**|Intel RealSense D435I|
|**Serial Number**|843112070584|
|**Firmware Version**|05.12.03.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|852323051288|
|**Firmware Update Id**|852323051288|

I'm experiencing huge latency when turning on just RGB camera with e.g. the following settings:
resolution: 640x480
frame fps: 60
color: RGB8
enable auto exposure: off
exposure: 40
auto exposure priority: off

I have tested using realsense-viewer both on build 2.33.1 and 2.32.1 and you can observe big latency with just naked eye (>100ms). On the other hand if I test at the same time with the same lightning condition and configuration on build 2.30 realsense-viewer video feed is very snappy and responsive without latency (even if my realsense firmware is using the latest one)

zacosik, 2020-03-12:
Some Errors and Warning I receive in he console:

12/03 15:44:48,477 WARNING [0x7000060cf000] (types.cpp:49) Accel Sensitivity:hwmon command 0x4f failed. Error type: No data to return (-21).
 12/03 15:44:48,479 WARNING [0x7000060cf000] (ds5-motion.cpp:463) IMU Calibration is not available, see the previous message
 12/03 15:44:51,873 WARNING [0x700006b8e000] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:51,886 WARNING [0x700006b8e000] (messenger-libusb.cpp:105) usb_request_cancel returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:51,898 WARNING [0x7000060cf000] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property Enable / disable backlight compensation
 12/03 15:44:51,909 WARNING [0x700006b8e000] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:51,922 WARNING [0x700006b8e000] (messenger-libusb.cpp:105) usb_request_cancel returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:51,934 WARNING [0x7000060cf000] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property UVC image brightness
 12/03 15:44:51,944 WARNING [0x700006b8e000] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:51,961 WARNING [0x700006b8e000] (messenger-libusb.cpp:105) usb_request_cancel returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:51,972 WARNING [0x7000060cf000] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property UVC image contrast
 12/03 15:44:51,985 WARNING [0x700006b8e000] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:52,035 WARNING [0x700006b8e000] (messenger-libusb.cpp:105) usb_request_cancel returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:52,061 WARNING [0x700006b8e000] (messenger-libusb.cpp:42) control_transfer returned error, index: 100, error: Undefined error: 0, number: 0
 12/03 15:44:52,061 WARNING [0x700006b8e000] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:52,072 WARNING [0x700006b8e000] (messenger-libusb.cpp:105) usb_request_cancel returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:52,084 WARNING [0x7000060cf000] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property Controls exposure time of color camera. Setting any value will disable auto exposure
 12/03 15:44:52,097 WARNING [0x700006b8e000] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:52,109 WARNING [0x700006b8e000] (messenger-libusb.cpp:105) usb_request_cancel returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:52,120 WARNING [0x7000060cf000] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property UVC image gamma setting
 12/03 15:44:52,133 WARNING [0x700006b8e000] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:52,144 WARNING [0x700006b8e000] (messenger-libusb.cpp:105) usb_request_cancel returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:52,155 WARNING [0x7000060cf000] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property UVC image hue
 12/03 15:44:52,169 WARNING [0x700006b8e000] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:52,656 WARNING [0x700006b8e000] (messenger-libusb.cpp:105) usb_request_cancel returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:52,668 WARNING [0x7000060cf000] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property UVC image saturation setting
 12/03 15:44:52,681 WARNING [0x700006b8e000] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:52,693 WARNING [0x700006b8e000] (messenger-libusb.cpp:105) usb_request_cancel returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:52,704 WARNING [0x7000060cf000] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property UVC image sharpness setting
 12/03 15:44:52,717 WARNING [0x700006b8e000] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:52,727 WARNING [0x700006b8e000] (messenger-libusb.cpp:105) usb_request_cancel returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:52,740 WARNING [0x7000060cf000] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
 12/03 15:44:52,753 WARNING [0x700006b8e000] (messenger-libusb.cpp:42) control_transfer returned error, index: 100, error: Undefined error: 0, number: 0
 12/03 15:44:52,753 WARNING [0x700006b8e000] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 81 error: Undefined error: 0, number: 0
[... cut ...]
 12/03 15:44:52,788 WARNING [0x700006b8e000] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:52,790 WARNING [0x700006b8e000] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Undefined error: 0, number: 0
 12/03 15:44:52,800 WARNING [0x700006b8e000] (messenger-libusb.cpp:105) usb_request_cancel returned error, endpoint: 81 error: Undefined error: 0, number: 0
 12/03 15:44:52,811 WARNING [0x7000060cf000] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property Enable / disable auto-white-balance

MartyG-RealSense, 2020-03-12:
I wonder if the latency is partially due to the large number of continuous error messages that are being generated whilst the program is running, giving more for the computer to think about.

Could you try running at 30 FPS please to see if you still get the errors at that FPS speed?

zacosik, 2020-03-12:
When setting frame rate to 30fps on any build version everything do works without noticable latency. Starting with build 2.31.0 it breaks when rgb has 60fps (works good on 2.30.0). In all builds realesense-viewer shows in top status bar we are running at 60fps. The errors mentioned above are not displayed in console continueaslly. Regarding calibrating IMU I'm more interested with video feed and AFAIK imu api is still not available on MacOS?

MartyG-RealSense, 2020-03-12:
Yes, MacOS support for D435i is scheduled but not yet implemented, according to the SDK Releases page. 

If the RGB works in 30 FPS mode but breaks at 60 FPS, that leads my thinking towards a USB port issue.  In the original RealSense R200 stereo cameras in 2015, there used to be an issue where applications could freeze if 60 FPS was used because the USB port was saturated with data, and the fix was to drop to 30 FPS.

zacosik, 2020-03-12:
Yes as mentioned I don't have problem on the same device and setup when using build 2.30.0 (rebuilt all versions between 2.30-2.33 and tested today). At least if to believe to realsense-viewer top status bar that indeed is displaying at 60fps. Previously also didn't have problem with my python sample app get RGB video feed at 60fps. That was back before when I reported that  manually adjusting exposure on MacOS is not working.

MartyG-RealSense, 2020-03-12:
I ran tests with your settings in SDK 2.33.1's version of the RealSense Viewer with a D435i in Windows and could not replicate the problem.

So the problem is centered around using manual exposure, and it works at 60 FPS if you have Auto Exposure enabled, please?

zacosik, 2020-03-12:
I managed only few times (with older firmware) with auto exposure get more than 30fps (I guess not enough lightning condition). Just tested auto exposure enabled on 2.33 and even though lots of light still provides only 30 fps. Another think I tested with manual exposure. When I set exposure: 120,  I get not noticable latency and around 45fps, but at exposure:100 and lower already can notice big latency (just by waving my hands with naked eye). I will test again tomorrow on another machine with windows 10.

MartyG-RealSense, 2020-03-12:
Ok, thank you very much.  Good luck with your tests.  Ultimately if you continue to have problems, I can give this discussion a "Bug" label so that the RealSense developers can track it officially.

zacosik, 2020-03-14:
Tested on another computer with also MacOS Catalina with 2 different cables - same thing and big latency. When tested on Windows 10 with the same cables and setup then it's working correctly with 2.33.1.

MartyG-RealSense, 2020-03-14:
Ok, thank you very much.   I will label this discussion as "Bug" for the RealSense developers to track and leave it open.

xEnVrE, 2020-03-18:
I can confirm the same issues, i.e. huge latency and also freezes, on Ubuntu 18.04, kernel `5.3.0-40-generic` using the version `2.32.1` from `apt` with default options (i.e. just plugging the camera, that is a `D415`, and running the `realsense-viewer`, i.e. a 30 fps, ont 60). 

I solved the problems by compiling from source and enabling `FORCE_RSUSB_BACKEND`. With this option enabled, the stream is very snappy and responsive. However, as reported in #5310, this option produces huge CPU usage in the order of `300%`.

zacosik, 2020-04-12:
Just to give an update after testing on 2.34. With this version is even worse even if I set fps to just 30fps I get around 2 seconds latency and that's completely unusable. Build with both -DFORCE_RSUSB_BACKEND true and false. Bellow the build I used:

cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -DBUILD_PYTHON_BINDINGS=true -DBUILD_PYTHON_DOCS=false -DBUILD_CV_EXAMPLES=true -DBUILD_CV_KINFU_EXAMPLE=true -DFORCE_RSUSB_BACKEND=false -DCMAKE_PREFIX_PATH=/Users/zaco/Developer/compile/installation/OpenCV-4.3.0

MartyG-RealSense, 2020-04-12:
@zacosik  There is an already reported issue with timing n 2.34.0.  A fix has been created and is awaiting final approval for merging into the SDK.

https://github.com/IntelRealSense/librealsense/issues/6189#issuecomment-607918458

@dorodnic  Can you take a look at the pull that is awaiting merging with an approving review please?   Thanks!

https://github.com/IntelRealSense/librealsense/pull/6150

jmachuca77, 2020-04-15:
@MartyG-RealSense is there a PR somewhere I can put into my build for now?

MartyG-RealSense, 2020-04-15:
@jmachuca77   If you mean the pull I mentioned earlier, it is here:

https://github.com/IntelRealSense/librealsense/pull/6150

MartyG-RealSense, 2020-05-29:
Closing this case because there was not a report of the same problem of huge RGB camera latency from other users in subsequent weeks.  The information in the case will remain findable with web search in case somebody else has the same issue in future.

closed 2020-05-29 by MartyG-RealSense