realsenseai/librealsense #6038: my Realsense D435i can not obtain pose data

hwenhui1997, 2020-03-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | 05.12.3.0 | 
| Operating System & Version |  ubuntu 16.04  | 
| Kernel Version (Linux Only)    | 4.15.0-88-generic                                   | 
| SDK Version                            |  RealSense 2.0           | 
| Language                            |  c++                        | 

### Issue Description
when I run example: ` ./rs-pose `
error occurs :
```
RealSense error calling rs2_pipeline_start_with_config(pipe:0x2614c00, config:0x25f81f0):
    Couldn't resolve requests
```
when I run program include              
```
     pipe_config.enable_stream(RS2_STREAM_POSE,RS2_FORMAT_6DOF);
     rs2::pipeline_profile profile=pipe.start(pipe_config);
```
error occurs too:
```
RealSense error callingrs2_pipeline_start_with_config(pipe:0x18aa7b0, config:0x188e1e0):
Couldn't resolve requests
```
but Gyro and Accel can return data ;

MartyG-RealSense, 2020-03-12:
**rs-pose** is a sample program for the T265 Tracking Camera.  The D435 depth camera models do not have support for pose data built in.  It is possible to get pose with D435 though by using OpenCV or ROS.

A user who was using a robot arm for bin picking got pose with a D435 with an OpenCV SolvePnP algorithm.

IntelRealSense/librealsense#3569 (comment)

The link below, meanwhile, has an approach to getting object pose with D435 and ROS using 'inference'.

https://github.com/NVlabs/Deep_Object_Pose/issues/10

hwenhui1997, 2020-03-12:
> **rs-pose** is a sample program for the T265 Tracking Camera. The D435 depth camera models do not have support for pose data built in. It is possible to get pose with D435 though by using OpenCV or ROS.
> 
> A user who was using a robot arm for bin picking got pose with a D435 with an OpenCV SolvePnP algorithm.
> 
> #3569 (comment)
> 
> The link below, meanwhile, has an approach to getting object pose with D435 and ROS using 'inference'.
> 
> [NVlabs/Deep_Object_Pose#10](https://github.com/NVlabs/Deep_Object_Pose/issues/10)

think you, but how can I use IMU data to estimate camera pose.

MartyG-RealSense, 2020-03-12:
There is a C++ SDK sample program called **Motion** that is listed as D435i compatible.  Its description says it "demonstrates how to use data from gyroscope and accelerometer to compute the rotation of the camera".  would this be helpful to you please?

https://dev.intelrealsense.com/docs/rs-motion

MartyG-RealSense, 2020-03-22:
This case will be closed after 7 days from the date of writing this if there are no further responses.   Thanks!

closed 2020-03-28 by MartyG-RealSense