realsenseai/librealsense #6039: Acquiring and processign frames in different threads

pascoa-pand, 2020-03-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | (05.12.01.00) | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| Language                            |  Python                          | 

### Issue Description

Hello,

I am trying to create a program that acquires frameset in one thread (with RGB and Depth enabled) and processes those frameset in a second thread

This is what I've implemented:

```python

import pyrealsense2 as rs
import time
import threading
from threading import Thread
import numpy as np
import multiprocess
from queue import Queue
import time

class CameraProcess(Thread):
    def __init__(self,frameQueue):
        Thread.__init__(self)
        self.frameQueue=frameQueue
    def run(self):
        pass
    def getProcessedFrame(self):
        frameset=self.frameQueue.get()
        proc=self.processFrame(frameset)
        return proc
    def processFrame(self,frameset):
        '''
        Function that performs processing to the frameset, such as alignment, depth post-processing and others
        '''
        time.sleep(2)
        return frameset

class CameraStreamer(Thread):
    depthShape = [640,480]
    rgbShape=[640,480]
    def __init__(self,frameQueue):
        Thread.__init__(self)
        
        self.pipeline = rs.pipeline()
        self.config = rs.config()
        self.config.enable_stream(rs.stream.color, self.rgbShape[0], self.rgbShape[1], rs.format.bgr8, 30)
        self.config.enable_stream(rs.stream.depth, self.depthShape[0], self.depthShape[1], rs.format.z16, 30)
        self.frameQueue=frameQueue
        
    def run(self):
        self.profile = self.pipeline.start(self.config)
        self.loopFrames()
        
    def loopFrames(self):
        t1=time.perf_counter()
        while True:
            frameset = self.pipeline.wait_for_frames()
            self.frameQueue.put(frameset)
            print(f"Looper Thread | Acquired frame {frameset.frame_number} in {np.round((time.perf_counter()-t1),3)}")
            t1=time.perf_counter()

frameQueue=Queue()

#start threads
cameraStreamer=CameraStreamer(frameQueue)
cameraProcess = CameraProcess(frameQueue)

#start threads
cameraProcess.start()
cameraStreamer.start()

nframes=0
iniTime=time.perf_counter()
t1=iniTime
while True:
    fs = cameraProcess.getProcessedFrame()
    nframes+=1
    print(f"Time since last frame {np.round((time.perf_counter()-t1)*1000,3)}ms, qsize {cameraProcess.frameQueue.qsize()},fps: {np.round(nframes/(time.perf_counter()-iniTime),3)}")
    t1=time.perf_counter()
c.pipeline.stop()

```

I believe my issue with this solution has to due with the frameQueue. When I put the frameset in queue, the cameraStreamer thread blocks while time.sleep() isn't finished. If I decide to switch the frameset for a number (for instance) this doesn't happen.

Do you have any idea on what might be happening? Is there a better way to implement such a solution?

MartyG-RealSense, 2020-03-12:
I must start with the disclaimer that there are members of the RealSense team with more advanced programming knowledge than myself.  In the interests of giving you a speedy response, I will try to assist though!

Your use of the **frame_queue** primitive to send frames from thread to thread seems to be the appropriate way to do things.  If you have not seen it already, I recommend looking at the frames and threads advice in Intel's documentation to see if they offer a useful suggestion for improving the efficiency of your script.

https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_lifetime.md#frames-and-threads

kafan1986, 2020-03-12:
@pascoa-pand I am not sure about the python api and how much functionality they expose but the if I remember correctly, underlying sdk uses frame queue of size 1. Based on my experience it is not possible to create your own queue of frameset and try to use it at a later stage. There is no easy deep cloning functionality for frameset. Although through the SDK, you can increase the internal queue size and if you are consuming the frameset fast enough then you can use it directly in your functionality.

Also, may I know why are you intend on using regular multi-threading on Python. There won't be much benefit as due to global interpreter lock, you will end up using single core even when using multithreading. You can spawn processes to bypass the limitation but you lose the benefit of threads ("light weight" process). Maybe the underlying SDK being in native bypasses the Python's global interpreter lock. I am not sure though.

To create a actual queue that can be used later, for my application I have used OpenCV that converts the frameset to individual pairs of Mat.

MartyG-RealSense, 2020-03-12:
@kafan1986   Thanks so much for your help with this case and others!

pascoa-pand, 2020-03-12:
Thank you both!

> 
> 
> @pascoa-pand I am not sure about the python api and how much functionality they expose but the if I remember correctly, underlying sdk uses default frame queue of size 1. Based on my experience it is not possible to create your own queue of frameset and try to use it at a later stage, as there is no easy deep cloning functionality for frameset. Although through the SDK, you can increase the internal queue size and if you are consuming the frameset fast enough then you can use it directly in your functionality.

Is it mandatory to do a deep cloning when using threading? Because the frameset memory reference is created by the first thread and put in a queue, and that reference will only be used again by the processing thread. Before I had tried to save 10 frameset in a queue, and when I read them they all had the original frameset.frame_number, so I assumed they were all being stored in memory

> 
> Also, may I know why are you intend on using regular multi-threading on Python. There won't be much benefit as due to global interpreter lock, you will end up using single core even when using multithreading. You can spawn processes to bypass the limitation but you lose the benefit of threads ("light weight" process). Maybe the underlying SDK being in native bypasses the Python's global interpreter lock. I am not sure though as I haven't used python in quite some time.

Maybe I'm wrong, but I thought this could avoid GIL as the only interaction between the two threads would be through a thread-safe queue. While the first thread is reading and putting frameset N in the queue, the second thread (processing) would be getting a previous frameset. What's funny and I don't understand is that it is the time.sleep() method in the processing thread that is slowing the streaming thread. 

> 
> To create a actual queue that can be used later, for my application I have used OpenCV that converts the frameset to individual pairs of Mat.

Good solution but doesn't seem to work in my scenario, because converting the Frameset to Mats and then pass on to a queue would mean that the heavy processing (alignment and depth post-processing) was already done

pascoa-pand, 2020-03-12:
> 
> 
> I must start with the disclaimer that there are members of the RealSense team with more advanced Librealsense programming knowledge than myself. In the interests of giving you a speedy response, I will try to assist though!
> 
> Your use of the **frame_queue** primitive to send frames from thread to thread seems to be the appropriate way to do things. If you have not seen it already, I recommend looking at the frames and threads advice in Intel's documentation to see if they offer a useful suggestion for improving the efficiency of your script.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_lifetime.md#frames-and-threads

Thank you Marty! I'll have a look to rs.frame_queue() a little bit better, but as far as I understood (I'm not very familiar with c++) I have to:
1) call wait_for_frame in the rs.frame_queue object (frame 'f')
2) call enqueue 'f' to rs.frame_queue ?

I'm not very familiar with c++ so this method was not very clear to me:
```
dev.start([](rs2::frame f){
    q.enqueue(std::move(f)); // enqueue any new frames into q
});
´´´

MartyG-RealSense, 2020-03-12:
@pascoa-pand   I will defer your question about frame_queue to the kind assistance of @kafan1986  as they can answer this particular programming subject better than I can!  Thanks again @kafan1986

kafan1986, 2020-03-12:
@pascoa-pand Although I am not sure but I believe the frameset memory is allocated by the SDK and by default only 1 set of frameset is allocated. The new frameset data is not placed into this allocated memory till you call pipeline.wait_for_frames, upon which the system believes you are not  using the previous frameset and replaces it with the latest frameset asap. But as you mentioned you have observed that adding frameset to the queue and after some considerable time later fetching them, you could still see the frameset were having their original frame_number, so maybe I am wrong but still it is better to confirm by looking at the actual data in the individual frames and check whether they are actually different or not.

Regarding Python and GIL, no communicating via queue will have no impact on GIL. All the threads  spawned will continue to work on the single core. There is a different python module which spawns processes but brings additional overhead and the communication among processes is done through message passing. The processes thus spawned may or may not work on the same physical CPU core, depends on the OS scheduling.

Regarding your doubt about Mat and alignment. I am not sure about the way you are using depth and color data later on, but quite frequently you don't need to align beforehand. You can individually process frames and depending on the result you can later combine result. Now alignment coordinates for limited number of pixels is required e.g. if object recognition is done, you can simply perform a rectangle overlap logic and thus only 4 corner point pixel alignment is only required. If you need the alignment of whole frames even that is quite easy from Mat. Look at the respective SDK code and change the few lines of code.

tgpsantos, 2020-03-15:
I'm now exploring the rs.frame_queue object. Is there any way to get the number of frameset in the queue? Or at least now if the queue is empty?

tgpsantos, 2020-03-15:
I've decided to change my approach and tried to combine the rs.frame_queue() with getting frames from the pipeline, but I'm getting an error: 

`null pointer passed for argument "frame_ref"`
This happens when I try to get data from my depthFrame in `depth_image = np.asanyarray(depthFrame.get_data()).` 
If I print the depthFrame right before I get a `<pyrealsense2.pyrealsense2.depth_frame object at 0x000002CE048D57F0>`. 

Here is the reproducible code:

```python 
import pyrealsense2 as rs
import time
import threading
import numpy as np
import multiprocess
import queue
import time

class CameraStreamer(threading.Thread):
    depthShape = [640,480]
    rgbShape=[640,480]
    def __init__(self,frameQueue):
        threading.Thread.__init__(self)
        threading.Thread.setName(self,name='CameraStreamer Thread')
        self.pipeline = rs.pipeline()
        self.config = rs.config()
        self.config.enable_stream(rs.stream.color, self.rgbShape[0], self.rgbShape[1], rs.format.bgr8, 30)
        self.config.enable_stream(rs.stream.depth, self.depthShape[0], self.depthShape[1], rs.format.z16, 30)
        self.frameQueue = frameQueue
        self.align=rs.align(rs.stream.color)
    def run(self):
        self.profile = self.pipeline.start(self.config)
        self.iniTime=time.perf_counter()
        self.loopFrames()
        self.pipeline.stop()
        print(f"CameraStreamer | Ended run in {np.round((time.perf_counter()-self.iniTime),1)} seconds")
    def loopFrames(self):
        t1=time.perf_counter()
        n=0
        while n<250:
            t_w=time.perf_counter()
            frameset = self.pipeline.wait_for_frames()
            self.frameQueue.enqueue(frameset)
            print(f"CameraStreamer | Acquired frame {frameset.frame_number} in {np.round((time.perf_counter()-t1),3)}")
            t1=time.perf_counter()
            n=frameset.frame_number
        
class FrameProcessor(threading.Thread):
    def __init__(self,queueFrames):
        threading.Thread.__init__(self)
        threading.Thread.setName(self,name='FrameProcessor Thread')
        self.queueFrames=queueFrames
    def run(self):
        self.iniTime=time.perf_counter()
        self.processFrames()
        print(f"FrameProcessor | Ended run in {np.round((time.perf_counter()-self.iniTime),1)} seconds")
    def processFrames(self):
        while True:
            frame=self.queueFrames.wait_for_frame()
            print(f"FrameProcessor | Going to process frame #{frame.frame_number}")
            depthFrame = frame.as_depth_frame()
            colorFrame = frame.as_video_frame()
            depth_image = np.asanyarray(depthFrame.get_data())
            self.queueOut.put(processedFrame)
            
frameQueue=rs.frame_queue(capacity=10,keep_frames=True)
#instantiate objects threads
cameraStreamer = CameraStreamer(frameQueue)
frameProcessor = FrameProcessor(frameQueue)
#start threads
cameraStreamer.start()
frameProcessor.start()
```

MartyG-RealSense, 2020-03-22:
@tgpsantos  Do you still need your last question answered please?  Or is it okay to close this case.   Thanks!

closed 2020-04-12 by MartyG-RealSense

PierreLouisT, 2022-01-14:
Hi ! I am working on multiprocessing as well, I would like put data frame in a queue. However I get this type error : 
**TypeError: cannot pickle 'pyrealsense2.pyrealsense2.BufData' object**

You can see my code below : 
`import multiprocessing as mp
import os
import threading
import time
import numpy as np
import pyrealsense2.pyrealsense2 as rs
import cv2
import sys
import math
from timecontext import Timer 
current = (0,0)

class Camera(): 
    

    def __init__(self): 
       
        self.pipeline = rs.pipeline()
        self.config = rs.config()
        self.cfg = self.pipeline.start(self.config)
        self.dev = self.cfg.get_device()
        self.colorizer = rs.colorizer()

        #self.align_to = rs.stream.depth

    def param(self): 
        return self.pipeline, self.config, self.cfg, self.dev, self.colorizer
        
        
        
    def read_camera():
    print("lecture de la caméra")

    pipeline, config, cfg, dev, colorizer = Camera().param()
    print(dev.get_info(rs.camera_info.serial_number))
    print(dev.get_info(rs.camera_info.name))
    print(dev.get_info(rs.camera_info.firmware_version))
    print(dev.get_info(rs.camera_info.product_id))
    print(dev.get_info(rs.camera_info.physical_port))

def acquisition_frames(framelist):
    print("Acquisition des frames")
    pipeline, config, cfg, dev, colorizer = Camera().param()
    
    
    
    while framelist.full() :
        time.sleep(0.1)
        print("buffer rempli")
    frames = pipeline.wait_for_frames()
    frames.keep()
    dataframe = frames.get_data()
    framelist.put(dataframe)
    

if __name__ == '__main__':
    print("Main Process : définition des process")
    framelist = mp.Queue()
    process1 = mp.Process(target = read_camera)
    process2 = mp.Process(target = acquisition_frames, args = (framelist,))
    print("lancement du process 1")
    start = time.time()
    process1.start()
    process1.join()
    end = time.time() - start
    print("temps realisé")
    print(end)
    print("fin du process 1")       
    print("début du process 2")
    start = time.time()
    process2.start()
    process2.join()
    end = time.time() - start
    print("temps realisé")
    print(end)
    print("fin du process 2")    `

I tried keep() fonction but it didn't work that way.. Do you have any idea ? I also viewed that it is possible to get a server to put the data but it is out of context in my case unfortunately. 
Thanks a lot for all the help you could bring me :)