realsenseai/librealsense #6045: How to get 6dof for D435i ?

raghav123456, 2020-03-12:
hey ,
I am working on a project related to odometry for which i need to record the 6DOF (Degree of Freedom) from intel D435i. So can someone help me that how to get this data. 
Thankyou.

MartyG-RealSense, 2020-03-12:
The RealSense T265 Tracking Camera has built-in support for accessing 6-DOF pose data but the D435 depth camera models do not.

If you want to get pose on a D435 (non-IMU), there are methods for doing so with OpenCV (Solve PNP algorithm) or ROS (Inference).

https://github.com/IntelRealSense/librealsense/issues/6038#issuecomment-598089815

There is also a D435i-compatible sample program called **rs-motion** that"demonstrates how to use data from gyroscope and accelerometer to compute the rotation of the camera".

https://dev.intelrealsense.com/docs/rs-motion
getting the D435i's rotation from the IMU.

raghav123456, 2020-03-13:
Thanks fro you reply @MartyG-RealSense 
Is their any document or Github material available regarding (Solve PNP algorithm) and ROS (Inference) which i can prefer to move foward?

MartyG-RealSense, 2020-03-13:
A RealSense community member who was using the SolvePNP algorithm recommended this guide:

https://www.learnopencv.com/head-pose-estimation-using-opencv-and-dlib/

In regard to ROS, I couldn't find a good example of using RealSense to get pose with inference other than the project I linked to before.  As a potential alternative approach though, a RealSense D435 user also recommended a program called OpenPose and its ROS wrapper:

https://github.com/stevenjj/openpose_ros

raghav123456, 2020-03-14:
I want to use d453i for making trajectory as t265 can do can it be possible to use it for trajectory.

MartyG-RealSense, 2020-03-14:
I couldn't find an easy way to track a trajectory with D435i.  Someone did render a trajectory with a D435i in ROS though through using the **RViz** program.

https://github.com/rpng/R-VIO/issues/8

Also using ROS, Intel published an article on setting up SLAM with D435i, though I'm not sure that is what you are looking for.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

raghav123456, 2020-03-14:
@MartyG-RealSense  
I am looking for rotation and translation matrix . like will it support RS2_STREAM_POSE (as t265 supports to get rotation and translation)

MartyG-RealSense, 2020-03-14:
The last I heard about it was around a year ago, in a comment where @dorodnic explained the difficulties of implementing SLAM for D435i.

https://github.com/IntelRealSense/librealsense/issues/3070#issuecomment-454689295

In an earlier discussion, Dorodnic recommended looking at the D435i SLAM example (which I highlighted earlier) and also the **RealSense Kinfu** example linked to below.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/kinfu

MartyG-RealSense, 2020-03-22:
This case will be closed after 7 days from the date of writing this if there are no further responses.   Thanks!

closed 2020-03-28 by MartyG-RealSense