realsenseai/librealsense #6046: LINK error when building rs-record-playback.sln

FancyBrush, 2020-03-12:
OS: windows 10
VS: 2019 community
device: SR300
CMake: 3.17
       Very grateful for attention :

       It turns out that using matlab to acquire bag file is less efficient than using C++, so I decided to 
try saving bag file with a C++ example. 
      Here is the detailed procedure I went through : 
      1. Navigated to "/librealsense-master/examples/record-playback";
      2. Configure CMakelist.txt in this folder with BUILD_GRAPHICAL_EXAMPLES set to TRUE
      3. After configuring, I open the "RealsenseExamplesRecord-Playback.sln"
      4. Add additional include and library which are librealsense2 and GLFW (did I miss something?)
      5. Then I build "RealsenseExamplesRecord-Playback.sln", but receive LINK errors:
          I think I did not miss head and library files, maybe it is that the problem?
[image]
 
      Thank you in advance !

MartyG-RealSense, 2020-03-12:
If you are using Windows 10 then there is a pre-built executable version of this example in the SDK folder so that it does not have to be compiled.  Its folder location on Windows should be:

**C: > Program Files (x86) > Intel RealSense SDK 2.0 > Tools**

and the filename is **rs-record-playback**.

[image]

MartyG-RealSense, 2020-03-12:
Guidance about methods for setting up the RealSense dependencies in your own project can be found in the link below:

https://github.com/IntelRealSense/librealsense/issues/4277

dorodnic, 2020-03-12:
Hi @FancyBrush 
You are going about CMake wrong - instead of running CMake on individual CMakeList.txt files, you need to run CMake on the top level one to general single solution for the SDK.
At that level, you can enable / disable wrappers and everything should be building no problem

FancyBrush, 2020-03-13:
Thank you for your suggestion!
      1.  I tried the executable rs-record-playback, but to find that I click the "record" button, the program crashed.
      2.  Did you mean CMake the librealsense/CMakelist.txt ?
      New to CMake, thank you for your patience...

MartyG-RealSense, 2020-03-13:
@FancyBrush   I believe what is being referred to by @dorodnic is building Librealsense with CMake, and using 'flags' in the build instruction to specify exactly which features you want to include ('true') and which you want to not be included in the build ('false').

A list of the CMake build customization flags can be found here:

https://github.com/IntelRealSense/librealsense/wiki/Build-Configuration

If you scroll down the list, you can see references to flags about building 'bindings'.  This refers to adding specific types of wrapper to your Librealsense build (Python, MATLAB, C#, NodeJS, etc).  

[image]

Re-reading your original question though, it looks as though you decided not to use the MATLAB wrapper and to use C++ instead.  So you should not need to add a wrapper if you only want to program in C++.

So, the first step then would be to build the basic RealSense SDK with support enabled for examples.  It would be a good idea to add support for "graphical" examples at the same time (programs such as RealSense Viewer and Depth Quality Tool that use OpenGL graphics).  So the CMake build instruction to build the Librealsense SDK with examples support may be like this:

cmake ../ -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true

If you did decide later to build Librealsense with the MATLAB wrapper included then the CMake installation instruction may look like this:

cmake ../ -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_MATLAB_BINDINGS=true

FancyBrush, 2020-03-14:
Thank you! I will try to CMake another time.

dorodnic, 2020-03-14:
Also, if you don't want to deal with CMake, you can install the SDK (from Releases page here). It comes with most examples already pre-build, as well as clean Visual Studio SLN that does not require CMake.

FancyBrush, 2020-03-15:
@dorodnic Thank you for your suggestion. I tried pre-build rs-record-playback.exe. But I'm so sad that whenever I press "record" button, the program crash and the window disappear.

dorodnic, 2020-03-15:
It is trying to write near the executable, probably permission issue with program files. You can try copying the folder to my documents or some place else it can write to

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

MartyG-RealSense, 2020-03-22:
This case will be closed after 7 days from the date of writing this if there are no further responses.   Thanks!

closed 2020-03-28 by MartyG-RealSense