realsenseai/librealsense #6049: Black RGB/IR texture with rs::gl::pointcloud

tom-garford, 2020-03-13:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (05.12.01.00) | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  (4.15.0-88-generic)                                         | 
| Platform                                 | i7, Nvidia RTX 2060, also Jetsons. |
| SDK Version                            |  { 2.31.0}                          | 
| Language                            |  {C++ }                          | 

Hello, 
I was encouraged to see how fast the rs-gl example rendered and colourised pointclouds. However, I would like to do the same but use either the RGB or IR stream as texture instead. 

I can use both of these as textures in the realsense viewer and RGB on the rs-pointcloud example but when I tweak the rs-gl code I get a "black" pointcloud when using either stream. I notice that the proj matrix in the rs-gl sample changes depending on camera used (d415/ d435) but not when streams are changed for texture. 

Graphical glXXX parameters seem similar in rs-gl function compared to the draw_pointcloud function used in rs-pointcloud in the example.hpp script. I have read previous posts about black pointclouds: 
[https://github.com/IntelRealSense/librealsense/issues/3213](url)
[https://github.com/IntelRealSense/librealsense/issues/3788](url)
    but none of them seem to be using librealsense2-gl functions. 

Please let me know if this is indeed possible (or is just a rendering problem) and if there is anything else I could try. My code is identical to rs-gl except: 
` rs2::frame  infrared;

    // Enable pipeline
    rs2::pipeline pipe;
    rs2::config cfg; 
    // For this example, we are only interested in depth:
    int x = 848, y = 480, f = 30;
    cfg.enable_stream(RS2_STREAM_DEPTH, x, y, RS2_FORMAT_Z16, f);
    cfg.enable_stream(RS2_STREAM_INFRARED, x, y, RS2_FORMAT_Y8, f);

    pipe.start(cfg);

    while (app) // Application still alive?
    {
        // Any new frames?
        rs2::frameset frames;
        if (pipe.poll_for_frames(&frames))
        {
	        auto infrared = frames.first(RS2_STREAM_INFRARED);

            infrared = upload.process(infrared);

            pc.map_to(infrared);

            auto depth = frames.get_depth_frame();

            depth = upload.process(depth);

            points = pc.calculate(depth);
        }

        // Populate projection and view matrices based on user input
        prepare_matrices(app_state, 
                         app.width(), app.height(), 
                         proj_matrix, view_matrix);

        // We need to get OpenGL texture ID for pointcloud texture
        uint32_t texture_id = 0;
        // First, check if the frame is already a GPU-frame
        if (auto gpu_frame = infrared.as<rs2::gl::gpu_frame>())
        {
            // If it is (and you have passed window for sharing to init_processing
            // you can just get the texture ID of the GPU frame
            texture_id = gpu_frame.get_texture_id(0);
        }
        else
        {
            // Otherwise, we need to upload texture like in all other examples
            tex.upload(infrared);
            texture_id = tex.get_gl_handle();
        }
`

And is successfully compiled with: 

`
g++ -std=c++11 gl-ir.cpp -I../librealsense/examples/ -lrealsense2 -lrealsense2-gl -lglfw3 -lglut -lGL -lGLU -o glir
`
[image]

dorodnic, 2020-03-13:
Hi @tom-garford 
This should absolutely be possible because this is what we are doing in the RealSense Viewer.
Same result in case of RGB stream?

tom-garford, 2020-03-13:
Hi @dorodnic 

Yes, exactly the same. I had wondered if it was a projection issue so tried IR but with same results. Coloriser runs as expected with original rs-gl code

dorodnic, 2020-03-15:
Found the bug :)
Change `auto infrared = frames.first(RS2_STREAM_INFRARED);` to `infrared = frames.first(RS2_STREAM_INFRARED);` - you are not assigning to external `infrared` variable

tom-garford, 2020-03-16:
Thanks very much @dorodnic, it worked!

Works too with RGB, however had to make sure format was RGB8. Thanks for your help!
[image]

closed 2020-03-16 by tom-garford