realsenseai/librealsense #6050: Optical Filters for Realsense D415

ToonTielen, 2020-03-13:
Dear Support,

I am looking for application of a longpassfilter in bright light conditions, to be less sensitive for variation in light conditions and have a higher response of the emitter. I was reviewing the whitepaper on `[optical filters](https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400)`  and the filters indicated in the graph below seem suitable (LP1 (gray) and LP2). However I am having trouble finding a product which matches this spectral response. Do you have any advice on which product to use?
[image]

MartyG-RealSense, 2020-03-13:
The 400 Series cameras do not have an IR-Cut filter on them and so do not have limitations on the nm light ranges that they can see.  The T265 Tracking Camera has a built-in IR-Cut filter to filter out the higher bands (e.g 900-940), so that it does not experience interference from 400 Series depth cameras when the depth and tracking cameras are used together.

RealSense community member @AntoineWefit provides a suggestion for a product in this link:

https://github.com/IntelRealSense/librealsense/issues/4082#issuecomment-496823480

If you are seeking to reduce fluctuations in FPS and have auto-exposure enabled, an alternative to filters to try may be to disable a setting called Auto-Exposure Priority (found in the RGB settings in the RealSense Viewer).  This enables a constant FPS to be enforced.

ToonTielen, 2020-03-17:
@MartyG-RealSense 
thanks for the response, but @AntoineWefit suggests to use a regular long pass filter or band pass filter. However in the [whitepaper optical filters for Intel Realsense](https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400) suggests to use a filter which is acting as a neutral density filter in the VIS spectrum and doesn't filter at all in the NIR, as shown in the image above, I am looking for a specific filter which shows that same response.

MartyG-RealSense, 2020-03-17:
The CTO of the RealSense Group at Intel (agrunnnet) provides links for providers of ND filters here:

https://github.com/IntelRealSense/librealsense/issues/2875#issuecomment-446793681

agrunnet, 2020-03-17:
Yes Astra Products is where we got the variable ND filters/IR-pass filters.

jnsweets, 2020-03-17:
They do not show them on their website but you can contact Astra about visible attenuating/IR-pass filters. Here's the link to their standard IR-pass filter material:

https://astraproducts.com/info-acrylic-nir-filters.asp

ToonTielen, 2020-03-18:
Ok, thanks for the information. I will contact Astra!

ToonTielen, 2020-03-18:
Thanks for the comments, I will get in contact with Astra

closed 2020-03-18 by ToonTielen