realsenseai/librealsense #6051: Link files missing????

jiapei100, 2020-03-13:
Just failed to build librealsense on my laptop with the following ERROR messages:
I'm using **PCL 1.10.0**
and **VTK 9.0.0**.

```console
[100%] Built target rs-rosbag-inspector
CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o: In function `Load_PCDFile()':
....../librealsense/wrappers/pcl/pcl-color/rs-pcl-color.cpp:267: undefined reference to `pcl::visualization::PCLVisualizer::PCLVisualizer(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)'
....../librealsense/wrappers/pcl/pcl-color/rs-pcl-color.cpp:270: undefined reference to `pcl::visualization::PCLVisualizer::setBackgroundColor(double const&, double const&, double const&, int)'
CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o: In function `Load_PCDFile()':
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:81: undefined reference to `pcl::visualization::PointCloudGeometryHandlerXYZ<pcl::PointXYZRGB>::PointCloudGeometryHandlerXYZ(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> const> const&)'
CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o: In function `Load_PCDFile()':
....../librealsense/wrappers/pcl/pcl-color/rs-pcl-color.cpp:274: undefined reference to `pcl::visualization::PCLVisualizer::setPointCloudRenderingProperties(int, double, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)'
....../librealsense/wrappers/pcl/pcl-color/rs-pcl-color.cpp:276: undefined reference to `pcl::visualization::PCLVisualizer::initCameraParameters()'
....../librealsense/wrappers/pcl/pcl-color/rs-pcl-color.cpp:281: undefined reference to `pcl::visualization::PCLVisualizer::spin()'
CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o: In function `pcl::visualization::PointCloudColorHandler<pcl::PointXYZRGB>::getColor(vtkSmartPointer<vtkDataArray>&) const':
/usr/local/include/vtk-9.0/vtkSmartPointer.h:137: undefined reference to `vtkSmartPointerBase::operator=(vtkObjectBase*)'
/usr/local/include/vtk-9.0/vtkSmartPointer.h:35: undefined reference to `vtkSmartPointerBase::~vtkSmartPointerBase()'
/usr/local/include/vtk-9.0/vtkSmartPointer.h:35: undefined reference to `vtkSmartPointerBase::~vtkSmartPointerBase()'
CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o: In function `pcl::visualization::PointCloudColorHandler<pcl::PointXYZRGB>::getColor() const':
/usr/local/include/vtk-9.0/vtkSmartPointer.h:62: undefined reference to `vtkSmartPointerBase::vtkSmartPointerBase()'
/usr/local/include/vtk-9.0/vtkSmartPointer.h:137: undefined reference to `vtkSmartPointerBase::operator=(vtkObjectBase*)'
/usr/local/include/vtk-9.0/vtkSmartPointer.h:35: undefined reference to `vtkSmartPointerBase::~vtkSmartPointerBase()'
/usr/local/include/vtk-9.0/vtkSmartPointer.h:35: undefined reference to `vtkSmartPointerBase::~vtkSmartPointerBase()'
/usr/local/include/vtk-9.0/vtkSmartPointer.h:35: undefined reference to `vtkSmartPointerBase::~vtkSmartPointerBase()'
CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o: In function `pcl::visualization::CloudActor::~CloudActor()':
/usr/local/include/vtk-9.0/vtkSmartPointer.h:35: undefined reference to `vtkSmartPointerBase::~vtkSmartPointerBase()'
/usr/local/include/vtk-9.0/vtkSmartPointer.h:35: undefined reference to `vtkSmartPointerBase::~vtkSmartPointerBase()'
CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o:/usr/local/include/vtk-9.0/vtkSmartPointer.h:35: more undefined references to `vtkSmartPointerBase::~vtkSmartPointerBase()' follow
CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o: In function `void pcl::visualization::PCLVisualizer::convertPointCloudToVTKPolyData<pcl::PointXYZRGB>(pcl::visualization::PointCloudGeometryHandler<pcl::PointXYZRGB> const&, vtkSmartPointer<vtkPolyData>&, vtkSmartPointer<vtkIdTypeArray>&)':
/usr/local/include/vtk-9.0/vtkSmartPointer.h:62: undefined reference to `vtkSmartPointerBase::vtkSmartPointerBase()'
/usr/local/include/vtk-9.0/vtkSmartPointer.h:62: undefined reference to `vtkSmartPointerBase::vtkSmartPointerBase()'
CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o: In function `void pcl::visualization::PCLVisualizer::convertPointCloudToVTKPolyData<pcl::PointXYZRGB>(pcl::visualization::PointCloudGeometryHandler<pcl::PointXYZRGB> const&, vtkSmartPointer<vtkPolyData>&, vtkSmartPointer<vtkIdTypeArray>&)':
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:302: undefined reference to `vtkPolyData::GetVerts()'
CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o: In function `void pcl::visualization::PCLVisualizer::convertPointCloudToVTKPolyData<pcl::PointXYZRGB>(pcl::visualization::PointCloudGeometryHandler<pcl::PointXYZRGB> const&, vtkSmartPointer<vtkPolyData>&, vtkSmartPointer<vtkIdTypeArray>&)':
/usr/local/include/vtk-9.0/vtkSmartPointer.h:173: undefined reference to `vtkSmartPointerBase::operator=(vtkObjectBase*)'
CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o: In function `void pcl::visualization::PCLVisualizer::convertPointCloudToVTKPolyData<pcl::PointXYZRGB>(pcl::visualization::PointCloudGeometryHandler<pcl::PointXYZRGB> const&, vtkSmartPointer<vtkPolyData>&, vtkSmartPointer<vtkIdTypeArray>&)':
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:306: undefined reference to `vtkCellArray::GetData()'
CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o: In function `void pcl::visualization::PCLVisualizer::convertPointCloudToVTKPolyData<pcl::PointXYZRGB>(pcl::visualization::PointCloudGeometryHandler<pcl::PointXYZRGB> const&, vtkSmartPointer<vtkPolyData>&, vtkSmartPointer<vtkIdTypeArray>&)':
/usr/local/include/vtk-9.0/vtkSmartPointer.h:102: undefined reference to `vtkSmartPointerBase::vtkSmartPointerBase(vtkObjectBase*)'
CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o: In function `void pcl::visualization::PCLVisualizer::convertPointCloudToVTKPolyData<pcl::PointXYZRGB>(pcl::visualization::PointCloudGeometryHandler<pcl::PointXYZRGB> const&, vtkSmartPointer<vtkPolyData>&, vtkSmartPointer<vtkIdTypeArray>&)':
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:307: undefined reference to `pcl::visualization::PCLVisualizer::updateCells(vtkSmartPointer<vtkIdTypeArray>&, vtkSmartPointer<vtkIdTypeArray>&, long long)'
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:309: undefined reference to `vtkCellArray::SetCells(long long, vtkIdTypeArray*)'
CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o: In function `void pcl::visualization::PCLVisualizer::convertPointCloudToVTKPolyData<pcl::PointXYZRGB>(pcl::visualization::PointCloudGeometryHandler<pcl::PointXYZRGB> const&, vtkSmartPointer<vtkPolyData>&, vtkSmartPointer<vtkIdTypeArray>&)':
/usr/local/include/vtk-9.0/vtkSmartPointer.h:35: undefined reference to `vtkSmartPointerBase::~vtkSmartPointerBase()'
/usr/local/include/vtk-9.0/vtkSmartPointer.h:35: undefined reference to `vtkSmartPointerBase::~vtkSmartPointerBase()'
/usr/local/include/vtk-9.0/vtkSmartPointer.h:35: undefined reference to `vtkSmartPointerBase::~vtkSmartPointerBase()'
CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o: In function `void pcl::visualization::PCLVisualizer::convertPointCloudToVTKPolyData<pcl::PointXYZRGB>(pcl::visualization::PointCloudGeometryHandler<pcl::PointXYZRGB> const&, vtkSmartPointer<vtkPolyData>&, vtkSmartPointer<vtkIdTypeArray>&)':
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:289: undefined reference to `pcl::visualization::PCLVisualizer::allocVtkPolyData(vtkSmartPointer<vtkPolyData>&)'
CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o: In function `void pcl::visualization::PCLVisualizer::convertPointCloudToVTKPolyData<pcl::PointXYZRGB>(pcl::visualization::PointCloudGeometryHandler<pcl::PointXYZRGB> const&, vtkSmartPointer<vtkPolyData>&, vtkSmartPointer<vtkIdTypeArray>&)':
/usr/local/include/vtk-9.0/vtkSmartPointer.h:218: undefined reference to `vtkCellArray::New()'
/usr/local/include/vtk-9.0/vtkSmartPointer.h:277: undefined reference to `vtkSmartPointerBase::vtkSmartPointerBase(vtkObjectBase*, vtkSmartPointerBase::NoReference const&)'
/usr/local/include/vtk-9.0/vtkSmartPointer.h:137: undefined reference to `vtkSmartPointerBase::operator=(vtkObjectBase*)'
/usr/local/include/vtk-9.0/vtkSmartPointer.h:35: undefined reference to `vtkSmartPointerBase::~vtkSmartPointerBase()'
CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o: In function `void pcl::visualization::PCLVisualizer::convertPointCloudToVTKPolyData<pcl::PointXYZRGB>(pcl::visualization::PointCloudGeometryHandler<pcl::PointXYZRGB> const&, vtkSmartPointer<vtkPolyData>&, vtkSmartPointer<vtkIdTypeArray>&)':
[... cut ...]
wrappers/pcl/pcl-color/CMakeFiles/rs-pcl-color.dir/build.make:198: recipe for target 'wrappers/pcl/pcl-color/rs-pcl-color' failed
make[2]: *** [wrappers/pcl/pcl-color/rs-pcl-color] Error 1
make[2]: Leaving directory '....../librealsense/build_18.04'
CMakeFiles/Makefile2:1565: recipe for target 'wrappers/pcl/pcl-color/CMakeFiles/rs-pcl-color.dir/all' failed
make[1]: *** [wrappers/pcl/pcl-color/CMakeFiles/rs-pcl-color.dir/all] Error 2
make[1]: Leaving directory '....../librealsense/build_18.04'
Makefile:132: recipe for target 'all' failed
make: *** [all] Error 2
```

Where to modify for a successful building?

Cheers
Pei

MartyG-RealSense, 2020-03-13:
I have looked at your case very carefully.  I am unfamiliar with VTK.  I am not aware of a need for VTK when using CMake to build Librealsense with PCL support though.

There are instructions for building Librealsense with PCL wrapper support  using Windows or Linux in the link below.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/pcl

jiapei100, 2020-03-13:
@MartyG-RealSense 

Thank you very much for your prompt reply.

Actually, it's just  [https://github.com/IntelRealSense/librealsense/blob/master/wrappers/pcl/pcl-color/rs-pcl-color.cpp](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/pcl/pcl-color/rs-pcl-color.cpp) . Line 33 `#include <pcl/visualization/pcl_visualizer.h>`. 

So, let's take a look at [https://github.com/PointCloudLibrary/pcl/blob/master/visualization/include/pcl/visualization/pcl_visualizer.h](https://github.com/PointCloudLibrary/pcl/blob/master/visualization/include/pcl/visualization/pcl_visualizer.h) .

MartyG-RealSense, 2020-03-14:
Thanks so much for the further details and sharing your solution of solving the error by including a VTK header file.   Do you need further assistance with this case please?   Thanks!

jiapei100, 2020-03-14:
@MartyG-RealSense 

If a VTK header file is included, that means when building librealsense's **PCL examples**, VTK libraries **must be included** as well, right? That will cause **undefined reference to ERRORs**.

So, actually, my final solution is to : NOT building PCL examples for now.... ^_^

Thank you... let's close it for now.

closed 2020-03-14 by jiapei100

MartyG-RealSense, 2020-03-14:
Thanks so much!   I appreciate how thorough you have been with documenting your errors in this discussion so that others can benefit from the information in future.   Good luck with your work.  :)