realsenseai/librealsense #6052: About SLAM & Visual Inertial odometry.

nych9811, 2020-03-13:
Hi. I'm studying SLAM, and I'm going to use D435i for Visual Inertial Odometry.

Thanks to your help, I understood the meaning of calibration parameter, and I also have questions about using IMU in SLAM. In order to use IMU sensor built-in D435i, we must do calibration. But, after I attach D435i camera on drone, it is impossible to doing IMU calibration as shown in tutorial.

So, I think before we test SLAM, we should take off camera, and do calibration in 6-axes as introduced in tutorial, then attach to drone, save parameter, then start testing. It makes be little bit tired.

So, is there any another way to do calibration of IMU sensor?

Please help me!

MartyG-RealSense, 2020-03-13:
I do not know of a means other than the Python calibration script to do the IMU calibration.

In theory though you could avoid taking the camera off by holding the drone that the camera is attached to and tilting the whole drone with your hands, so that the camera changes angle with it because it is mounted rigidly to the drone.  

If you are able to use ROS, Intel has a guide to setting up SLAM with D435i and ROS to make use of the IMU.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

nych9811, 2020-03-13:
Dear @MartyG-RealSense 
Thanks a lot. It makes sense to me.
I have another question. After I finished calibration, I got norm values like these.

norm (raw data ): 9.651282
norm (fixed data): 9.802097 A good calibration will be near 9.806650

Can you explain the meaning of "norm"? I tried hard to understand, but I couldn't. Please help me.

MartyG-RealSense, 2020-03-13:
The value of the strength of the gravity on the planet Earth is usually rounded up to 9.81 ('Newtons' scientific measurement unit).    My rough understanding is that the IMU takes a raw reading and then does motion correction based on known modeling about inaccuracies in the IMU to arrive at the final fixed value (the norm).

nych9811, 2020-03-13:
Perfectly understood. Thanks a lot.

closed 2020-03-13 by nych9811