realsenseai/librealsense #6054: T265 mapper confidence is always 0

smartynenko, 2020-03-13:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0879 | 
| Operating System & Version |   {Win (8.1/10)  | 
| Kernel Version (Linux Only)    |  N/A                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.31.0                          | 
| Language                            |  C                          | 
| Segment			|  Robot/Smartphone                          | 

### Issue Description
I am running rs_pose example with small change to report mapper and tracker confidence:
`// Print the x, y, z values of the translation, relative to initial position
        std::cout << "\r" << "Device Position: " << std::setprecision(3) << std::fixed << pose_data.translation.x << " " <<
            pose_data.translation.y << " " << pose_data.translation.z << " (meters). Mapper: " << pose_data.mapper_confidence << " Tracker: " << pose_data.tracker_confidence;`

Tracker confidence settles quickly and bounces back and forth between 2 and 3, but mapper confidence is always 0. Are there any additional changes required to enable mapping?

neilyoung, 2020-03-16:
I think you are good to go with the tracker confidence. I'm not even sure that this element is relevant vor T265 at all. Tracker confidence 3 is excellent.

arjunsengupta98, 2020-09-04:
Hello. 
Was anyone able to figure out what mapper confidence signifies? 

> Tracker confidence settles quickly and bounces back and forth between 2 and 3, but mapper confidence is always 0.

I am facing this as well. 

I'm trying to obtain a transformation between an imported map and the current map. As suggested on various threads, I tried querying a static node present in the saved map, but the sensor is unable to identify the node.
 
I thought that the issue may be due to mapper confidence being 0. Any idea what the problem may be?.

alessandromorra, 2021-02-03:
Dear @RealSense-Customer-Engineering @RealSenseCustomerSupport @RealSenseSupport 
I am facing this issue as well, can you give some explanations or pointers?
Many thanks!

closed 2026-10-23 by MartyG-RealSense