realsenseai/librealsense #6055: Consider `static` in rsutil.h

jerry73204, 2020-03-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | (no tool on linux) | 
| Operating System & Version |   Arch Linux  | 
| Kernel Version (Linux Only)    |  5.5.6-arch1-1 | 
| Platform                                 | PC  |
| SDK Version                            |  2.23.1 | 
| Language                            |  C | 
| Segment			|  any | 

### Issue Description
Currently I'm working on Rust bindings for librealsense2. I noticed all functions defined in [rsutil.h](https://github.com/IntelRealSense/librealsense/blob/8c2eff00655eeca365f94b2a81f62c3990eda50d/include/librealsense2/rsutil.h#L19)
are static. It would be a trouble to generate FFI for static functions. Because static functions are limited to its object file, they cannot be linked by outer programs. I think it's not necessary to place `static` there since they are somewhat public to C/C++. Removing `static` would give convenience for building other language bindings.

MartyG-RealSense, 2020-03-14:
I will label your message as a Feature Request and leave it open so that the RealsSense developers can track it and consider it.  Thanks!

MartyG-RealSense, 2021-11-16:
Closed due to the issue being addressed in librealsense version 2.49.0.

closed 2021-11-16 by MartyG-RealSense