realsenseai/librealsense #6057: attempt to record .bag file using matlab wrapper, but crashed

FancyBrush, 2020-03-14:
I read the record-playback example in 
https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback
and follow the overview to build matlab version of record-playback. But came across a problem.
Here is my testing code:
`pipe = realsense.pipeline();
colorizer = realsense.colorizer();
pipe.start();
fs = pipe.wait_for_frames();
depth = fs.get_depth_frame();
depth_color = colorizer.colorize(depth);
rgb = fs.get_color_frame();
pipe.stop();
% create a new pipeline
cfg = realsense.config();
pipe = realsense.pipeline();
% matlab crashed if cfg.enable_record_to_rile(...) is executed
cfg.enable_record_to_file("F:\REALSENSE_PROJECT\testrecord.bag");
pipe.start(cfg);
device = pipe.get_active_profile().get_device();
pipe.stop();
`
When matlab execute "cfg.enable_record_to_file()", it crashed. Am I doing record in a right way with matlab?

FancyBrush, 2020-03-14:
pipe = realsense.pipeline();
colorizer = realsense.colorizer();

pipe.start();
fs = pipe.wait_for_frames();
depth = fs.get_depth_frame();
depth_color = colorizer.colorize(depth);
rgb = fs.get_color_frame();
pipe.stop();

% create a new pipeline
cfg = realsense.config();
pipe = realsense.pipeline();
% matlab crashed if cfg.enable_record_to_rile(...) is executed
cfg.enable_record_to_file("F:\CameraSync\REALSENSE_PROJECT\testrecord.bag");

pipe.start(cfg);
device = pipe.get_active_profile().get_device();
pipe.stop();

MartyG-RealSense, 2020-03-14:
I am not a MATLAB programmer but I compared your code to the example programs in the wrapper.  The first thing I noticed is that the very first line has a **function** instruction of the name of the program that is the same as the script's filename.  So if your script file was saved as **rosbag_recorder.m** then the first line of your script would be:

**function rosbag_recorder()**

In other programming languages, the opening and closing of a function may be defined by the open and close brackets { and }.  In MATLAB scripts the word **end** seems to be used instead to close a function.  So on the very last line of the script, put:

**end**

I tried now to organise the rest of the code according roughy to the structure of the wrapper's rosbag reader example, though it may need adjustment.

cfg = realsense.config();
cfg.enable_record_to_file("F:\REALSENSE_PROJECT\testrecord.bag")
pipe = realsense.pipeline();
colorizer = realsense.colorizer();
profile = pipe.start(cfg);
dev = profile.get_device();
name = dev.get_info(realsense.camera_info.name);

% Get frames. We discard the first couple to allow
    % the camera time to settle
    for i = 1:5
        fs = pipe.wait_for_frames();
    end

% NOTE:. You still need an **end** at the script end.  The one here is just closing the **for** loop.

% Stop streaming
    pipe.stop()

% Select depth frame
    depth = fs.get_depth_frame();
    % Colorize depth frame
    color = colorizer.colorize(depth);

% the actual end of script

end

FancyBrush, 2020-03-14:
Thank you for your reply! @MartyG-RealSense .  There might be some misunderstanding.
     The testing code is actualy a script, not a function. Another script I wrote which is aimed to playback a .bag file and save every frame into .avi video file.This script is revised from matlab wrapper rosbag_examplt.m,  dose not crash. Therefore, I don't think the crash is caused by function issue. Instead, if a function script name is not consistent, matlab reports an error, it would not crash .
      I checked the crash report just now, it says : 
> 
This error was detected while a MEX-file was running. If the MEX-file
is not an official MathWorks function, please examine its source code
for errors. Please consult the External Interfaces Guide for information
on debugging MEX-files.
       I still wander if I do the work flow of managing the camera in a right way.

FancyBrush, 2020-03-14:
Thank you for your reply! There might be some misunderstanding. 
     The testing code is actualy a script, not a function. Another script I wrote which is aimed to
playback a .bag file and save every frame into .avi video file.This script is revised from matlab
wrapper rosbag_examplt.m,  dose not crash. Therefore, I don't think the crash is caused by 
function issue. Instead, if a function script name is not consistent, matlab reports an error, 
it would not crash .
      I checked the crash report just now, it says : 
      "This error was detected while a MEX-file was running. If the MEX-file is not an official 
MathWorks function, please examine its source code for errors. Please consult the External 
Interfaces Guide for information on debugging MEX-files."
       I still wander if I do the work flow of managing the camera in a right way.

MartyG-RealSense, 2020-03-14:
@lramati and @ev-mp seem to be the main contributors of example programs in the MATLAB wrappers, so hopefully they can offer advice when they are available to do so.  I apologise for the delay in progressing your work that this may cause.

FancyBrush, 2020-03-14:
Thank you for your detailed explanation. The matlab crash report says the .mex file might be the source of prolem. Thank you for helping me sorting out the matlab contributors in realsense community!

lramati, 2020-03-15:
Do you get any sort of error message?

FancyBrush, 2020-03-15:
Thank you for your attention! @lramati 
MATLAB sent me a crash report:
MATLAB 崩溃文件: C:\Users\WHR\AppData\Local\Temp\matlab_crash_dump.7944-1:

--------------------------------------------------------------------------------
          Access violation detected at 周日 3月 15 06:55:34 2020 +0800
--------------------------------------------------------------------------------

Configuration:
  Crash Decoding           : Disabled - No sandbox or build area path
  Crash Mode               : continue (default)
  Default Encoding         : GBK
  Deployed                 : false
  Graphics Driver          : NVIDIA Corporation GeForce GTX 1050 Ti/PCIe/SSE2 Version 4.6.0 NVIDIA 441.66
  Graphics card 1          : NVIDIA ( 0x10de ) NVIDIA GeForce GTX 1050 Ti Version 26.21.14.4166 (2019-12-6)
  Graphics card 2          : Intel Corporation ( 0x8086 ) Intel(R) UHD Graphics 630 Version 24.20.100.6287 (2018-8-15)
  Java Version             : Java 1.8.0_144-b01 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
  MATLAB Architecture      : win64
  MATLAB Entitlement ID    : 4453461
  MATLAB Root              : D:\MATLAB_2018
  MATLAB Version           : 9.4.0.813654 (R2018a)
  OpenGL                   : hardware
  Operating System         : Microsoft Windows 10 家庭中文版
  Process ID               : 7944
  Processor ID             : x86 Family 6 Model 158 Stepping 10, GenuineIntel
  Session Key              : c3c0d7a3-0676-4086-b686-26fffc9e1bf5
  Window System            : Version 10.0 (Build 18363)

Fault Count: 1

Abnormal termination

Register State (from fault):
  RAX = 0000000000000000  RBX = 0000000000000000
  RCX = 00000000ec397660  RDX = 0000000010a6b5b8
  RSP = 00000000043f8900  RBP = 00000000043f8d50
  RSI = 00000000043f89c0  RDI = 0000000000000000
 
   R8 = 00000001720c8ff0   R9 = 0000000000000002
  R10 = 00000001720cc090  R11 = 00000000043f8880
  R12 = 00000000043f8d50  R13 = 000000000bd9a8f0
  R14 = 00000001720cc080  R15 = 0000000000000000
 
  RIP = 00007ffe03fd3891  EFL = 00010202
 
   CS = 0033   FS = 0053   GS = 002b

Stack Trace (from fault):
[  0] 0x00007ffe03fd3891 F:\CameraSync\REALSENSE_PROJECT\+realsense\librealsense_mex.mexw64+01521809 librealsense::processing_block::processing_block+01382128
[  1] 0x00007ffe03f9a7c6 F:\CameraSync\REALSENSE_PROJECT\+realsense\librealsense_mex.mexw64+01288134 librealsense::processing_block::processing_block+01148453
[  2] 0x00007ffe03fac09a F:\CameraSync\REALSENSE_PROJECT\+realsense\librealsense_mex.mexw64+01360026 librealsense::processing_block::processing_block+01220345
[  3] 0x00007ffe03fdbdda F:\CameraSync\REALSENSE_PROJECT\+realsense\librealsense_mex.mexw64+01555930 librealsense::processing_block::processing_block+01416249
[  4] 0x00000000fc626594                               bin\win64\libmex.dll+00288148 MexRetrieveVersion+00002916
[  5] 0x00000000fc62678c                               bin\win64\libmex.dll+00288652 MexRetrieveVersion+00003420
[  6] 0x00000000fc6268de                               bin\win64\libmex.dll+00288990 MexRetrieveVersion+00003758
[  7] 0x00000000fc612ec3                               bin\win64\libmex.dll+00208579 mexUnlock_800+00024147
[  8] 0x000000001653c51a                     bin\win64\pgo\m_dispatcher.dll+00050458 Mfh_file::dispatch_fh_impl+00001114
[  9] 0x000000001653c0b2                     bin\win64\pgo\m_dispatcher.dll+00049330 Mfh_file::dispatch_fh+00000062
[ 10] 0x0000000016534674                     bin\win64\pgo\m_dispatcher.dll+00018036 Mfunction_handle::dispatch+00000420
[ 11] 0x00000000262d1953                        bin\win64\pgo\mcos_impl.dll+00596307
[ 12] 0x00000000262baf02                        bin\win64\pgo\mcos_impl.dll+00503554
[ 13] 0x00000000262ba62e                        bin\win64\pgo\mcos_impl.dll+00501294
[ 14] 0x00000000262b9e1b                        bin\win64\pgo\mcos_impl.dll+00499227
[ 15] 0x000000002628676a                        bin\win64\pgo\mcos_impl.dll+00288618
[ 16] 0x000000002628625a                        bin\win64\pgo\mcos_impl.dll+00287322
[ 17] 0x0000000026286052                        bin\win64\pgo\mcos_impl.dll+00286802
[ 18] 0x00000000262b957f                        bin\win64\pgo\mcos_impl.dll+00497023
[ 19] 0x000000001653b499                     bin\win64\pgo\m_dispatcher.dll+00046233 Mfh_MATLAB_fn::dispatch_fh+00000297
[ 20] 0x0000000016534674                     bin\win64\pgo\m_dispatcher.dll+00018036 Mfunction_handle::dispatch+00000420
[ 21] 0x00000000179b2aa1                            bin\win64\pgo\m_lxe.dll+00404129
[... 52 lines cut ...]
[ 74] 0x00000000fcfce722                                  bin\win64\iqm.dll+00255778 iqm::Iqm::deliver+00003730
[ 75] 0x00000000fcfcf749                                  bin\win64\iqm.dll+00259913 iqm::Iqm::deliver+00007865
[ 76] 0x00000001001aa211                        bin\win64\libmwservices.dll+01810961 services::system_events::PpeDispatchHook::dispatchOne+00030865
[ 77] 0x00000001001b7ff3                        bin\win64\libmwservices.dll+01867763 sysq::addProcessPendingEventsUnitTestHook+00004659
[ 78] 0x00000001001b81e0                        bin\win64\libmwservices.dll+01868256 sysq::addProcessPendingEventsUnitTestHook+00005152
[ 79] 0x00000001001b9576                        bin\win64\libmwservices.dll+01873270 sysq::getCondition+00003446
[ 80] 0x00000001001bb0e6                        bin\win64\libmwservices.dll+01880294 svWS_ProcessPendingEvents+00000230
[... cut ...]
[ 89] 0x000000014000862f                               bin\win64\MATLAB.exe+00034351
[ 90] 0x00007ffe77e47bd4                   C:\WINDOWS\System32\KERNEL32.DLL+00097236 BaseThreadInitThunk+00000020
[ 91] 0x00007ffe7816ced1                      C:\WINDOWS\SYSTEM32\ntdll.dll+00446161 RtlUserThreadStart+00000033

This error was detected while a MEX-file was running. If the MEX-file
is not an official MathWorks function, please examine its source code
for errors. Please consult the External Interfaces Guide for information
on debugging MEX-files.

Thank you for your time!

lramati, 2020-03-15:
There is an unfortunate implementation detail in how Matlab handles strings vs char arrays, and C++ can't access string data. If you switch to `cfg.enable_record_to_file('FILEPATH');` it should work. I will update the existing input validation to require char arrays in an upcoming version

FancyBrush, 2020-03-15:
It is the format issue, Thank you for your help!
     I'm not so professional in MATLAB object oriented programming, so it is somewhat difficult for me to configure the camera parameters such as minimum/maximum depth, resolution, fps and so on.
     Can I require any samples about configuring the camera parametres? 
     So glad for your guidance!

lramati, 2020-03-15:
Right now we don't have a ton of examples written in Matlab, but the API is almost identical to the C++ API and python APIs. So if you can understand those examples, accomplishing things in Matlab is basically the same

FancyBrush, 2020-03-15:
Understood,  @lramati 
This issue can be closed ! Thank you everyone !

closed 2020-03-22 by MartyG-RealSense