realsenseai/librealsense #6058: can't use D435 after update Firmware

fenneishi, 2020-03-14:
| | |
|---|---|
|**librealsense**|2.33.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435|
|**Serial Number**|819612071062|
|**Firmware Version**|05.12.02.100|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|818413023426|
|**Firmware Update Id**|818413023426|

Please provide a description of the problem

MartyG-RealSense, 2020-03-14:
If you are able to access the RealSense Viewer program, does the camera appear at all in it when inserted into the USB port please?

fenneishi, 2020-03-14:
> If you are able to access the RealSense Viewer program, does the camera appear at all in it when inserted into the USB port please?

I could have used it normally,but when I updated the firmware,it failed to connect.
And  I can't update the firmware again
[image]
[image]

fenneishi, 2020-03-14:
> If you are able to access the RealSense Viewer program, does the camera appear at all in it when inserted into the USB port please?
> 
> **Edit** Thanks for adding the Viewer images!

thanks for your help!

MartyG-RealSense, 2020-03-14:
It is a good sign that the camera is displayed in the Viewer.   Could you go to the 'More' icon at the top of the Viewer's options side-panel please and select ''Install Recommended Firmware' from its menu.  This should cause the Viewer to try to update the firmware.  You do not need to provide a firmware file for this option, as the recommended file is included within the Viewer package.

fenneishi, 2020-03-14:
thanks very much for your suggest !
but it's not working.
when select  ''Install Recommended Firmware',firstly it will show this for a few seconds
[image]
and then will show this:
[image]

MartyG-RealSense, 2020-03-14:
Thanks for your patience and the additional images you posted.

Could you check please whether the camera works with the Depth Quality Tool program (which is similar to the Viewer) to eliminate whether it is a problem with the Viewer program specifically please?

fenneishi, 2020-03-14:
Thanks for your patience too.
 Depth Quality Tool program can't work, It's always in the launch page like this:(already at least 5min before I close it)
[image]
and the rs_examples is also not work(I use VS2019)
but they were normal before
should I use  Firmware Update Tool?

MartyG-RealSense, 2020-03-14:
The firmware update tool (**rs-fw-update** may be worth a try, though the firmware looks healthy.  The tool has the ability to recover cameras if the camera can be detected.  My recollection though is that it needs a USB 3 connection and that only the RealSense Viewer firmware update method works with USB 2.

https://dev.intelrealsense.com/docs/firmware-update-tool

If the firmware is okay and there is some other cause of the problem (such as a corrupted calibration table inside the camera, which has been known to disrupt firmware update) then you could try a "Gold Reset" of the camera to its factory defaults, which fixes the calibration table too.

This is done by installing a program called the Dynamic Calibrator and using a program within it called **CustomFW** to perform the reset.  The details of doing so are in the link below.

The Dynamic Calibrator software can be downloaded here:

https://downloadcenter.intel.com/download/28517/Intel-RealSense-D400-Series-Calibration-Tools-and-API

Download the upper of the two links at the side of the page, **CalibrationTool-2.6.8.0-Setup.zip**

When it is installed, use the CustomRW program to activate a gold reset:

**Intel.Realsense.CustomRW.exe -g**

fenneishi, 2020-03-14:
> The firmware update tool (**rs-fw-update** may be worth a try, though the firmware looks healthy. The tool has the ability to recover cameras if the camera can be detected. My recollection though is [...]

Amazing,"RealSense Viewer program" is working properly again!!!!
But I did nothing！
I just went to the kitchen and ate an apple ,and then restart the "RealSense Viewer program" 

Thanks for your patience very much!!!! 
thanks!!!!

MartyG-RealSense, 2020-03-14:
You are very welcome.  I'm glad that your problem was resolved!  

I will close the case now, as your camera is working again.   Good luck!

closed 2020-03-14 by MartyG-RealSense