realsenseai/librealsense #6059: do not enable the Frame Metadata

fenneishi, 2020-03-14:
| | |
|---|---|
|**librealsense**|2.33.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435|
|**Serial Number**|819612071062|
|**Firmware Version**|05.12.02.100|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|818413023426|
|**Firmware Update Id**|818413023426|

if I click "Enable",then the Viewer is wrong(described in the previous problem)
[image]

MartyG-RealSense, 2020-03-14:
I tested this on my own D435i on a USB 2 connection on Windows.  When I clicked the 'Enable' button, I was asked by a Windows pop-up (not a RealSense one) to give permission to access the program and I clicked OK   I did not encounter the problem that you did.

For others reading this, the previous issue referenced was this one:

https://github.com/IntelRealSense/librealsense/issues/6058

dorodnic, 2020-03-16:
Issue description is not clear

closed 2020-03-16 by dorodnic