realsenseai/librealsense #6060: Calibration at 2ft

tispratik, 2020-03-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.03.00| 
| Operating System & Version |   Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.15.0-88-generic                                         | 
| Platform                                 |    - |
| SDK Version                            |  -                          | 
| Language                            |  python                         | 
| Segment			|  Robot                          | 

### Issue Description
Our installation has camera installed at 24" height facing a conveyor belt. The dynamic calibrator says that we should place the target between 600mm (23.6 inches) and 850mm from the camera. We use the iPhone8 target app for calibration. I wonder if the calibration is of good quality as the target is almost outside of the lower range. We are looking for millimeter level accuracy.

RealSenseCustomerSupport, 2020-03-18:
----------------------------------------------

The depth performance of a well-calibrated camera has an z-accuracy of ≤ 2% at 2 meter. Please refer to the RealSense D400 Calibration User guide here:
https://dev.intelrealsense.com/docs/intel-realsensetm-d400-series-calibration-tools-user-guide#section-7-2-accuracy-check
Thank you.

tispratik, 2020-03-19:
2 meters = 78.74 inches.
Error ~ 78.74 * 0.02 = 1.57 inches.

How does the error translate to at 24" of installation? Is it safe to assume that the error will be 24 * 0.02 ~ 0.48 inches = 12 mm ?

agrunnet, 2020-03-19:
Please look at equation in this white paper
https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance

RealSenseCustomerSupport, 2020-03-23:
----------------------------------------------

Please let us know if there is any question.
Thank you.

tispratik, 2020-03-24:
The linked document doesn't really answer the question.

agrunnet, 2020-03-24:
The equation and graph show the RMS error (precision) and it scales as the square of distance away. The absolute accuracy scales also quadratically, with the limit being the RMS error. It is possible there is a bias and you may want to correct that with the new Tare function that is now included in the viewer. You can read more about that in the newly released white paper on “Self-Calibration”.

RealSenseCustomerSupport, 2020-03-30:
----------------------------------------------

Please let us know if there is any update or question. Thank you.

closed 2020-04-06 by RealSenseCustomerSupport