realsenseai/librealsense #6061: What dose "system_time" in rosbag file save by realsense Viewer stands for ?

FancyBrush, 2020-03-15:
I open the metadata in rosbag file in matlab, and find the variable "system_time".
    Is it the exact time when this frame was recorded in computer OS?
    If it is computer time of this frame, it marks the arrival of this frame or time of accomplishing saving it?
[image]
    I also plotted the system_time, which is not a straight line, maybe bacause OS delay.
[image]
So looking forward to having this confusion solved!!

FancyBrush, 2020-03-15:
I check the real time metadata and found that "system_time" is equal to "time of arrival" and backend timestamp. But 1584275500.255 seconds is 50 years. What's going on here ?
[image]

ev-mp, 2020-03-16:
@FancyBrush , the `system time` is host-generated timestamp of frame when it was captured by the SDK and measured in [Unix's EPOCH](https://en.wikipedia.org/wiki/Unix_time) that actually applies to MS as well.
Documentation link - https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1frame.html#a25f71d45193f2f4d77960320276b83f1.
Also see #2188

RealSenseCustomerSupport, 2020-04-03:
----------------------------------------------

Hi @FancyBrush,

Do you need any further clarifications regarding this topic?

Thank you!

FancyBrush, 2020-04-10:
@RealSenseCustomerSupport 
Problem solved, you may close this issue.

closed 2020-04-10 by FancyBrush