realsenseai/librealsense #6062: 15/03 16:32:14,498 WARNING [546761056640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

kaushik98y, 2020-03-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2020-03-16:
Other Jetson users have recently been experiencing variations on this error in the discussion linked to below.  It remains an open case at the time of writing this.  It may be worth adding a comment to that discussion to highlight your own problem.

https://github.com/IntelRealSense/librealsense/issues/4031

Have you tried the JetsonHacks installation guide for Nano please?

https://github.com/JetsonHacksNano/installLibrealsense

kaushik98y, 2020-03-16:
yes i have tried the jetsonhacks guide, even did the kernel patch. Could upgrading the kernel solve the issue? Currently i have 4.9.140-tegra kernel version, would kernel version 5 solve the issue as it supports librealsense2-dkms ??

MartyG-RealSense, 2020-03-16:
If you want to test for whether the issue may be kernel related, you could try the "bypass" method of installation for Librealsense that does not depend on kernel versions or patches.  The method below requires SDK version 2.30.0 or newer and an internet connection.

The CMake instruction for the bypass in SDK 2.30.0 and newer is:

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

dorodnic, 2020-03-16:
I think @kaushik98y already using RSUSB because of `messenger-libusb.cpp` in the error. 
It's not entirely clear from the description if the issue happens all the time, or only some times. Also, I'd recommend checking if different USB setup can give better results. We have seen issues with some kinds of USB hubs that do not reproduce when connecting directly, also there is USB-C connector I think (I may be confusing with the Xavier)
In general, @doronhi is working on stability on Jetson, and @ev-mp is looking into issues introduced by kernel 5. So I hope we will have a solution soon

jorgemia, 2020-03-26:
I'm running Realsense 2.33.1. and D435i firmware 05.12.03.00 and also see similar issues. 

I get `[548212306368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61` when starting the depth stream of the D435i with a Jetson Nano in realsense viewer. 

Then I sometimes get `WARNING [548212306368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`.

I also have ros-realsense 2.2.13 set up and get
 `[548212306368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61` 
and 
`[ WARN] [1585260191.657304589]: Hardware Notification:Motion Module failure,1.58526e+12,Error,Hardware Error`
when launching the ros node.

MartyG-RealSense, 2020-03-27:
@jorgemia   ROS wrapper 2.2.13 was the first one to officially support Melodic.  My understanding is that earlier wrapper versions could work unofficially with Melodic though.  In JetsonHacks' Jetson Nano installation guide in October 2019, they used SDK 2.25.0.  

https://github.com/IntelRealSense/librealsense/releases/tag/v2.28.0

They say that the ROS wrapper they used was 2.28.0, but I think they meant **2.2.8**.  

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.8

JetsonHacks said they did not encounter problems with using Melodic with this setup, even though it wasn't officially supported by the ROS wrapper at the time. 

The JetsonHacks Nano guide is here:

https://www.jetsonhacks.com/2019/10/25/realsense-ros-wrapper-jetson-nano/

jorgemia, 2020-03-27:
@MartyG-RealSense it seems that for L4T 32.3.1  (which I'm running) Jetson Hacks used realsense 2.33.1 and ros 2.2.11. I'm going to give that a try. 

If I want to install a specific version using the prebuilt packages, how can I choose the version to install? (Following [distribution_linux.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md)) Do I have to install from source?

MartyG-RealSense, 2020-03-27:
If you have unusual hardware that differs from the average desktop PC, compiling from source is recommendable in order to make sure that the build matches up well to the hardware.

Having said that, I recalled that JetsonHacks released an update to their installation guide in December 2019, where they stated that starting with L4T 32.2.1 and Librealsense SDK 2.30.0, installation on Jetson had become easier.

https://github.com/JetsonHacksNano/installLibrealsense

jorgemia, 2020-03-27:
Yes I'm aware, I have a Jetson Nano and followed his guide previously. Just want to know if there is a way to select the version to install using apt-get packages?

When I run this `sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u` and continue with the other steps, I believe it installs the latest realsense version available.

MartyG-RealSense, 2020-03-27:
@jorgemia   I feel that the key phrase here is probably "version control" to achieve what you are seeking to do with apt-get.  It is not a subject I am an expert on though at present.

I did learn though that JetsonHacks has a Releases page with a list of past repositories for their Jetson Nano installation, if that is of any use.

https://github.com/JetsonHacksNano/installLibrealsense/releases

jorgemia, 2020-03-27:
@MartyG-RealSense Ok so I installed v2.31.0 from source using the JetsonHacks build script but I still get the following warning when running realsense-viewer stereo module:

`[548212306368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
`

When turning the stereo module **off**, I sometimes get:
` 27/03 12:48:59,352 WARNING [546727518656] (messenger-libusb.cpp:96) usb_request_cancel returned error, endpoint: 130 error: Success, number: 0`

As a side note, when building from source I got warnings of the following style at times:

```
/home/jorge/librealsense/third-party/glad/glad.c:902:23: warning: ISO C forbids conversion of object pointer to function pointer type [-Wpedantic]
  glad_glGetIntegerv = (PFNGLGETINTEGERVPROC)load("glGetIntegerv");
                       ^
```
I don't know if this affected the installation?

MartyG-RealSense, 2020-03-27:
@doronhi the RealSense ROS wrapper developer has said the following about the control_transfer error:

********************

The warning message "control_transfer returned error" reflects communication error with the device that is resolved by repeating the attempt. These messages are still under investigation but as long as they appear only now and then, say at the beginning of the run, they can safely be ignored for now.

I believe that if you can see the topics in RViz then as far as realsense2_camera node is concerned they are published correctly.

tispratik, 2020-03-30:
We're seeing the same issue in a two Realsense camera setup on a regular beefy machine setup with Intel(R) Core(TM) i9-9900K CPU @ 3.60GHz, 32 GiB RAM, Realsense SDK 2.33.1. One of the cameras keeps showing this error. Following is what i see in the debug logs.

```
 30/03 00:00:00,932 WARNING [140510995904256] (types.cpp:49) set_xu(ctrl=1) failed! Last Error: Success
 30/03 00:00:00,932 ERROR [140510995904256] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: set_xu(ctrl=1) failed! Last Error: Success
 30/03 00:00:07,946 WARNING [140510995904256] (types.cpp:49) set_xu(ctrl=1) failed! Last Error: Success
 30/03 00:00:07,946 ERROR [140510995904256] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: set_xu(ctrl=1) failed! Last Error: Success
 30/03 00:00:14,959 WARNING [140510995904256] (types.cpp:49) set_xu(ctrl=1) failed! Last Error: Success
 30/03 00:00:14,959 ERROR [140510995904256] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: set_xu(ctrl=1) failed! Last Error: Success

30/03 00:05:45,220 ERROR [140511408912128] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 30/03 00:05:50,609 WARNING [140511051962112] (types.cpp:49) set_xu(ctrl=1) failed! Last Error: File exists
 30/03 00:05:50,609 ERROR [140511051962112] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: set_xu(ctrl=1) failed! Last Error: File exists
 30/03 00:05:51,215 ERROR [140511408912128] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 30/03 00:05:56,622 WARNING [140510995904256] (types.cpp:49) set_xu(ctrl=1) failed! Last Error: Success
 30/03 00:05:56,622 ERROR [140510995904256] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: set_xu(ctrl=1) failed! Last Error: Success
 30/03 00:06:02,635 WARNING [140511051962112] (types.cpp:49) set_xu(ctrl=1) failed! Last Error: File exists
 30/03 00:06:02,635 ERROR [140511051962112] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: set_xu(ctrl=1) failed! Last Error: File exists
 30/03 00:06:03,205 ERROR [140511408912128] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 30/03 00:06:08,649 WARNING [140510995904256] (types.cpp:49) set_xu(ctrl=1) failed! Last Error: Success
 30/03 00:06:08,649 ERROR [140510995904256] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: set_xu(ctrl=1) failed! Last Error: Success
 30/03 00:06:14,662 WARNING [140511051962112] (types.cpp:49) set_xu(ctrl=1) failed! Last Error: File exists
 30/03 00:06:14,662 ERROR [140511051962112] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: set_xu(ctrl=1) failed! Last Error: File exists
 30/03 00:06:15,196 ERROR [140511408912128] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 30/03 00:06:20,675 WARNING [140510995904256] (types.cpp:49) set_xu(ctrl=1) failed! Last Error: Success

 30/03 13:42:10,021 WARNING [139879551850240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 30/03 13:42:11,102 WARNING [139879551850240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 30/03 13:42:12,183 WARNING [139879551850240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 30/03 13:42:16,325 WARNING [139879551850240] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 30/03 13:42:19,447 WARNING [139879551850240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 30/03 13:42:21,548 WARNING [139879551850240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 30/03 13:42:21,608 WARNING [139879551850240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 30/03 13:42:22,689 WARNING [139879551850240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 30/03 13:42:23,770 WARNING [139879551850240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 30/03 13:42:23,830 WARNING [139879551850240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 30/03 13:42:24,911 WARNING [139879551850240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```

MartyG-RealSense, 2020-03-30:
@tispratik  So the messages above are appearing continuously instead of only at the start of a run?

tispratik, 2020-03-30:
I restarted the cameras. I still keep seeing these messages continuously. Generally what happens is, after 24+ hours of running, the second camera fails to get frames, thats when we have to restart both cameras and the same cycle repeats.

MartyG-RealSense, 2020-03-30:
I found a case where this error is occurring: **set_xu(ctrl=1) failed! Last Error: Success**

The user in that case had the idea of configuring hardware_reset to reset automatically when the error "frame not received" occurred.

I was not able to find a method for preventing this error from occurring though.

tispratik, 2020-03-30:
Thanks Marty. Its interesting. We are running the camera at 1 frame every 2 seconds. Could that be a cause of the problem?

MartyG-RealSense, 2020-03-30:
In general, errors are more likely to occur when using an FPS speed of less than 15 FPS.  6 FPS especially can be erratic.

MartyG-RealSense, 2020-04-12:
Closed due to no further comments being received.

closed 2020-04-12 by MartyG-RealSense

Abdob, 2020-09-23:
For getting rid of the resource not available problem , it definately has to do with the usb drivers as libusb is complaining: disabling autosuspend seems to get rid of it:
echo "bash -c 'echo -1 > /sys/module/usbcore/parameters/autosuspend'" >> ~/.bashrc

reinzor, 2021-04-13:
> echo "bash -c 'echo -1 > /sys/module/usbcore/parameters/autosuspend'" >> ~/.bashrc

Did not work for us

MartyG-RealSense, 2021-04-13:
Hi @reinzor  If you are using Jetson Nano, the link below leads to a librealsense installation guide that has worked consistently with Nano.

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

chrissunny94, 2022-04-22:
`kernal upgrade seems to have fixed it for me`

PR #6083 (2020-03-25): Fix 100% CPU usage when using libusb
Replace broken dispatcher with a thread to fix 100% CPU usage

The dispatcher can't tell the difference between "timeout" and "empty", so it ends up infinte looping.  Using a thread directly is much simpler and more correct.

Fixes #5783.  Fixes #6062.