realsenseai/librealsense #6063: How to use Realsense Cameras D435 and T265 for navigation

demoncoderad, 2020-03-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 - D435(Without IMU) | 
| Firmware Version                   | v2.33.1 (Latest after 1 March 2020)| 
| Operating System & Version |   {Win (8.1/10)   | 
| Kernel Version (Linux Only)    |  X                                      | 
| Platform                                 | PC  |
| SDK Version                            | 2.33.1 }                          | 
| Language                            |  opencv/python }                          | 

### Issue Description

I want to use Intel Realsense cameras D435 (without the imu) and T265 together for navigation. I have seen examples of end results but I need some example code to implement in my project. I also want to announce where to go next like if the person needs to go Left after 2m then it should say "Left after 2 meter". Please provide me with some knowledge on how to do it or link to examples either by Intel or Others

neilyoung, 2020-03-15:
This could be a start

https://neilyoung.serveblog.net

It provides you with a stable GPS context (or xyz if you like). I think to know, where you (or the camera) is, is a pre-requirement to decide, where to go if there is a target.

demoncoderad, 2020-03-15:
And if i want to make a sort of map and then also know where I am then? Target is on map

neilyoung, 2020-03-15:
I'm sorry, could you try to clarify your problem?

demoncoderad, 2020-03-15:
I want to first make a map of the surroundings where it will go. Then I want to know where the cameras are and navigate to another point in the map. I also want to announce the next turn or directions for going.

neilyoung, 2020-03-15:
Ah ok. Thanks for the clarification. I'm sorry I don't have experiences with maps taken from depth cams. From all I know you will be able to create a 3D map of the environment, from which it will be possible to derive the position of the camera and of objects in the environment. But I guess those will always just positions relative to the camera itself. 

I would bet you would be better off doing "plain old map navigation", having a GPS coordinate to start from, a destination GPS coordinate and a bunch of routes possible between both. Then you could follow one of these routes and use the depth cam to identify obstacles and find a collision free paths around.

But I'm not really an expert in depth/tracking fusion, I'm sorry.

demoncoderad, 2020-03-16:
Coordinate system would be better but Ultimately I am using this for real people
I think i can make a 2D map for navigati9n after I know where I am. This leaves 5he problem of identifying where the cameras are. Can you help me in this task

demoncoderad, 2020-03-16:
And previously I had the idea of navigation probes. So they would be in the house all around 
If 2 Probes are visible than the camera would detect them and calculate position on the map with triangulation

neilyoung, 2020-03-16:
>  Can you help me in this task
I can tell you with all the precision the T265 gives, where you are in either the WGS84 coordinate system or a cartesian system with an origin of your choice. Reproducible and stable. Not more, not less. You can't achieve this with the pure T265 SDK, there is some outer computation necessary. This is done in my python scripts and supported/presented in the web app linked above.

demoncoderad, 2020-03-16:
I am sorry but I don't understand the terms WGS84 and cartesian system. Please elaborate and explain

neilyoung, 2020-03-16:
WGS84 = GPS. Cartesian = The well known X/Y 2D diagram.

demoncoderad, 2020-03-16:
Actually I am not that great programmer but what does this script do https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud if you know

neilyoung, 2020-03-16:
AFAIK it creates a 3D model of your environment, which you can rotate, scale and zoom.

demoncoderad, 2020-03-16:
So in your opinion what is the best way to know where you are using these cameras? Please explain in noob programmer's words

neilyoung, 2020-03-16:
Well, in my opinion it would be awesome to have the ability to see the position of the cam (or the device, which carries it) on a floor plan or a map. This is what you can do with my tool.

I'm preparing a new video which probably makes the benefit of my work more obvious. In the meantime you can checkout some older videos. You'll find the links in the documentation section of https://neilyoung.serveblog.net

demoncoderad, 2020-03-17:
Previously you were telling that it needs a End Point as well as The Starting Point ?

neilyoung, 2020-03-17:
Not sure, where I told about an end point... Hmm..

Well, maybe it is a bit difficult to understand. I was hoping, that my solution as well as the very detailed documentation tells it all, but obviously it does not.

This is not a big problem. If you don't understand that, then it is more or less my problem to make it better understandable, what my solution makes. Not because it is hard, but because it's easy :)

Let me try again in other words.

Basically, the T265 and its software works like our sensory organs: it is able to orient itself in space thanks to its "eyes" (the camera's lenses) and its organ of balance (the IMU), analogous to our ears. Her brain is the memory, remembering the environment in the form of a 3D model (a kind of map).

Now imagine that the camera is about as smart at every "awakening" as you are when you wake up one morning in a completely unknown room without windows. Like you, the camera is able to recognise the dimensions of the space surrounding it, but none of you has the slightest idea where you are in the world.

Now someone opens a door and you immediately recognise where the sun rises. You are immediately aware of where the other points of the compass are now, and if that someone now additionally tells you which city you are in, then you will also be able to find your way home. Even more, if surprisingly someone locks you up in this room again, you will immediately remember, where this door was, just from the inventory of the room. And all the rest comes back to mind automatically...

The same is true for the camera (supported by my software) from the moment you told it for the first time: "Hello software, you are now standing right here and looking in this and that direction" ("set_marker" operation). My software then marks this point to the camera, in her map. From then on the algorithm has the context and can blithely chat about where it is, into what direction it is going and all that, even though, how fast it is, but all that now in a bigger, outer context - the world reference frame. 

Actually, not the camera can this out of the box, but my software as a proxy with help of the camera can, only because someone showed me the sun through the door once and told me what city it is. Figuratively. And I'm doing some math on the poses delivered by the camera, once this is clear, before I forward coordinates to you.

And if I load the marked map back into the camera the next time in the same environment, what do you think will happen? She looks around, recognises the room and after she remembers: "Ah here right behind me is the marked point, which you told me is important, do you wanna know?". She tells my software about her flash back and I combine that with  "Someone opened the door last time and said: This is East, and this is Sparta". Now I know - and the coordinates flow again :) Correctly, properly, reproducibly. Well, if the lighting is OK, if the furnitures haven't changed al lot and all the restricting conditions which apply to Visual SLAM are all eliminated for the moment....

And no - I don't need an end point. For the initial mapping of the surroundings, I need someone to show me the direction and tell me the city.

The T265 is perfectly adequate for this functionality. The D435 could do that as well, but in other ways (which are not so clear to me). In combination with the T265, it is certainly a great, though perhaps a little too expensive, help with obstacle detection (which is certainly easier to do with ultrasound). 

I hope that wasn't too flowery, but there's nothing more to it.

HTH

demoncoderad, 2020-03-17:
Thank You For The Great Explanation. In my comment above, the end point I was talking about is when I have saved one map. The initial mapping. After that when I switch On the T265, then through your software, can we get the point where the camera is on the map (Which you told above) ? If yes then can we download the recorded map ? Please tell this last question along with that what format the map is in

demoncoderad, 2020-03-17:
Your documentation was very detailed. Actually it is my problem that I don't understand many terms Easily. I am a minor and English is not my native language so I don't get some words easily

neilyoung, 2020-03-17:
Sure. You can download the map from the cam. But it would not help you. It is only relevant for the cam and the format is completely undocumented

neilyoung, 2020-03-17:
> Your documentation was very detailed. Actually it is my problem that I don't understand many terms Easily. I am a minor and English is not my native language so I don't get some words easily

Ask what you don’t understand

demoncoderad, 2020-03-17:
In which pattern do we have to take the camera along the map that we have made for it to record everything ?

neilyoung, 2020-03-17:
What do you mean? I’m sorry I don’t understand

demoncoderad, 2020-03-17:
I mean that at what pace and how well do we have to move the camera in the surroundings for it to map properly

demoncoderad, 2020-03-17:
And can I use your web app through the python script i.e. can I use them offline?

neilyoung, 2020-03-17:
I have no idea what the maximum pace is. I just tested with max 5-10 km/h pedestrian speed and once I drove with my car around the block, maybe at 30. That was no problem. I expect the algorithm to be capable to support a much higher pace. I mean it can produce 200 poses per second, so it should be able to provide a fix every 27 cm at 200 km/h. Given, the environment can still be recognised correctly at that speed and doesn't just "warp by"...

Also pure rotations have not been a problem at all. Right now my algorithm supports a forward pointing camera only, which does not mean, that you can't go backward. Next step will be to handle downward pointing cameras (for drone usage)

The Python script has no online dependency, while the web app has (since the coordinate display happens on either a Google Map or OpenStreetMap). But this does not necessarily have to be so forever. It is easily possible to remove the Internet dependency by bringing down the coordinate display to floor plan level, aka graphics level.

Other than that the web app is just an optional thing. You don't need it.

demoncoderad, 2020-03-17:
Last thing to understand, Say I have done the initial mapping. And now I switch of my camera. Disconnect it and come back tomorrow. Then I connect it and switch it on, will The python scripts be able to detect it where it is on the map that I drew for my reference. If yes then what coordinate will it give as its position

neilyoung, 2020-03-17:
You will have to load the map to the camera. Since the "magic point"  (the marker) is stored in the map, the camera will - if it recognises the environment again - give me the coordinates of the magic point. i her _current_ reference frame (which might be completely differently as during the previous run). But  I will be able to restore your _current_ real world position and pose from that. So you don't have to start from the same point. You don't have to repeat the route exactly as you did with initial mapping. You can start from whatever point in the environment. If the camera has seen this site it will remember the magic point and give it to me. This is the key for all what comes.

What you can't do is this: You map one room and on next occasion you go into a completely different room, with no relation to the first. Then the camera will not be able to restore the magic point.

neilyoung, 2020-03-17:
Just simply give it a try. It is so easy. https://neilyoung.serveblog.net

Python scripts here: 

https://github.com/neilyoung/t265pose/releases

demoncoderad, 2020-03-17:
Well That is just super awsome. I appreciate your making of this Web App and python scripts

demoncoderad, 2020-03-17:
I will surely give it a try

neilyoung, 2020-03-17:
Thanks for the encouragement. The feedback is not that big and it is always nice to be pushed to continue. 

I will come out with a new release today or tomorrow. It will bring the ability to record a video stream along with the poses (for debugging and accuracy assessments) and to playback the positions from a recording. I will also produce a new video, which will show a split screen: One half the original video seen by the cam. Other half: The positions displayed on a floor plan.

Then I guess I will proceed with the offline version and maybe I will make the python app host the web app, so that it is all in one package and you can run the web app from the device, which hosts the API. It must not necessarily reside on the web.

These are the short term plans, but I have no pressure on it. As said - spare time activity.

neilyoung, 2020-03-17:
FYI: https://youtu.be/ogALlpxyApU

neilyoung, 2020-03-18:
I have uploaded 0.5.0 to https://github.com/neilyoung/t265pose (releases), which bundles the web app (https://neilyoung.serveblog.net) with the Python scripts. You will now have the same web app available at `localhost:4000` hosted by the Python scripts. This should boost your start and allow instant success.

RealSenseSupport, 2021-01-07:
Thanks so much @neilyoung and @demoncoderad for this conversation and interaction.  @neilyoung you have done amazing work with the T265, very commendable.  

Does this thread need to stay open at this time?

demoncoderad, 2021-01-09:
Hi, and I do not believe this thread needs to be open at the time, I think i'll close it, Thanks to @neilyoung for his great help.

demoncoderad, 2021-01-16:
Hi @neilyoung again
I have come back to this project recently. I want to ask given all the WGS84 coordinates and the XYZ coordinates (using a map, giving the width and breadth) How can I get like a whole number indexed location of the camera. Like say each of tiles are 25 cm wide and high, so how can I figure out how to get their index as if they were a 2D array. Or if the XY coordinates are already a indexed location, are they indexed from center of image or the top left, or the starting point.

neilyoung, 2021-01-16:
Well, if you have an origin of your coordinate system (say bottom/left is (0,0)) then you need to make sure to receive reproducible results (one way would to always start the camera at 0,0 (which is impractical) or to use my code, which stores the geo-reference in the map and aligns the delivered coordinates after re-determining the environment and retrieving the geo-reference). 

Once you have that, you simply take the x/y coords and multiply each by 4. This gives you the coordinate in your tile coordinate system (so if Y points forward and X right and the raw X/Y would be 4/8, then your tile coordinate would be 16/32)

If I got that right.

demoncoderad, 2021-01-16:
So correct me if I'm wrong, If my marker set in georeferencing tab is on 0,0 according to my tile based system, the X/Y will be relative to that particular point?
Also, using curl through pycurl getting the coordinates (as mentioned in docs) would work the same?

neilyoung, 2021-01-16:
> So correct me if I'm wrong, If my marker set in georeferencing tab is on 0,0 according to my tile based system, the X/Y will be relative to that particular point?

Sure

> Also, using curl through pycurl getting the coordinates (as mentioned in docs) would work the same?

Not sure what you mean. Are you talking about my API? Then yes.

demoncoderad, 2021-01-19:
Ok thanks neil.
So I don't know its a deal with what, but with the T265, starting in like one cm of the area where i've put the marker, the marker sometimes gets outside of the map. The dimensions are perfectly correct. Its kinda like I start from one corner of room, and goto the an adjacent corner and its outside the walls. Happens everywhere. So is this about me walking too fast or not enough light or something?

neilyoung, 2021-01-19:
I think you should be prepared for a drift of +-50cm. You might improve accuracy with more features and excessive mapping. 

Reagan ding the marker: Would you mind to elaborate a bit more on your solution?

closed 2026-10-23 by MartyG-RealSense